Koenkk/zigbee2mqtt #9776: Develco HESZB-120 also available with "frient A/S"

ffery, 2021-11-21:
Hi there,

I bought some frient fire/smoke sensors ( one SMSZB-120 which got configured without problem, with provider marked as "frient A/S"), the HESZB-120 and couldn't be configured.

    Info Successfully interviewed '0x0015bc0034000e09', device has successfully been paired
    Warning Device '0x0015bc0034000e09' with Zigbee model 'HESZB-120' and manufacturer name 'frient A/S' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
    Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x0015bc0034000e09","ieee_address":"0x0015bc0034000e09","status":"successful","supported":false},"type":"device_interview"}'
    Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x0015bc0034000e09","supported":false},"type":"pairing"}'
    Debug Received Zigbee message from '0x0015bc0034000e09', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 35 with groupID null
    Warning Received message from unsupported device with Zigbee model 'HESZB-120' and manufacturer name 'frient A/S'
    Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
    Debug Received Zigbee message from '0x0015bc0034000e09', type 'read', cluster 'genTime', data '["time"]' from endpoint 35 with groupID null
    Warning Received message from unsupported device with Zigbee model 'HESZB-120' and manufacturer name 'frient A/S'
    Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

So I tried to add a new configuration myself following the tutorial, but ran into few issues I don't know how to fix (first experience in trying to add a device)

I took the configuration example and copied the info from the device in devices/develco.js which ends up like:

Created the file:

    /share/zigbee2mqtt/HESZB-120.js

containing:

    // Frient Fire detector with Siren   
    const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
    const tz = require('zigbee-herdsman-converters/converters/toZigbee');
    const exposes = require('zigbee-herdsman-converters/lib/exposes');
    const reporting = require('zigbee-herdsman-converters/lib/reporting');
    const extend = require('zigbee-herdsman-converters/lib/extend');
    const e = exposes.presets;
    const ea = exposes.access;
    
    const definition = {
            zigbeeModel: ['HESZB-120'],
            model: 'HESZB-120',
            vendor: 'frient A/S',
            description: 'Fire detector with siren',
            fromZigbee: [fz.temperature, fz.battery, fz.ias_smoke_alarm_1_develco, fz.ignore_basic_report, fz.develco_fw, fz.ias_enroll, fz.ias_wd, fz.develco_genbinaryinput],
            toZigbee: [tz.warning, tz.ias_max_duration, tz.warning_simple],
            meta: {battery: {voltageToPercentage: '3V_2500'}},
            configure: async (device, coordinatorEndpoint, logger) => {
                const options = {manufacturerCode: 4117};
                const endpoint = device.getEndpoint(35);
                await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'ssIasZone', 'ssIasWd', 'genBasic', 'genBinaryInput']);
                await reporting.batteryVoltage(endpoint);
                await endpoint.read('genBasic', [0x8000], options);
                await endpoint.read('ssIasZone', ['iasCieAddr', 'zoneState', 'zoneId']);
                await endpoint.read('genBinaryInput', ['reliability', 'statusFlags']);
                await endpoint.read('ssIasWd', ['maxDuration']);
                const endpoint2 = device.getEndpoint(38);
                await reporting.bind(endpoint2, coordinatorEndpoint, ['msTemperatureMeasurement']);
                await reporting.temperature(endpoint2, {min: constants.repInterval.MINUTE, max: constants.repInterval.MINUTES_10, change: 10});
            },
            endpoint: (device) => {
                return {default: 35};
            },
            exposes: [e.temperature(), e.battery(), e.smoke(), e.battery_low(), e.test(),
                exposes.numeric('max_duration', ea.ALL).withUnit('s').withValueMin(0).withValueMax(600).withDescription('Duration of Siren'),
                exposes.binary('alarm', ea.SET, 'START', 'OFF').withDescription('Manual Start of Siren'),
                exposes.enum('reliability', ea.STATE, ['no_fault_detected', 'unreliable_other', 'process_error'])
                    .withDescription('Indicates reason if any fault'),
                exposes.binary('fault', ea.STATE, true, false).withDescription('Indicates whether the device are in fault state')],
    };
    
    module.exports = definition;

In the zigbee2mqtt config I have added:

[... 39 lines cut ...]
        at /app/lib/util/externally-loaded.js:7:16
        at Script.runInContext (vm.js:143:18)
        at Script.runInNewContext (vm.js:148:17)
        at Object.runInNewContext (vm.js:303:38)
        at loadModuleFromText (/app/lib/util/utils.js:156:8)
        at loadModuleFromFile (/app/lib/util/utils.js:162:12)
    (Use `node --trace-warnings ...` to show where the warning was created)
    (node:329) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
    (node:329) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
    [10:57:25] INFO: Handing over control to Zigbee2mqtt Core ...

## Information about the device + link
The device is this one:

[HESZB-120](https://www.zigbee2mqtt.io/devices/HESZB-120.html)

## data/database.db entry of the device

    {"id":56,"type":"EndDevice","ieeeAddr":"0x0015bc0034000e09","nwkAddr":18976,"manufId":4117,"manufName":"frient A/S","powerSource":"Battery","modelId":"HESZB-120","epList":[1,35,38],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[3,5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"35":{"profId":260,"epId":35,"devId":1026,"inClusterList":[0,1,3,15,32,1280,1282],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"HESZB-120","manufacturerName":"frient A/S","powerSource":3,"zclVersion":1,"dateCode":"2021-01-22 10:58"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff0530d6","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}},"38":{"profId":260,"epId":38,"devId":770,"inClusterList":[0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"2021-01-22 10:58","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1637488075222}

in comparison the other smoke detector which got configured (not the same model):

    {"id":55,"type":"EndDevice","ieeeAddr":"0x0015bc003100ab65","nwkAddr":65343,"manufId":4117,"manufName":"frient A/S","powerSource":"Battery","modelId":"SMSZB-120","epList":[1,35,38],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[3,5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"35":{"profId":260,"epId":35,"devId":1026,"inClusterList":[0,1,3,15,32,1280,1282],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"SMSZB-120","manufacturerName":"frient A/S","powerSource":3,"zclVersion":1,"dateCode":"20210717 00:33"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff0530d6","zoneState":1}},"genPowerCfg":{"attributes":{"batteryVoltage":31,"batteryAlarmState":0}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff0530d6","endpointID":1},{"cluster":1280,"type":"endpoint","deviceIeeeAddress":"0x00212effff0530d6","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff0530d6","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}},"38":{"profId":260,"epId":38,"devId":770,"inClusterList":[0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}},"msTemperatureMeasurement":{"attributes":{"measuredVal [... cut ...] t","deviceIeeeAddress":"0x00212effff0530d6","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}}},"dateCode":"20210717 00:33","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1637486287622}

I'm kinda blocked and don't know what to do next... I want to add full fire/smoke coverage of my house but first I need to get the 2 first tests to work as expected. Thanks in advance for your help :)

rindlerblabla, 2021-11-21:
I've got both those heat detectors and smoke detectors added correctly to z2m with the same modelIds. What coordinator are you using?

ffery, 2021-11-28:
I bought a bunch of smoke detectors, they all connected fine, I just ran into problem with the heat detector.
I am using a deconz RaspBeeII.
Maybe there is a way to force the config in the DB?

ffery, 2021-11-28:
Well... @rindlerblabla I found out where my issue was.
Analysing manually the versions in my HA and in git, I noticed that my add-on was version 1.18.1... I checked why it was not updated, and actually I needed to uninstall the old version I had and add again the repository.
Deleted the old device from the DB, adding the device again, and all good now, the heat detector has been added to my devices.
Sorry for that and thank you for your support

closed 2021-11-28 by ffery