Koenkk/zigbee2mqtt #9782: IKEA LED2005R5 does not transition when both power and temperature provided

Otah, 2021-11-21:
## What happened
If I turn on [IKEA LED2005R5](https://www.zigbee2mqtt.io/devices/LED2005R5.html) only using the state=ON, it transitions as expected. If I add color temperature as well, it ignores the transition and "blinks" into full/last brightness right away (as if it has transition=0)

## What did you expect to happen
Transition shouldn't be ever ignored, no matter what combination of parameters is sent.

## How to reproduce it (minimal and precise)
`mosquitto_pub -t "zigbee2mqtt/thatIkeaDevice" -m '{"state": "ON", "transition": 2}'` transitions to ON for 2 seconds, as expected.
`mosquitto_pub -t "zigbee2mqtt/thatIkeaDevice" -m '{"state": "ON", "transition": 2, "color_temp": "370"}'` immediately turns on with no transition.

Both work correctly with other bulbs I have (Philips Hue, Immax Neo). Not sure about other IKEA models, except of this one I only have one other IKEA which doesn't support color temperature.

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: CC2531
Adapter firmware version: `{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}`

Koenkk, 2021-11-21:
This is expected, IKEA bulbs can only transition on 1 property, so if you provide coler_temp it will skip the brightness transition.

Otah, 2021-11-21:
OK thanks, so there is no plan to somehow simulate this on the level of Z2M (e.g. split both actions and send the commands serially – brightness transition first, color temp. setting second)? Do I have to serialize the commands on the application level and that's the ultimate answer?

Koenkk, 2021-11-22:
If you want to have a transition on both, you have to do this from your application level indeed.

closed 2021-11-22 by Koenkk

Otah, 2021-11-22:
In fact, it's not like my use-case would be "transitioning both metrics". My intention is just to turn on the bulb with already specified temperature and transition its brightness from zero to target level. I even kinda don't want the temperature to be visibly transitioned, I want it to be there from the moment of turning on, but I don't think this is how it works / is possible.

(I haven't really tested it recently, but some time ago I wasn't able to specify temperature to a bulb with state OFF, so that I could send the temperature changes only when necessary - like when evening comes - and not with every turning on)