Koenkk/zigbee2mqtt #9783: Tradfri color bulb E12 450 Lumen

CzechSebastian, 2021-11-21:
Hi,
I just bought an Ikea Trader color bulb and it's not support by the HA Addon.
Maybe in the edge version, it will be work ?
Can you give more about this integration ?

Thanks a lot

## Information about the device + link
Ikea Tradfri Bulb E12 450 Lumen
https://www.ikea.com/ca/en/p/tradfri-led-bulb-e12-450-lumen-wireless-dimmable-color-and-white-spectrum-globe-opal-white-20439194/

Koenkk, 2021-11-22:
Added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

CzechSebastian, 2021-11-22:
If I want update the edge version on HA , I need to uninstall , reload add on and install it right ?
Can you send me a message here, when it will be deployed ?

Thanks a lot

Koenkk, 2021-11-22:
> If I want update the edge version on HA , I need to uninstall , reload add on and install it right ?

yes

> Can you send me a message here, when it will be deployed ?

4 hours from now

closed 2021-12-31 by github-actions[bot]