Koenkk/zigbee2mqtt #9785: Temperature Support for MCCGQ14LM (lumi.magnet.acn001)

DanPatten, 2021-11-22:
## Is your feature request related to a problem? Please describe
I have the latest edge version of zigbee2mqtt (HA) and got the sensor working but I am not seeing a temperature reading.

## Describe the solution you'd like
It looks like this is possible, see ZHA's version: https://github.com/zigpy/zha-device-handlers/pull/1151/commits/3d82d0246d44eccc88b2422c2e6ab88743d4ac13

Thank you!

danieledwardgeorgehitchcock, 2021-11-22:
I am not sure where you are looking, but it doesn't appear this device reports back temperature values on any standardised clusters..

Are you sure this is something that the device supports if it is linked to its native hub?

The pull request that you referenced above seems to be part of a larger refactor for the new Aqara E1 / T1 devices so, the new references to temperature are only used if the device (such as the new lumi weather) supports / reports it.

The lumi magnet only seems to report generic (for general device attributes) and IAS Zone (for open / close events)

Koenkk, 2021-11-22:
@dmulcahey can you confirm this device supports temperature measurements? It is not clear to me based on https://github.com/zigpy/zha-device-handlers/pull/1151

DanPatten, 2021-11-22:
I think someone did get it working in deconz (assuming that the value isnt static or something. It's obviously not very accurate but I think that same thing occurred with the older version. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5403#issuecomment-972172659

danieledwardgeorgehitchcock, 2021-11-23:
I see that in the comments someone has indeed mentioned the temperature, and looking at the above comments, this may come from a Xiaomi / Aqara specific cluster.

However, looking at the [Deconz REST Plugin device handler](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/xiaomi/xiaomi_mccgq14lm_openclose_sensor.json), I cannot see any mention of temperature. I also tried decoding the hex attributes that were mentioned in [this comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5403#issuecomment-950730003) and cannot get anything like a temperature reading

Koenkk, 2021-11-23:
@DanPatten could you check if the temperature is in the published payload now? (could take one day to show up)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

blcetux, 2021-12-18:
@Koenkk, I'm using same MCCGQ14LM device. With "koenkk/zigbee2mqtt:latest-dev" docker image, I can see the temperature:
```
zigbee2mqtt    | Zigbee2MQTT:info  2021-12-18 10:10:05: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20211115,"transportrev":2},"type":"zStack12"}'
(...)
zigbee2mqtt    | Zigbee2MQTT:info  2021-12-18 10:10:05: 0x54ef4410002bc16d (0x54ef4410002bc16d): MCCGQ14LM - Xiaomi Aqara E1 door & window contact sensor (EndDevice)
(...)
zigbee2mqtt    | Zigbee2MQTT:info  2021-12-18 10:15:01: MQTT publish: topic 'zigbee2mqtt/0x54ef4410002bc16d/temperature', payload '25'
```

For now, the value is much higher than other sensors. This shows 25C while others on the same room show 19C. I will wait a couple of hours and check it again.

avi9899, 2022-01-07:
> @Koenkk, I'm using same MCCGQ14LM device. With "koenkk/zigbee2mqtt:latest-dev" docker image, I can see the temperature:
>
> ```
> zigbee2mqtt    | Zigbee2MQTT:info  2021-12-18 10:10:05: Coordinator [...]

Did you get it working?
It's 25C all the time!!

Koenkk, 2022-01-19:
@z-x-f do you know if this device has a temperature sensor?

ToXinE, 2022-02-10:
how can this device (a door sensor) could have a temperature sensor ?

closed 2022-03-21 by github-actions[bot]