Koenkk/zigbee2mqtt #9791: First pairing device (supported project) - unsupported

fast0490f, 2021-11-22:
I have a door sensor xiaomi / aqara
If I do the first pairing with the coordinator, i am getting an unsupported device.
I see only device_joined, device_interview, not parring data

If restart Z2M or wait 2 minus - device works fine!
How many times have I tried to remove and add a device the situation is repeated
this does not happen with all devices

Is this a hardware feature of the zigbee network or the software bug of Z2M?

fast0490f, 2021-11-22:
Если сразу после собеседования перезапустить сервис Z2M, то все сразу заработает
В противном случае нужно подождать 1-2 минуты до Приходит сообщение device_anonce.

после перезагрузки получается пройти собеседование сразу, навсегда

lrodorigo, 2021-12-18:
https://github.com/Koenkk/zigbee2mqtt/issues/10214

I think we are in the same boat... when do you restart Z2M service to  complete the interview?

closed 2022-01-25 by github-actions[bot]