Koenkk/zigbee2mqtt #9794: Silvercrest motion sensor (Lidl HG06335) does no longer send valid updates

thackel, 2021-11-22:
## What happened

The device worked before a battery change perfect. Every result got published to the broker when it happened.
After the device got a new battery it no longer provides values for `occupancy`, `battery_low` and `tamper`.
When ever there is movement detected Z2M publishes the data without these values.
Quite odd is also the doubled log output.
I have this behavior now for a couple of dev releases of Z2M after the battery change.

## What did you expect to happen

The values for `occupancy`, `battery_low` and `tamper` should contain values.

## Debug info
```
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02', payload '{"battery":100,"battery_low":null,"last_seen":"2021-11-22T18:37:54.708Z","linkquality":207,"occupancy":null,"tamper":null,"voltage":3000}'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02', payload '{"battery":100,"battery_low":null,"last_seen":"2021-11-22T18:37:54.708Z","linkquality":215,"occupancy":null,"tamper":null,"voltage":3000}'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/voltage', payload '3000'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/voltage', payload '3000'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/battery', payload '100'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/battery', payload '100'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/last_seen', payload '2021-11-22T18:37:54.708Z'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/last_seen', payload '2021-11-22T18:37:54.708Z'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/linkquality', payload '207'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/linkquality', payload '215'
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/occupancy', payload ''
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/occupancy', payload ''
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/battery_low', payload ''
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/battery_low', payload ''
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/tamper', payload ''
Zigbee2MQTT:info  2021-11-22 19:37:54: MQTT publish: topic 'zigbee2mqtt/occupancy02/tamper', payload ''

```

The device is a Lidl "Silvercrest smart motion sensor" HG06335 with firmware 2019.12.12.
Zigbee2MQTT Version 1.22.0-dev commit: 9128838
Coordinator Type ConBee2/RaspBee2
Coordinator Version 0x26660700

thackel, 2021-11-28:
After complete removal of the device and repairing it, the problem persists.
No (`null`) data in for  `occupancy`, `battery_low` and `tamper`.

Status as seen from the frontend:
```
{
    "battery": 100,
    "last_seen": "2021-11-28T10:02:20.830Z",
    "linkquality": 255,
    "voltage": 3000
}
```

Details in frontend:
[image]

woestric, 2021-12-02:
I've got the same problems since a day or three. Occupancy got stuck. I removed battery and did multiple resets. It connects but I get the same null values. I am thinking off coupling through gateway to lidl app and see if it works... Otherwise, the device is probably broken.

woestric, 2021-12-02:
OK. So I coupled the sensor through Silvercrest gateway in Lidl Home. And the sensor is working fine. So it does seem to be a problem with zigbee2mqtt. I also see a couple of ping problems to my ikea lights..

i'll try to investigate further.

woestric, 2021-12-03:
I've got my sensor working again. Not sure if there is a real relation (doubt it) but after upgrading firmware on my conbee II stick and pairing again it is working.... Hopefully it stays this way

Koenkk, 2021-12-03:
can you try to: remove device, restart z2m, pair device

thackel, 2021-12-04:
According to your ideas I tried
1. Remove the device from z2m, restarting the container. This did not help.
2. Remove the device, stopping the container unplugging the Conbee2 to have a "power cycle". Starting and pairing again.

The 2. solution did the trick, so I assume the Conbee2 was in some sort of strange state. It was powered several months, so maybe some sort of overflow happened here.
I think its not related to z2m, so I close the ticket.
Thanks for your help!

closed 2021-12-04 by thackel

gmarchart, 2021-12-06:
i had the same problem with this device, not possible to go get the right values after paring - the only solution was in my case to stop z2m, start deconz, pair the device in deconz, stop deconz, start z2m - then i got the correct values ... this happens 1 time, after a longer time of power off of the sensor (battery was removed because the sensor was not used). not sure what happens when i now remove and reinsert the battery from the sensor. i will test it maybe today.

gmarchart, 2021-12-06:
tested it now, removed the battery and reinsert it - motion sensor is allready paired in z2m but no reaction on motion (always active), tamper manipulation is true, lqi shows value, also other values are available, but did not change and i guess it is not true what i see on this device, did not react on any motion, no value changes. z2m restart (i used edge) did not change the behaviour.

then i removed the device from z2m and powercycled the complete HA (i use conbee 2) - after that all is back again. works now...

thackel, 2021-12-07:
@gmarchart this matches with my scenario, because all started with a power-cycle of the sensor (exchange of battery).

The question is now, does this happen only for Conbee2 users or also others. If others are affected as well, then it would be probably a z2m issue, otherwise blame the Conbee2 ;-)

dxaudio, 2022-01-18:
Same issue for me. Tried different things without success. Only works again if force to remove sensor from z2m and repair. Adapter Slaesh's CC2652RB stick. Latest z2m.

AlexFindeis, 2022-06-25:
Using a CC2531 stick the same problem occurs. My previous adapter died and the silvercrest motion sensors had to be repaired. Only got battery and signal strength, but no tamper and occupancy signal. The described procedure worked for me, too: Remove devices in z2m, power off RPi + stick, reboot, then pair sensors. All back to normal now.