Koenkk/zigbee2mqtt #9796: Sercomm SZ-DWS08N-CZ3 magnetic contact sensor shows up as unsupported device despite being on the supported list

nathan425, 2021-11-22:
## What happened
Sercomm SZ-DWS08N-CZ3 Magnetic door & window contact sensor shows up as unspported device despite being on the supported list.

## What did you expect to happen
The sensor should already be supported as well as a supplemental fix? 
https://www.zigbee2mqtt.io/devices/SZ-DWS08.html & https://github.com/Koenkk/zigbee2mqtt/issues/9322

## How to reproduce it (minimal and precise)

Pairing SZ-DWS08N-CZ3 shows up as an Unsupported device on both standard and dev branch. 

```
Warning Received message from unsupported device with Zigbee model 'SZ-DWS08N-CZ3' and manufacturer name 'Sercomm Corp.'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

## Debug info
Zigbee2MQTT version: 1.22.0-3 & Edge (1.22.0-dev commit:9128838)
Adapter hardware: CC2531
Adapter firmware version: 20180815

Koenkk, 2021-11-22:
Yours has a slightly different model number, I've added it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

nathan425, 2021-11-23:
This works, thanks! Sorry didn’t know since it started with the same model # string. 

Can I transfer this to my stable branch or do I need to continue using edge?

Koenkk, 2021-11-23:
You can switch back to stable with the next release since the device will be included (1 december)

closed 2021-11-23 by Koenkk