Koenkk/zigbee2mqtt #9797: Socket TS011F manufacturer _TZ3000_w0qqde0g not reporting anything except linkquality and on/off

Toma82, 2021-11-22:
I have 3 plug TS011F manufacturer _TZ3000_w0qqde0g https://pl.aliexpress.com/item/1005003501027768.html?spm=a2g0o.productlist.0.0.7ce0408brLHZAI&algo_pvid=8526ef5d-0967-4c0e-af74-7f2d5b7f9470&algo_exp_id=8526ef5d-0967-4c0e-af74-7f2d5b7f9470-13&pdp_ext_f=%7B%22sku_id%22%3A%2212000026075390421%22%7D
It is not reporting power, current, voltage,energy ,only report the state when changed from physical button and  reports is linkquality.

[image]
[image]
[image]

I also tried external configuration:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [
        {
            modelID: 'TS011F',
            manufacturerName: '_TZ3000_w0qqde0g'
        },
    ],
    model: 'Test',
    vendor: 'TuYa',
    description: 'Zigbee Socket',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 1, min: 10, max: 600});
        await reporting.rmsCurrent(endpoint, {change: 50, min: 10, max: 600});
        await reporting.activePower(endpoint, {change: 1, min: 10, max: 600});
        await reporting.currentSummDelivered(endpoint, {change: 1, min:10, max: 600});
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage')],
};

module.exports = definition;

when I type from the console, the report appears.

[image]
[image]
[image]

{"id":10,"type":"Router","ieeeAddr":"0xa4c138e1620dedbf","nwkAddr":35154,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_w0qqde0g","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":237,"activePower":73,"rmsCurrent":217}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff070ed0","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff070ed0","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff070ed0","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":1,"maxRepIntval":600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":1,"maxRepIntval":600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":1,"maxRepIntval":600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":1,"maxRepIntval":600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1637614629330}

All 3 plug behave the same

## Debug info
Zigbee2MQTT version: 1.22.0-3 
Adapter hardware: Conbee II
Adapter firmware version: _0x26720700

Debug
Info MQTT publish: topic 'zigbee2mqtt/test', payload '{"current":0.22,"energy":null,"linkquality":208,"power":null,"power_outage_memory":"restore","state":"ON","voltage":238}'
Info MQTT publish: topic 'zigbee2mqtt/test', payload '{"current":0.22,"energy":null,"linkquality":188,"power":null,"power_outage_memory":"restore","state":"ON","voltage":238}'
Info Succesfully changed options
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"qcu6x-1"}'
Debug Received MQTT message on 'zigbee2mqtt/test/set' with data '{"state":"OFF"}'
[... 2 lines cut ...]
Debug Received Zigbee message from 'test', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/test', payload '{"current":0.22,"energy":null,"linkquality":208,"power":null,"power_outage_memory":"restore","state":"OFF","voltage":238}'
Debug Received Zigbee message from 'Gniazdko', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":229}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/Gniazdko', payload '{"current":0,"energy":27.58,"linkquality":255,"power":0,"power_outage_memory":"restore","state":"ON","voltage":229}'
Debug Received MQTT message on 'zigbee2mqtt/test/set' with data '{"power_outage_memory":"on"}'
Debug Publishing 'set' 'power_outage_memory' to 'test'
Info MQTT publish: topic 'zigbee2mqtt/test', payload '{"current":0.22,"energy":null,"linkquality":212,"power":null,"power_outage_memory":"on","state":"OFF","voltage":238}'
Debug Received Zigbee message from 'test', type 'attributeReport', cluster 'genOnOff', data '{"moesStartUpOnOff":1}' from endpoint 1 with groupID null
Debug Received Zigbee message from 'test', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53264":1}' from endpoint 1 with groupID null
Debug No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53264":1}'
Debug Received MQTT message on 'zigbee2mqtt/test/set' with data '{"power_outage_memory":"restore"}'
Debug Publishing 'set' 'power_outage_memory' to 'test'
Info MQTT publish: topic 'zigbee2mqtt/test', payload '{"current":0.22,"energy":null,"linkquality":192,"power":null,"power_outage_memory":"restore","state":"OFF","voltage":238}'
Debug Received Zigbee message from 'test', type 'attributeReport', cluster 'genOnOff', data '{"moesStartUpOnOff":2}' from endpoint 1 with groupID null
Debug Received Zigbee message from 'test', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53264":2}' from endpoint 1 with groupID null
Debug No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53264":2}'
Debug Received MQTT message on 'zigbee2mqtt/test/set' with data '{"state":"ON"}'
Debug Publishing 'set' 'state' to 'test'
Info MQTT publish: topic 'zigbee2mqtt/test', payload '{"current":0.22,"energy":null,"linkquality":180,"power":null,"power_outage_memory":"restore","state":"ON","voltage":238}'
Debug Received Zigbee message from 'test', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/test', payload '{"current":0.22,"energy":null,"linkquality":204,"power":null,"power_outage_memory":"restore","state":"ON","voltage":238}'
Debug Received Zigbee message from 'Gniazdko', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,2758]}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/Gniazdko', payload '{"current":0,"energy":27.58,"linkquality":255,"power":0,"power_outage_memory":"restore","state":"ON","voltage":229}'
Debug Received Zigbee message from 'Gniazdko', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":229}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/Gniazdko', payload '{"current":0,"energy":27.58,"linkquality":255,"power":0,"power_outage_memory":"restore","state":"ON","voltage":229}'

Koenkk, 2021-11-22:
Could you share the data/database.db entry of this device?

Toma82, 2021-11-22:
{"id":10,"type":"Router","ieeeAddr":"0xa4c138e1620dedbf","nwkAddr":35154,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_w0qqde0g","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":237,"activePower":73,"rmsCurrent":217}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff070ed0","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff070ed0","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff070ed0","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":1,"maxRepIntval":600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":1,"maxRepIntval":600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":1,"maxRepIntval":600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":1,"maxRepIntval":600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1637614629330}

Toma82, 2021-11-22:
It is with a different plug
[image]
{"id":11,"type":"Router","ieeeAddr":"0xa4c1385cb5918afd","nwkAddr":50131,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_w0qqde0g","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff070ed0","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff070ed0","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff070ed0","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1637615931204,"useImplicitCheckin":true}

Koenkk, 2021-11-22:
Could you check if this is fixed now for the plug with ieeeAddr `0xa4c138e1620dedbf`?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

kaweksl, 2021-11-23:
Hi, i have 3 plugs that identify itself as "_TZ3000_w0qqde0g", One of them have similar look but different colours, this one report values without issues i can change reporting timings as well. For another two i can't get reporting to work, 
[image]

kaweksl, 2021-11-23:
Hi, i have 3 plugs that identify itself as "_TZ3000_w0qqde0g", One of them have similar look but different colours, this one report values without issues i can change reporting timings as well. For another two i can't get reporting to work, 
[image]

Edit: no change in latest-dev

Edit: my database.db for those plugs, note that i was playing with reporting timings
```
{"id":11,"type":"Router","ieeeAddr":"0xa4c1381caa91d2d5","nwkAddr":19382,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_w0qqde0g","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791644","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791644","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791644","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":"10","repChange":[1,1]},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":"10","repChange":"1"}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1637631848352,"useImplicitCheckin":true}

{"id":4,"type":"Router","ieeeAddr":"0xa4c138295068d73c","nwkAddr":2355,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_w0qqde0g","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791644","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791644","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791644","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1637630422047,"useImplicitCheckin":true}

{"id":6,"type":"Router","ieeeAddr":"0xa4c1384281b90290","nwkAddr":14790,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53264":2,"switchType":0}},"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_w0qqde0g","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":238,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,46]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791644","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791644","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791644","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":"3600","repChange":[1,1]}],"meta":{}},"242":{"profId":4144 [... cut ...]
On plugs with reporting issue i can get values via dev console read 
Zigbee2MQTT version:  1.22.0-dev commit: a9e3fb3 
Adapter: Egony Stick V4
Coordinator revision 20210708

Did commit https://github.com/Koenkk/zigbee-herdsman-converters/commit/a8deb79fa78e25dfb2b51bbccf54ad784634bca2 changed something for _TZ3000_w0qqde0g ? I can't see that

EDIT: 
@Toma82 does your plug named "Test2" is not reporting even after changing max rep interval to something faster (like 60sec) ?
Maybe plugs with appVersion 68 have issues and we can filter by that

Toma82, 2021-11-23:
Hello,
Surprisingly, socket Test 2 started to report correctly, while socket Test still does not.

[image]
[image]

[image]
[image]

Koenkk, 2021-11-23:
Could you both check if it is fixed now?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

kaweksl, 2021-11-23:
Seams not fixed.
Zigbee2MQTT version 1.22.0-dev commit: 167c056 

Plugs are recognized as Model: TS011F_plug_1
I have checked and above patch to tuya,js is in place
I have removed and paired plug again, no change still as TS011F_plug_1
does applicationVersion checks for appVersion ?

kaweksl, 2021-11-23:
Seams not fixed.
Zigbee2MQTT version 1.22.0-dev commit: 167c056 

Plugs are recognized as Model: TS011F_plug_1
I have checked and above patch to tuya,js is in place
I have removed and paired plug again, no change still as TS011F_plug_1
does applicationVersion checks for appVersion ?

EDIT: Seams like there is issue with converters priority, in tuya,js i have switched places between converter with applicationVersion and without and it looks ok.

[image]

avanlievenoogen, 2021-11-23:
I am not sure if this is the same issue but it looks like it.
I just bought four of these. (TS011F - _TZ3000_ew3ldmgx)   Three configure fine and I get updates on power usage. The fourth one however does not. I can only turn it on and off. I run Z2M with a conbee ii. When I connect the failing plug to another setup with a CC2652RB it configures fine. But I don't want it on that setup :)  I have tried deleting it and reconfiguring it but no luck.
I pasted the logging below. 
Any tips how to resolve this?  Should I install the development branch as mentioned above?

thanks in advance.

Logging:
Publish 'set' 'state' to '0x2c1165fffec75baa' failed: 'Error: Command 0x2c1165fffec75baa/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
debug Error: Command 0x2c1165fffec75baa/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received) at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:645:23)
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x2c1165fffec75baa' failed: 'Error: Command 0x2c1165fffec75baa/1 genOnOff.off({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"0x2c1165fffec75baa"},"type":"zigbee_publish_error"}'
error Publish 'set' 'state' to '0x2c1165fffec75baa' failed: 'Error: Command 0x2c1165fffec75baa/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
debug Error: Command 0x2c1165fffec75baa/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received) at DeconzAdapter.sendZclFrameToEndpoint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:645:23) at runMicrotasks (<anonymous>) at runNextTicks (internal/process/task_queues.js:62:5) at processTimers (internal/timers.js:494:9)
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x2c1165fffec75baa' failed: 'Error: Command 0x2c1165fffec75baa/1 genOnOff.off({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"0x2c1165fffec75baa"},"type":"zigbee_publish_error"}'

kaweksl, 2021-11-23:
@avanlievenoogen post data/database.db for all four plugs when paired with setup that have issues

Toma82, 2021-11-24:
There is no improvement at the moment, no reporting via these 2 sockets

Koenkk, 2021-11-24:
@kaweksl 

> EDIT: Seams like there is issue with converters priority, in tuya,js i have switched places between converter with applicationVersion and without and it looks ok.

What is the order now, first `TS011F_plug_1` after that `TS011F_plug_3` or `TS011F_plug_3` after that `TS011F_plug_4`?

kaweksl, 2021-11-24:
First TS011F_plug_1 , then TS011F_plug_3. That works for me

avanlievenoogen, 2021-11-24:
@kaweksl 
> @avanlievenoogen post lines from data/database.db for all four plugs when paired with setup that have issues. Entry when paired with working setup might be useful too

Three plugs where energy mon. is working:
{"id":5,"type":"Router","ieeeAddr":"0x2c1165fffec75bce","nwkAddr":26315,"manufId":4098,"manufName":"_TZ3000_ew3ldmgx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_ew3ldmgx","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"rmsCurrent":101,"activePower":0,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":236}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,199]}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a322","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a322","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a322","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1637737286876}
{"id":28,"type":"Router","ieeeAddr":"0x2c1165fffea2a42e","nwkAddr":43495,"manufId":4098,"manufName":"_TZ3000_ew3ldmgx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_ew3ldmgx","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":237,"rmsCurrent":340,"activePower":86,"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,122]}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a322","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a322","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a322","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]},{"cluster":2820,"attrId":1288,"minRepIntval":"60","maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1285,"minRepIntval":"60","maxRepIntval":3600,"repChange":5}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1637737621639}
{"id":26,"type":"Router","ieeeAddr":"0x2c1165fffec3f7f2","nwkAddr":45869,"manufId":4098,"manufName":"_TZ3000_ew3ldmgx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_ew3ldmgx","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"rmsCurrent":0,"activePower":0,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":235}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a322","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a322","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff06a322","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1637737560318}

failing on same setup as above:
[... cut ...] repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1637741719494}

Toma82, 2021-11-24:
I confirm how I changed places first TS011F_plug_1 after that TS011F_plug_3, everything works fine and the socket is already reporting

Thank you very much @Koenkk and kaweksl for your help.
You guys are gorgeous
[image]
[image]

Koenkk, 2021-11-24:
Thanks, fixed this now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-11-24 by Koenkk

avanlievenoogen, 2021-11-25:
great! switching to development branch also fixed it for me. Thank you very much!

iisworks, 2021-12-04:
Does the power_outage_memory setting work for you on these new plugs? Mine always seem to be turned off when power comes back on, regardless of the setting or previous state?

kaweksl, 2021-12-05:
On TS011F_plug_3 it doesn't, after plugging into socket plug turns on and off, not matter how i set ower_outage_memory setting .
On TS011F_plug_1 this setting work as expected.

Webserve, 2021-12-08:
Hi @Koenkk 

Zigbee2MQTT versie: 1.22.1-dev commit: d0195d0
Coordinator type: zStack3x0
Coordinator revisie: 20200925
Versie frontend: 0.6.59

Just recieved four plugs and they are exposed as _TZ3000_ew3ldmgx (TS011F_plug_1) thats good right ?
But the values are Null.

[image]

But as in the opening post i can read the values.

[image]

This is my db 

```
{"id":60,"type":"Router","ieeeAddr":"0x540f57fffe897074","nwkAddr":2583,"manufId":4098,"manufName":"_TZ3000_ew3ldmgx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_ew3ldmgx","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsCurrent":135}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf135","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf135","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf135","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"reporting":1,"configured":680236614},"lastSeen":1638997455252,"useImplicitCheckin":true}
```

What am i doing wrong?

Koenkk, 2021-12-09:
@Webserve yours has appVersion 69 so it only support measurements by polling, fixed this now for your device.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Webserve, 2021-12-12:
Thanks fixed it :)

Toma82, 2021-12-12:
For me, _TZ3000_w0qqde0g appVersion  64 also works correctly only with polling

Koenkk, 2021-12-14:
@Toma82 fixed

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gabric098, 2021-12-14:
I could test my tuya TS011F sockets (_TZ3000_ew3ldmgx). with firmware version 69.
[image]

They work correctly:
[image]

Thanks for fixing it 👍

Toma82, 2021-12-22:
Hello, a new version of these sockets has appeared "{modelID: 'TS011F', manufacturerName: '_TZ3000_gjnozsaz', applicationVersion: 68},". Could You @Koenkk  add them to TS011F_plug_3  I made my own entry, the sockets work and are recognized. I'm sorry, I did not check if they would work in the TS011F_plug version

{
        fingerprint: [
            {modelID: 'TS011F', manufacturerName: '_TZ3000_dpo1ysak', applicationVersion: 69},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_cphmq0q7', applicationVersion: 69},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_5f43h46b', applicationVersion: 69},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_dpo1ysak', applicationVersion: 68},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_cphmq0q7', applicationVersion: 68},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_5f43h46b', applicationVersion: 68},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_ps3dmato', applicationVersion: 69},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_w0qqde0g', applicationVersion: 68},
	    {modelID: 'TS011F', manufacturerName: '_TZ3000_w0qqde0g', applicationVersion: 64},
	    {modelID: 'TS011F', manufacturerName: '_TZ3000_gjnozsaz', applicationVersion: 68},

        ],
        model: 'TS011F_plug_3',
        description: 'Smart plug (with power monitoring by polling)',
        vendor: 'TuYa',
        whiteLabel: [{vendor: 'VIKEFON', model: 'TS011F'}],
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        options: [exposes.options.measurement_poll_interval()],
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage')],
        onEvent: (type, data, device, options) => {
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(globalStore.getValue(device, 'interval'));
                globalStore.clearValue(device, 'interval');
            } else if (!globalStore.hasValue(device, 'interval')) {
                const seconds = options && options.measurement_poll_interval ? options.measurement_poll_interval : 60;
                const interval = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                        await endpoint.read('seMetering', ['currentSummDelivered']);
                    } catch (error) {/* Do nothing*/}
                }, seconds*1000);
                globalStore.putValue(device, 'interval', interval);
            }

Koenkk, 2021-12-22:
@Toma82 should already work in the dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Toma82, 2021-12-22:
> @Toma82 should already work in the dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Yes ,This is  works thanks

Only this option after a power outage, the socket always goes off despite the fact that power outage memory is set to on or restore

majkers, 2022-01-04:
The same for me. I opened new issue because this one is closed https://github.com/Koenkk/zigbee2mqtt/issues/10487

iisworks, 2022-01-12:
FYI: Just got a FW update for _TZ3000_w0qqde0g in the Tuya App to 1.0.10 (AppVersion 74), which seems to fix the On/Off state memory, and the device does seem to report updates again instead of requiring polling!

[image]

Koenkk, 2022-01-12:
@iisworks good to know! checked the z2m code and the plug should be identified as `TS011F_plug_1` which relies on the reporting instead of polling.

majkers, 2022-01-12:
But is it possible to update firmware without tuya gateway?

iisworks, 2022-01-12:
Don' t think so. Bought one for €15 on Ali for situations like these. :)

Toma82, 2022-01-15:
I confirm that the On / Off state memory is working properly in the new FW 1.0.10. My socket is still detected as TS011F_plug_3. I just did not remove it, only reconfigured it in 2ZM

majkers, 2022-01-27:
After update I removed it and re add it but still recognized as TuYa TS011F_plug_3

iisworks, 2022-01-30:
FYI: the orange compact plugs now come in a new version: _TZ3000_gjnozsaz.
They also support the new FW update 1.0.10 (AppVersion 74).

elenril2, 2022-04-15:
Hello, I've just bought two of these and had a problem with power, energy and current reports.
```{"id":13,"type":"Router","ieeeAddr":"0xa4c138e742559d19","nwkAddr":52,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":241}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0c7db","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0c7db","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0c7db","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1650051087586,"defaultSendRequestWhen":"immediate"}
{"id":14,"type":"Router","ieeeAddr":"0xa4c1386d24a24808","nwkAddr":64939,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":241,"rmsCurrent":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0c7db","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0c7db","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0c7db","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1650052033429,"defaultSendRequestWhen":"immediate"}```

elenril2, 2022-04-15:
Strangely second one starts reporting, but the first one (id: 13) has these values marked as N/A.

knipknap, 2023-04-28:
> I confirm how I changed places first TS011F_plug_1 after that TS011F_plug_3, everything works fine and the socket is already reporting
> 
> Thank you very much @Koenkk and @kaweksl for your help.

It seems like nobody noticed here that the reported fix actually proves that there is still an error: 230*0.2=46, not 22. So it's off by a factor of 2.

stockno, 2023-12-06:
Hello
i have this drawer https://www.aliexpress.com/item/1005004857105081.html?spm=a2g0o.order_list.order_list_main.23.372c1802ioY7nR
model and manufacturer: TS011F _TZ3000_w0qqde0g

zigbee2mqtt is not consumption measurement, just on/off and signal strength.

DB: {"id":24,"type":"Router","ieeeAddr":"0xa4c1383fd5cfe7ba","nwkAddr":5088,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":" Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266 ,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{ "modelId":"TS011F","manufacturerName":"_TZ3000_w0qqde0g","powerSource":1,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode": ""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002281111d ","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode": null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33], "clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode ":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1701851233817,"defaultSendRequestWhen":"immediate"}
[image]
[image]
[image]
[image]
[image]

Hello
i have this drawer https://www.aliexpress.com/item/1005004857105081.html?spm=a2g0o.order_list.order_list_main.23.372c1802ioY7nR
model and manufacturer: TS011F _TZ3000_w0qqde0g

zigbee2mqtt is not consumption measurement, just on/off and signal strength.

DB: {"id":24,"type":"Router","ieeeAddr":"0xa4c1383fd5cfe7ba","nwkAddr":5088,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_w0qqde0g","powerSource":1,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002281111d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1701851233817,"defaultSendRequestWhen":"immediate"}

[image]
[image]
[image]
[image]
[image]

Can you advise me, I don't know what to do with it at all. Thank you

TonKro, 2024-01-05:
I moved from Deconz to Z2M. My switch did work earlier: 

_TZ3000_gjnozsaz
TS011F

Now it is only passing the voltage / power but not able to turn on / off and does have the state unkown in Home Assisitant.

What am I doing wrong? Or can I upgrade it?