Koenkk/zigbee2mqtt #9800: Multiple mqtts publishes?

robertalexa, 2021-11-22:
Hi @Koenkk 

I was wondering if you could shed some light on this, something i have noticed after the latest update. For one of my automations i basically used a toggle when the action = single. Now i have noticed that this stopped working, or even, it doubles up in fact.

At a quick glance in the z2m logs i have found this:
```
Info MQTT publish: topic 'zigbee2mqtt/Steph's Nightstand Light Switch', payload '{"action":"single","battery":92,"last_seen":1637619612680,"linkquality":99}'
Info MQTT publish: topic 'zigbee2mqtt/Steph's Nightstand Light Switch/action', payload 'single'
```

To your knowledge, did anything change recently that would duplicate these MQTT messages like this? In my case all of this goes through HA

Thanks

Koenkk, 2021-11-23:
Does the problem go away when setting `last_seen` to `disable` in `configuration.yaml`?

sjorge, 2021-11-23:
Is this not because output is set to json + attrubute ?

closed 2021-12-31 by github-actions[bot]