Koenkk/zigbee2mqtt #9806: Show database.db entry in Frontend - "Nice to have" feature

chrisrock1984, 2021-11-23:
## Is your feature request related to a problem? Please describe
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

To request a new device more easily, it would be nice to show the database entry in the frontend. 

## Describe the solution you'd like
A clear and concise description of what you want to happen.

Add a new tab which shows the database entry of that device

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.
-

## Additional context
Add any other context or screenshots about the feature request here.

An "upload new request" to github would be nice to have, which automatically adds the database entry within the template for adding a new device

danieledwardgeorgehitchcock, 2021-11-23:
I raised this a while back.

Please see answers here nurikk/zigbee2mqtt-frontend#822

Koenkk, 2021-11-23:
Let's continue in https://github.com/nurikk/zigbee2mqtt-frontend/issues/822

closed 2021-11-23 by Koenkk