Koenkk/zigbee2mqtt #9809: DDS238-2 Zigbee (TS0601_din) energy counter not reported (always null)

edillmann, 2021-11-23:
## What happened
Configured device DDS238-2 Zigbee (TS0601_din) worked well, but energy meter values remain null

Exemple mqtt log:

Zigbee2MQTT:info  2021-11-23 11:19:37: MQTT publish: topic 'zigbee2mqtt/switch_boiler', payload '{"current":0.057,"energy":null,"linkquality":42,"power":6.6,"state":"ON","voltage":236}'

## What did you expect to happen

I would expect that the energy value reflect the one displayed on the device (in kWh)

## How to reproduce it (minimal and precise)

Connect & pair the device, it appears in home assistant, but energy value is always "null"

## Debug info
Zigbee2MQTT version:1.22.0
Adapter hardware: CC2531
Adapter firmware version: 20190425/ zStack30x

danieledwardgeorgehitchcock, 2021-11-23:
Hi,

I created the converter so, should be able to assist you in getting it sorted.

The energy consumed is reported less frequently than the other metrics as it is a cumulative value so, it may well be that you need to let it gather the data for a while.

I will also check on my device to make sure that it is actually working as it should.

edillmann, 2021-11-23:
I will let it live a whole day and see if it get updated,

Thanks for your feedback

edillmann, 2021-11-23:
It took 1 hour, which make sense kW per hour

Thanks,

closed 2021-11-23 by edillmann