Koenkk/zigbee2mqtt #9810: New Device: AduroSmart Smart Plug with power measurements

Inrego, 2021-11-23:
## Information about the device + link
Smart plug with power measurements. AduroSmart website sucks, and doesn't list all their products. So here's a link from a reseller:
https://www.turascandinavia.com/en/products/smart-home/sensor-wall-sockets/wall-sockets/615527

Here's the JS file I created that works for me. I didn't really know what I was doing, I just more or less copy-pasted everything from the Heiman metering plug. And it works for me. I tested on/off, and I get values in Power, Current and Voltage. Externally it looks identical to the other AduroSmart plugs (I own all 3 variants), so you can use the same photo as those.
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['AD-SmartPlug3001'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '81848', // Vendor model number, look on the device for a model number
    vendor: 'AduroSmart', // Vendor of the device (only used for documentation and startup logging)
    description: 'ERIA smart plug (power measuring)', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.electrical_measurement], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch(), e.power(), e.current(), e.voltage()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            await reporting.onOff(endpoint);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.rmsVoltage(endpoint);
            await reporting.rmsCurrent(endpoint);
            await reporting.activePower(endpoint);
        }
};

module.exports = definition;
```

Inrego, 2021-11-23:
Here's the state json:

```
{
    "current": 0.05,
    "linkquality": 255,
    "power": 3.9,
    "state": "ON",
    "voltage": 234.82
}
```

arjanvanleent, 2021-11-23:
You know away to add it in my system? Just received this power switch

Inrego, 2021-11-23:
> You know away to add it in my system? Just received this power switch

Just create a file named something.js with the code i posted. Put it next to configuration.yaml and add it to the config as described in the article about adding support for new devices

Koenkk, 2021-11-23:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-11-23 by Koenkk