Koenkk/zigbee2mqtt #9811: Aeotec (Smartthings) Multipurpose Sensor IM6001-MPP03

Papi77, 2021-11-23:
## Information about the device + link
Add support for latest version of Multipurpose Sensor from Aeotec. There is currently already support for the older version of a similar device from [Smartthings (2018)](https://www.zigbee2mqtt.io/devices/IM6001-MPP01.html). This request if for the newer version: https://aeotec.freshdesk.com/support/solutions/articles/6000242293

When pairing I'm getting the device recognized but number of entities are missing:

[image]

## data/database.db entry of the device
..

danieledwardgeorgehitchcock, 2021-11-23:
Please add the data/database.db for this device

Papi77, 2021-11-24:
Sorry, but I don't know what data/database.db is :(

danieledwardgeorgehitchcock, 2021-11-24:
How are you using Z2M? Through HA addon? Through Docker image? other?

closed 2022-01-01 by github-actions[bot]