Koenkk/zigbee2mqtt #9814: Immax NEO LITE Smart E27 bulb identified as wrong GU10 bulb

domsvoboda, 2021-11-23:
## What happened
I bought pack of Immax RGB E27 bulbs, however two of them shows as GU10 bulbs from Tuya. Zigbee model is TS0505A and the Zigbee manufacturer is _TZ3000_hlijwsai. Also the red light on these bulbs doesn't work when i choose it from HA. I guess this needs the red color fix.

## What did you expect to happen
I have another bulbs from Immax and they are showing correctly as Immax and the red color work.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: CC2531
Adapter firmware version: zStack12, 20190608

pborky, 2021-12-24:
Same issue here.

standahabich, 2022-01-14:
Immax NEO 07115L

reporting:
`Read result of 'genBasic': {"manufacturerName":"_TZ3000_hlijwsai","modelId":"TS0505A"}`

but identify as "TuYa TS0505A" and full red color not worked.

closed 2022-02-22 by github-actions[bot]