Koenkk/zigbee2mqtt #9815: Philips Smart plug with bluetooth UK LOM009 unsupported

bmorris591, 2021-11-23:
Looking at https://github.com/Koenkk/zigbee2mqtt/issues/4244 and https://github.com/Koenkk/zigbee-herdsman-converters/pull/1527 do I just need to copy paste the device again with the `LOM009` model number?

## Information about the device + link
Philips UK model Smart Plug with Bluetooth support
https://www.philips-hue.com/en-gb/p/hue-smart-plug/8718699689308#specifications

## data/database.db entry of the device
```json
{
  "id": 2,
  "type": "Router",
  "ieeeAddr": "0x001788010b572083",
  "nwkAddr": 43561,
  "manufId": 4107,
  "manufName": "Signify Netherlands B.V.",
  "powerSource": "Mains (single phase)",
  "modelId": "LOM009",
  "epList": [11, 242],
  "endpoints": {
    "11": {
      "profId": 260,
      "epId": 11,
      "devId": 266,
      "inClusterList": [0, 3, 4, 5, 6, 8, 4096, 64514],
      "outClusterList": [25],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "LOM009",
            "manufacturerName": "Signify Netherlands B.V.",
            "powerSource": 1,
            "zclVersion": 2,
            "appVersion": 2,
            "stackVersion": 1,
            "hwVersion": 1,
            "dateCode": "20201117",
            "swBuildId": "1.76.11"
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [33],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 2,
  "stackVersion": 1,
  "hwVersion": 1,
  "dateCode": "20201117",
  "swBuildId": "1.76.11",
  "zclVersion": 2,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1637685323581
}

```

Koenkk, 2021-11-23:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-11-23 by Koenkk

bmorris591, 2021-11-23:
Awesome, thanks! 

I assume matching it done on `zigbeeModel` and to the manufacturer will be overridden to Philips by the mapping?

Koenkk, 2021-11-23:
manufName will stay at is, the frontend just shows the `vendor` from the definition in zigbee-herdsman-converters.

bmorris591, 2021-11-23:
Confirmed working. Thanks again.