Koenkk/zigbee2mqtt #9820: TuYa TS0601_thermostat_1 can not set schedule

biejay, 2021-11-24:
## What happened
I got the TuYa TS0601_thermostat_1 working, but in the manual it says that for setting the schedule I have to publish to "zigbee2mqtt/FRIENDLY_NAME/set". But i can not set any schedule data. 
Inserting my thermostat name here works fine for the current heatpoint, so thats not the matter.

But The payload in the manual `"{"undefined": {"monday_schedule": VALUE, "tuesday_schedule": VALUE, "wednesday_schedule": VALUE, "thursday_schedule": VALUE, "friday_schedule": VALUE, "saturday_schedule": VALUE, "sunday_schedule": VALUE}}"` does not work. 

I replaced VALUE by `" 6h:0m 20°C,  12h:0m 15°C,  18h:0m 22°C,  22h:0m 15°C "` but Zigbee2mqtt tells me **`"No converter found for undefined" `**
## What did you expect to happen
Setting the schedule to my preferred times.

## Debug info
Zigbee2MQTT version: 1.22.0-3
Adapter hardware: slae.sh_cc2652rb_stick

Koterak, 2021-11-24:
Try instead of "undefined": 
- "workdays"
- "holidays"

biejay, 2021-11-24:
Also not working.

biejay, 2021-12-02:
Still no luck after update to 1.22.1-1

biejay, 2021-12-24:
Push

closed 2022-02-01 by github-actions[bot]

maciekn, 2022-04-06:
I am struggling with this issue as well. @biejay were you able to set the schedule?

chodar, 2022-09-26:
any update here? I'm facing the same issue.