Koenkk/zigbee2mqtt #9833: TuYa TS0601 valve Heat mode opens valve 100%

mwasowski, 2021-11-25:
## What happened

When I change mode on the TRV to `Heat` the valve fully opens and temperature setting is not respected. To control temperature I need to set it to `Auto`. That is actually kind of wrong and not how Home Assistant handles thermostats.

## What did you expect to happen

After setting to `Heat` I expected to be able to control temperature. `Auto` mode means for me that I am using schedules on the TRV etc.

## How to reproduce it (minimal and precise)

Set TRV to `Auto`, change temperature - it works.
Set TRV to `Heat`, set temperature - see it is not respected and valve opens to 100%.

## Debug info
Zigbee2MQTT version: 1.22.0
Adapter hardware: CC2652P (SmartlightMe)
Adapter firmware version: zStack3x0 20210708

groenmarsmannetje, 2021-11-25:
Duplicate of #5578 which is stale

mwasowski, 2021-11-25:
So should I open the one referenced one or start the conversation here?

groenmarsmannetje, 2021-11-25:
I just added it as a reference, because I remember that there where quite a lot of discussions about this multiple times earlier.
Probably there are also people now who depend on current implementation. Which need to be taken in consideration when behavior is changed.

mwasowski, 2021-12-14:
Well, ok. Any comment/update?

mwasowski, 2022-01-14:
Bumping the bot.

Inferi0r, 2022-01-27:
> Bumping the bot.

system_mode and force are duplicates values for this valve. 
force:normal = system_mode=auto

But it does not matter in HA, just use auto and it has nothing to do with HA climate. Any use case with HA climate can work with the 'auto' mode.

closed 2022-03-06 by github-actions[bot]

mwasowski, 2022-03-06:
Crap, missed the email. sorry for the late reply.

The issue for me is that I can't use it with scheduler component because it relies on setting mode to heat and then temperature. So I need to add some custom logic for that that is not working super well :(