Koenkk/zigbee2mqtt #9843: HSB color_temp publish over or under values as documented values

Wikibear, 2021-11-25:
Vers: 1.21.2
Müller Light RGBW
Command you see at debug.
[image]
I get values under 153 or over 556 as documented:
https://www.zigbee2mqtt.io/devices/404000_404005_404012.html#light
[image]
Via Frontend:
[image]
[image]

Koenkk, 2021-11-25:
What is the temp range given by the bulb? https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

sjorge, 2021-11-25:
@Koenkk I think this is because the color_mode is set to hs, so we're calculating the color_temp based on the hue/saturation values.

Not sure what the correct thing to do here is, do we clamp them to the support colorTemperature range when color_sync=true or do we stick to what we have now which is probably a close approximation of what the actual color is?

I'd be inclined to clamp them to the actual support colorTemperature range, because we do pass this information along using expose.

Koenkk, 2021-11-25:
> I'd be inclined to clamp them to the actual support colorTemperature range

Agree

sjorge, 2021-11-25:
> > I'd be inclined to clamp them to the actual support colorTemperature range
> 
> Agree

I'm pretty busy at the moment, but if you don't get to it before weekend after next, I should be able to look at this.

sjorge, 2021-11-26:
I had a quick look during my break, looks like we do not have access to 'entity' in https://github.com/Koenkk/zigbee-herdsman-converters/blob/05f0795cbf3742a17d59152bbe2186ec6bc8a48e/lib/color.js#L644

`findColorTempRange()` which is used to do the range lookup to feed to `clampColorTemp()` requires it, not sure what a clean way is to get the information, aside from perhaps trying to pass 'entity' to `syncColorState()`.

Wikibear, 2021-11-26:
> What is the temp range given by the bulb? https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

I don't know how to help here. I don't know the real limitation of bulb.

Koenkk, 2021-11-27:
@sjorge for fromZigbee, the device and endpoint can be accessed through `msg.endpoint`/`msg.device`

closed 2021-11-28 by Koenkk