Koenkk/zigbee2mqtt #9845: Datek Meter Reader stopped creating sensors for Voltage at some point

TheStigh, 2021-11-26:
@Koenkk 

Hi, at some point between early summer and now, suddenly sensors for `voltage` are gone.
In attributes, everything are in place, but the `sensor.<entity>_voltage` + `...voltage_phase_b` + `...voltage_phase_c` are missing

Re-pairing do not help.

Have anything been done on the seMetering/haElectricalMeasurement after the summer?

Look at `await endpoint.configureReporting('haElectricalMeasurement', payload);`
`sensor.<entity>_current_phase_b` + ...`_current_phase_c` are showing as sensors? Why not Voltage?

Here's the converter I published before the summer, and everything was working:
```
    {
        zigbeeModel: ['Meter Reader'],
        model: 'HSE2905E',
        vendor: 'Datek',
        description: 'Datek Eva AMS HAN power-meter sensor',
        fromZigbee: [fz.metering_datek, fz.electrical_measurement, fz.temperature],
        toZigbee: [],
        ota: ota.zigbeeOTA,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering', 'msTemperatureMeasurement']);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            const payload = [{
                attribute: 'rmsVoltagePhB',
                minimumReportInterval: 10,
                maximumReportInterval: 3600,
                reportableChange: 0,
            },
            {
                attribute: 'rmsVoltagePhC',
                minimumReportInterval: 10,
                maximumReportInterval: 3600,
                reportableChange: 0,
            },
            {
                attribute: 'rmsCurrentPhB',
                minimumReportInterval: 10,
                maximumReportInterval: 3600,
                reportableChange: 0,
            },
            {
                attribute: 'rmsCurrentPhC',
                minimumReportInterval: 10,
                maximumReportInterval: 3600,
                reportableChange: 0,
            }];
            await endpoint.configureReporting('haElectricalMeasurement', payload);
            await reporting.rmsVoltage(endpoint, {min: 10, max: 3600, change: 0});
            await reporting.rmsCurrent(endpoint, {min: 10, max: 3600, change: 0});
            await reporting.instantaneousDemand(endpoint, {min: 10, max: 3600, change: 0});
            await reporting.currentSummDelivered(endpoint, {min: 10, max: 3600, change: [1, 1]});
            await reporting.currentSummReceived(endpoint);
            await reporting.temperature(endpoint);
        },
        exposes: [e.power(), e.energy(), e.current(), e.voltage(), e.current_phase_b(), e.voltage_phase_b(), e.current_phase_c(),
            e.voltage_phase_c(), e.temperature()],
    },
```

Koenkk, 2021-11-26:
I believe these sensor are disabled by default in HA, you can enable them in HA.

TheStigh, 2021-11-26:
> I believe these sensor are disabled by default in HA, you can enable them in HA.

Never seen this... You say 'believe'?... Tried some google searches but didn't find anything about this. Can uou point me to the solution?

closed 2021-11-26 by TheStigh