Koenkk/zigbee2mqtt #9847: ORVIBO ST30 often reports none for temperature

simulot, 2021-11-26:
## What happened
The device temperature is often seen as none in Home Assistant. This is visible on the graph:
[image]

In the same time, the humidity report is continuous.
[image]

## What did you expect to happen
I'm expecting having a continuous reading of the temperature.

## How to reproduce it (minimal and precise)
This is like that since I have added the device 

## Debug info
Zigbee2MQTT version:     1.22.0 commit: 0a1fccf 
Adapter hardware: ConBee2/RaspBee2
Adapter firmware version:     0x26580700

simulot, 2021-11-27:
For Zigbee2MQTT device panel, we read null
[image]

This explains the "none" in Home Assistant

simulot, 2021-11-27:
And a log entry 
```
info MQTT publish: topic 'zigbee2mqtt/0x5c0272fffe02bb79', payload '{"battery":100,"device":{"applicationVersion":4,"dateCode":"20201123","friendlyName":"0x5c0272fffe02bb79","hardwareVersion":1,"ieeeAddr":"0x5c0272fffe02bb79","manufacturerID":4098,"manufacturerName":"MultIR","model":"ST30","networkAddress":21795,"powerSource":"Battery","softwareBuildID":"ST30_v1.0.4","type":"EndDevice","zclVersion":3},"humidity":44,"linkquality":255,"temperature":null}'
```

**"temperature":null**

closed 2022-01-04 by github-actions[bot]

traian, 2022-01-17:
Having the exact same problem with it

simonbaudart, 2022-03-15:
Hello, I have the same issue, I don't know how to fix or to help fix that.
Adapter : Sonoff USB

[image]

I'll try to set the log in debug to see what is received on Zigbee Network.

I have no issue with same device on ZHA and reports are done more often.

Can somebody help ?

zipitrik, 2022-06-08:
I just bought this model  and I have the exact same problem. I feel like I wasted my money :( ...  I wish this would have been reported/noted on this page https://www.zigbee2mqtt.io/devices/ST30.html  because I checked there before buying to be sure I was going to be fine (I am not blaming anybody for anything of course)

simulot, 2022-06-12:
> I wish this would have been reported/noted on this page https://www.zigbee2mqtt.io/devices/ST30.html
Report done.

zipitrik, 2022-06-12:
Thanks ! This may never gets resolved but at least the device integration info page will be accurate

Koenkk, 2022-06-16:
@zipitrik can you provide the debug log when temperature null is reported?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

simulot, 2022-06-17:
Hi @Koenkk 
I still have 2 of those devices in my setup.
Here is the log for ST30 devices

One is located in a bedroom, the other is in the freezer (so -20° is expected as value.

```
Zigbee2MQTT:info  2022-06-16 19:57:25: 0x5c0272fffe02bb79 (0x5c0272fffe02bb79): ST30 - ORVIBO Temperature & humidity sensor (EndDevice)
Zigbee2MQTT:info  2022-06-16 19:57:25: Capteur Chambre Thomas (0x84fd27fffe714d42): ST30 - ORVIBO Temperature & humidity sensor (EndDevice)
Zigbee2MQTT:info  2022-06-16 19:57:27: MQTT publish: topic 'homeassistant/sensor/0x5c0272fffe02bb79/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x5c0272fffe02bb79"],"manufacturer":"ORVIBO","model":"Temperature & humidity sensor (ST30)","name":"0x5c0272fffe02bb79","sw_version":"ST30_v1.0.4"},"device_class":"humidity","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x5c0272fffe02bb79","name":"0x5c0272fffe02bb79 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0x5c0272fffe02bb79","unique_id":"0x5c0272fffe02bb79_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
Zigbee2MQTT:info  2022-06-16 19:57:27: MQTT publish: topic 'homeassistant/sensor/0x5c0272fffe02bb79/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x5c0272fffe02bb79"],"manufacturer":"ORVIBO","model":"Temperature & humidity sensor (ST30)","name":"0x5c0272fffe02bb79","sw_version":"ST30_v1.0.4"},"device_class":"temperature","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x5c0272fffe02bb79","name":"0x5c0272fffe02bb79 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x5c0272fffe02bb79","unique_id":"0x5c0272fffe02bb79_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Zigbee2MQTT:info  2022-06-16 19:57:27: MQTT publish: topic 'homeassistant/sensor/0x5c0272fffe02bb79/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x5c0272fffe02bb79"],"manufacturer":"ORVIBO","model":"Temperature & humidity sensor (ST30)","name":"0x5c0272fffe02bb79","sw_version":"ST30_v1.0.4"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x5c0272fffe02bb79","name":"0x5c0272fffe02bb79 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x5c0272fffe02bb79","unique_id":"0x5c0272fffe02bb79_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2022-06-16 19:57:27: MQTT publish: topic 'homeassistant/sensor/0x5c0272fffe02bb79/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x5c0272fffe02bb79"],"manufacturer":"ORVIBO","model":"Temperature & humidity sensor (ST30)","name":"0x5c0272fffe02bb79","sw_version":"ST30_v1.0.4"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x5c0272fffe02bb79","name":"0x5c0272fffe02bb79 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x5c0272fffe02bb79","unique_id":"0x5c0272fffe02bb79_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2022-06-16 19:57:28: MQTT publish: topic 'homeassistant/sensor/0x84fd27fffe714d42/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x84fd27fffe714d42"],"manufacturer":"ORVIBO","model":"Temperature & humidity sensor (ST30)","name":"Capteur Chambre Thomas","sw_version":"ST30_v1.0.4"},"device_class":"humidity","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/Capteur Chambre Thomas","name":"Capteur Chambre Thomas humidity","state_class":"measurement","state_topic":"zigbee2mqtt/Capteur Chambre Thomas","unique_id":"0x84fd27fffe714d42_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
Zigbee2MQTT:info  2022-06-16 19:57:28: MQTT publish: topic 'homeassistant/sensor/0x84fd27fffe714d42/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x84fd27fffe714d42"],"manufacturer":"ORVIBO","model":"Temperature & humidity sensor (ST30)","name":"Capteur Chambre Thomas","sw_version":"ST30_v1.0.4"},"device_class":"temperature","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/Capteur Chambre Thomas","name":"Capteur Chambre Thomas temperature","state_class":"measurement","state_topic":"zigbee2mqtt/Capteur Chambre Thomas","unique_id":"0x84fd27fffe714d42_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Zigbee2MQTT:info  2022-06-16 19:57:28: MQTT publish: topic 'homeassistant/sensor/0x84fd27fffe714d42/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x84fd27fffe714d42"],"manufacturer":"ORVIBO","model":"Temperature & humidity sensor (ST30)","name":"Capteur Chambre Thomas","sw_version":"ST30_v1.0.4"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Capteur Chambre Thomas","name":"Capteur Chambre Thomas battery","state_class":"measurement","state_topic":"zigbee2mqtt/Capte [... cut ...] 26.2}'
Zigbee2MQTT:info  2022-06-16 21:33:35: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffe02bb79', payload '{"battery":90,"device":{"applicationVersion":4,"dateCode":"20201123","friendlyName":"0x5c0272fffe02bb79","hardwareVersion":1,"ieeeAddr":"0x5c0272fffe02bb79","manufacturerID":4098,"manufacturerName":"MultIR","model":"ST30","networkAddress":21795,"powerSource":"Battery","softwareBuildID":"ST30_v1.0.4","type":"EndDevice","zclVersion":3},"humidity":49,"linkquality":255,"temperature":-20}'
```

simulot, 2022-06-17:
Here is the graph produced in Home Assitant:
[image]

Holes in the graph matches the null value.

I think that the device transmits sometime null if the temperature hasn't changed since last transmission. However humidity is reported correctly. It looks like a firmware bug.

Is it possible to compensate the problem by report last known value instead of null?

Koenkk, 2022-06-17:
I'm a bit puzzled by your logging, e.g.:

```
Zigbee2MQTT:info  2022-06-16 21:23:31: MQTT publish: topic 'zigbee2mqtt/Capteur Chambre Thomas', payload '{"battery":93.5,"device":{"applicationVersion":4,"friendlyName":"Capteur Chambre Thomas","hardwareVersion":1,"ieeeAddr":"0x84fd27fffe714d42","manufacturerID":4098,"manufacturerName":"MultIR","model":"ST30","networkAddress":15591,"powerSource":"Battery","softwareBuildID":"ST30_v1.0.4","type":"EndDevice","zclVersion":3},"humidity":43.5,"linkquality":255,"temperature":null}'
Zigbee2MQTT:info  2022-06-16 21:24:10: MQTT publish: topic 'zigbee2mqtt/Capteur Chambre Thomas', payload '{"battery":93.5,"device":{"applicationVersion":4,"friendlyName":"Capteur Chambre Thomas","hardwareVersion":1,"ieeeAddr":"0x84fd27fffe714d42","manufacturerID":4098,"manufacturerName":"MultIR","model":"ST30","networkAddress":15591,"powerSource":"Battery","softwareBuildID":"ST30_v1.0.4","type":"EndDevice","zclVersion":3},"humidity":43.5,"linkquality":255,"temperature":26.2}'
Zigbee2MQTT:info  2022-06-16 21:33:15: MQTT publish: topic 'zigbee2mqtt/Capteur Chambre Thomas', payload '{"battery":93.5,"device":{"applicationVersion":4,"friendlyName":"Capteur Chambre Thomas","hardwareVersion":1,"ieeeAddr":"0x84fd27fffe714d42","manufacturerID":4098,"manufacturerName":"MultIR","model":"ST30","networkAddress":15591,"powerSource":"Battery","softwareBuildID":"ST30_v1.0.4","type":"EndDevice","zclVersion":3},"humidity":46.6,"linkquality":255,"temperature":26.2}'
```

The messages seems to be published out of no where (normally you would see `Received zigbee message..` before this). Can you provide your `configuration.yaml`?

simulot, 2022-06-17:
Sure: 

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.10.20
  user: broker
  password: 28a4be0cfe
  client_id: zigb2mqtt
  include_device_information: true
serial:
  adapter: deconz
  port: /dev/ttyACM1
frontend:
  port: 8080
  host: 0.0.0.0
advanced:
  log_level: debug
  network_key:
    ** redacted **
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: homeassistant/status
devices:
  '0xf0d1b800001391a7':
    friendly_name: Luminaire Ikea 2-3
  '0xf0d1b80000138984':
    friendly_name: Luminaire Ikea 3-3
  '0xf0d1b80000138fdf':
    friendly_name: Luminaire Ikea 1-3
  '0x00158d00012e539d':
    friendly_name: Globe RVB Thomas
    retrieve_state: true
  '0x7cb03eaa0a0730b3':
    friendly_name: Chambre de Thomas 2
    transition: 0
  '0x7cb03eaa0a098bc6':
    friendly_name: Prise 1
  '0x04cf8cdf3c78d0d6':
    friendly_name: Luminosité façade sud
  '0x60a423fffedb2eab':
    friendly_name: 4 gangs Thomas
  '0x00158d000651825c':
    friendly_name: Humidité salle sous-sol
  '0x5c0272fffe051588':
    friendly_name: Chambre Paul
  '0x5c0272fffe043dbe':
    friendly_name: Chambre Thomas
  '0x804b50fffefb4e22':
    friendly_name: Lampe cheminée
  '0xb4e3f9fffe584ee4':
    friendly_name: Variateur Etagères salon
  '0x847127fffe25e3b7':
    friendly_name: LED Etagères Salon
  '0xec1bbdfffe6d9236':
    friendly_name: RVB Bureau RdC
    transition: 0.1
    color_sync: true
  '0xb4e3f9fffe22a7ee':
    friendly_name: Interrupteur Led cuisine
  '0xbc33acfffe13a089':
    friendly_name: LED Lit Thomas
  '0xb4e3f9fffe8644e1':
    friendly_name: Variateur lit Thomas
  '0x5c0272fffeb62e42':
    friendly_name: Lampe JF
  '0x5c0272fffe4faa5c':
    friendly_name: Télécommande lampe chevet JF
  '0x00158d00068b8973':
    friendly_name: Porte garage
    debounce: 1
    optimistic: true
  '0xcc86ecfffe3ab355':
    friendly_name: Fontaine Mermoz
  '0x00158d0007068811':
    friendly_name: Exterieur
  '0xcc86ecfffe3ab337':
    friendly_name: Prise 2
  '0xb4e3f9fffe22a95e':
    friendly_name: Radiateur salle de bain parents
  '0x5c0272fffe02bb79':
    friendly_name: '0x5c0272fffe02bb79'
    temperature_precision: 0
    humidity_precision: 0
    temperature_calibration: 0
  '0x84fd27fffe714d42':
    friendly_name: Capteur Chambre Thomas
    humidity_precision: 1
    temperature_precision: 1
  '0x3c6a2cfffed0cf0e':
    friendly_name: Master Chauffage
  '0x3c6a2cfffed0cf05':
    friendly_name: Master Pompe à chaleur
  '0x00158d0005a21c07':
    friendly_name: Entrée
  '0x680ae2fffefdcbbf':
    friendly_name: Télécommande Chauffage SdB Enfants
  '0x54ef441000125f49':
    friendly_name: Chauffage Salle de bain enfants
  '0x04cf8cdf3c78dcbb':
    friendly_name: Luminosité façade nord
  '0x00158d0006517f4b':
    friendly_name: Buanderie
  '0x804b50fffe5904a9':
    friendly_name: Mouvement cuisine
  '0x7cb03eaa0a0172f2':
    friendly_name: '0x7cb03eaa0a0172f2'
groups:
  '1':
    friendly_name: Luminaire Ikea
    devices:
      - 0xf0d1b80000138fdf/1
      - 0xf0d1b800001391a7/1
      - 0xf0d1b80000138984/1
  '2':
    friendly_name: Groupe Etagères salon
    devices:
      - 0x847127fffe25e3b7/1
      - 0xb4e3f9fffe584ee4/1
  '3':
    friendly_name: Groupe LED Lit Thomas
    devices:
      - 0xbc33acfffe13a089/1
  '4':
    friendly_name: Groupe RVB Bureau
    devices:
      - 0xec1bbdfffe6d9236/1
  '5':
    friendly_name: Groupe commde chambre parents
    devices:
      - 0x5c0272fffe4faa5c/1
      - 0x5c0272fffeb62e42/1
  '65289':
    friendly_name: Trafri_scenes
    devices:
      - 0x5c0272fffe4faa5c/1
      - 0x5c0272fffeb62e42/1
blocklist:
  - '0x7cb03eaa0a09ac06'
external_converters:
  - CCB432.js```

Koenkk, 2022-06-18:
What is in the `CCB432.js` converter?

simulot, 2022-06-18:
The is the converter for energy meter https://www.zigbee2mqtt.io/devices/CCB432.html#casaia-ccb432
The problem with ST30 predates the installation of CCB432
It's now in the master branch. I can remove it.

Koenkk, 2022-06-18:
Could you provide the herdsman debug log when the temperature goes from a value to `null`?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

simulot, 2022-06-18:
0x5c0272fffe02bb79 is the thermometer in the fridge
I hope I have kept enough data.

When temperature is reported:
```
2022-06-18T08:41:44.075Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":81,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":33,"dataType":32,"attrData":180}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":21795,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-06-18 08:41:44: Received Zigbee message from '0x5c0272fffe02bb79', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":180}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-06-18 08:41:44: Configuring '0x5c0272fffe02bb79'
2022-06-18T08:41:44.088Z zigbee-herdsman:controller:endpoint Bind 0x5c0272fffe02bb79/1 msTemperatureMeasurement from '0x00212effff06a3c7/1'
2022-06-18T08:41:44.091Z zigbee-herdsman:controller:endpoint DefaultResponse 0x5c0272fffe02bb79/1 1(10, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-06-18T08:41:44.092Z zigbee-herdsman:deconz:adapter no response expected
2022-06-18T08:41:44.092Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2022-06-18 08:41:44: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffe02bb79', payload '{"battery":90,"device":{"applicationVersion":4,"dateCode":"20201123","friendlyName":"0x5c0272fffe02bb79","hardwareVersion":1,"ieeeAddr":"0x5c0272fffe02bb79","manufacturerID":4098,"manufacturerName":"MultIR","model":"ST30","networkAddress":21795,"powerSource":"Battery","softwareBuildID":"ST30_v1.0.4","type":"EndDevice","zclVersion":3},"humidity":58,"linkquality":255,"temperature":-15}'

```

Then later, temperature is null
```
2022-06-18T09:02:23.671Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-06-18T09:02:23.672Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-06-18T09:02:23.675Z zigbee-herdsman:deconz:driver query aps data indication
2022-06-18T09:02:23.680Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 173
2022-06-18T09:02:23.684Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 173 srcAddr: 0x5523 destAddr: 0x0 profile id: 0x104 cluster id: 0x402 lqi: 255
2022-06-18T09:02:23.684Z zigbee-herdsman:deconz:frameParser response payload: 8,94,10,62,0,24,0
2022-06-18T09:02:23.692Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":94,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":62,"dataType":24,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":21795,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-06-18 09:02:23: Received Zigbee message from '0x5c0272fffe02bb79', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"62":0}' from endpoint 1 with groupID null
2022-06-18T09:02:23.709Z zigbee-herdsman:controller:endpoint DefaultResponse 0x5c0272fffe02bb79/1 1026(10, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-06-18T09:02:23.710Z zigbee-herdsman:deconz:adapter no response expected
2022-06-18T09:02:23.710Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2022-06-18 09:02:23: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffe02bb79', payload '{"battery":90,"device":{"applicationVersion":4,"dateCode":"20201123","friendlyName":"0x5c0272fffe02bb79","hardwareVersion":1,"ieeeAddr":"0x5c0272fffe02bb79","manufacturerID":4098,"manufacturerName":"MultIR","model":"ST30","networkAddress":21795,"powerSource":"Battery","softwareBuildID":"ST30_v1.0.4","type":"EndDevice","zclVersion":3},"humidity":58,"linkquality":255,"temperature":null}'

```

Koenkk, 2022-06-18:
Found the issue, fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

simulot, 2022-06-18:
A couple of hours later, it's confirmed to be working.
[image]

Thank you sir!
We can close this definitively.

zipitrik, 2022-06-21:
@Koenkk  @simulot  Sorry I've been away for a few days. I see you have solved it !! Thank you so much. Merci Jean-Francois for providing your logs and helping . HA is amazing and that's only because you guys are amazing :)