Koenkk/zigbee2mqtt #9858: Moes TV01-ZB: Many errors while join and actual temperature is freezed

ChrisLSaar, 2021-11-26:
Hello,

when connecting the device, there are very many error messages.
After that it works in principle, but the actual temperature is not up to date. 
Attached is a download of the status.

Can someone here give me more detailed info?

Christian

[state.json (1).zip](https://github.com/Koenkk/zigbee2mqtt/files/7610508/state.json.1.zip)

thiscantbeserious, 2021-11-29:
From how I see it this is the Datapoint 115 - its called Check01

[image]

`{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "115",
    "startTime": 1638025243968,
    "endTime": 1638198043969,
    "projectCode": "CENSORED",
    "sourceId": "CENSORED",
    "sourceType": "4",
    "deviceId": "CENSORED",
    "pageStartRow": "",
    "region": "EU"
}`

Specifically this payload here seems interesting when I go onto the App - the temperature refreshes and doesn't light up the display.

`<html>
<body>

2021-11-29 15:52:02 | Publish | Check01 | off | app client | appName:涂鸦智能,ios-packageName:com.tuya.smart,android-packageName:com.tuya.smart,model:Pixel 6,os_system:12,app_version:3.33.0,model:,os_system:,app_version:
-- | -- | -- | -- | -- | --

</body>
</html>`

thiscantbeserious, 2021-11-29:
It seems like the TV01-ZG and TV01-ZB variant from Moes are currently implemented in a way that's not complete and bugged. 

I'm in the process of merging @vladi1234's code over from Add TV02-Zigbee (_TZE200_hue3yfsn) (PR #3291) which should also add working boost mode, schedule and a way to query for is_online (DP 115 mentioned above, which gives you the current local_temprature without lightning up the screen, which the Tuya App also uses).

thiscantbeserious, 2021-11-30:
So, took me a while on how to test his thing on Hass OS (on a Raspi 4).

Pushing to the container doing a full build within the Alpine .. could be better documented / scripted ...

Good news:

It works. 

No more unkown DPs and online check works too without lighting up the screen.

The schedule is looking not that good tought, so maybe the format is different.

Also holiday mode only shortly turns on, then turns back to auto.

No clue why, but code-base looks better to me now ... so that be worked on.

vladi1234, 2021-11-30:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/3142#issuecomment-962460788

closed 2022-01-07 by github-actions[bot]