Koenkk/zigbee2mqtt #9860: New Philips HUE White and Color 1100 lm Not supported

LagunaPower, 2021-11-27:
## Information about the device + link
New philips hue white and color 1100
Model : 9290024688

https://www.philips-hue.com/fr-fr/p/hue-white-and-color-ambiance-pack-de-1-e27/8719514291171#specifications

## data/database.db entry of the device

{
	"id":20,
	"type":"Router",
	"ieeeAddr":"0x001788010b1d376e",
	"nwkAddr":11812,
	"manufId":4107,
	"manufName":"Signify Netherlands B.V.",
	"powerSource":"Mains (single phase)",
	"modelId":"LCA006",
	"epList":[11,242],
	"endpoints": {
		"11":{
			"profId":260,
			"epI
			d":11,
			"devId":269,
			"inClusterList":[0,3,4,5,6,8,4096,64514,768,64513],
			"outClusterList":[25],
			"clusters":{
				"genBasic":{
					"attributes":{
						"modelId":"LCA006",
						"manufacturerName":"Signify Netherlands B.V.",
						"powerSource":1,
						"zclVersion":2,
						"appVersion":2,
						"stackVersion":1,
						"hwVersion":1,
						"dateCode":"20201117",
						"swBuildId":"1.76.11"
					}
				}
			},
			"binds":[],
			"configuredReportings":[],
			"meta":{}
		},
		"242":{
			"profId":41440,
			"epId":242,
			"devId":97,
			"inClusterList":[],
			"outClusterList":[33],
			"clusters":{},
			"binds":[],
			"configuredReportings":[],
			"meta":{}
		}
	},
	"appVersion":2,
	"stackVersion":1,
	"hwVersion":1,
	"dateCode":"20201117",
	"swBuildId":"1.76.11",
	"zclVersion":2,
	"interviewCompleted":true,
	"meta":{},
	"lastSeen":1637966518964
}

Koenkk, 2021-11-27:
Already supported in the dev branch (or next release on 1 December). (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-11-27 by Koenkk

skela, 2021-12-16:
Does not seem to work, at least not with docker-compose and the latest-dev image.

koenkk/zigbee2mqtt:latest-dev
Digest:sha256:d885c84c18d7d278ee9a11ae0b5581413e9544303b99c3298acad9e9a28ae558

According to the the docker image above aka d885c84c18d7d278ee9a11ae0b5581413e9544303b99c3298acad9e9a28ae558

zigbee2mqtt reports the following in Settings/About
1.21.2 commit: 4492212 

So maybe latest-dev no longer has the fix?