Koenkk/zigbee2mqtt #9865: CC2530+CC2591 adapter works only the fist time but needs a power cycle to work again

damocles-dev, 2021-11-27:
## What happened

Using a CC2530+CC2591 adapter  works perfectly the first time it starts up but if you try to start it again it will no longer be able to start. 

This problem happens from v1.21.x onwards, the last time it worked  was v1.20.0.

## How to reproduce it (minimal and precise)

I did several tests with v1.21 and v1.22 and these were the results:

Intel Nuc + pl2303 usb/ttl adapter + CC2530+CC2591 adapter :   second boot fails
raspberry pi 4 / raspberry pi zero v2 integrated uart + CC2530+CC2591 adapter: second boot fails

Intel Nuc +  CC2531 USB adapter :   second boot woks
raspberry pi 4 / raspberry pi zero v2 + CC2531 USB adapter: second boot works

## Debug info
Zigbee2MQTT version: the last working version that worked was 1.20.0 and fails with 1.21.x and 1.22.0 
Adapter hardware: CC2530+CC2591
Adapter firmware version: zStack12 / 20201127

Failed second boot:
<pre>
$ DEBUG="zigbee-herdsman*" npm start

> zigbee2mqtt@1.22.0 start
> node index.js

Zigbee2MQTT:info  2021-11-27 12:09:19: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-11-27.12-09-15' filename: log.txt
Zigbee2MQTT:info  2021-11-27 12:09:20: Starting Zigbee2MQTT version 1.22.0 (commit #0a1fccf6)
Zigbee2MQTT:info  2021-11-27 12:09:20: Starting zigbee-herdsman (0.13.164)
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[<edited>,"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyAMA0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyAMA0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +19ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +0ms
  zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload +271ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +260ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +1s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352 +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +713ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +713ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2021-11-27 12:09:41: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-11-27 12:09:41: Failed to start zigbee
Zigbee2MQTT:error 2021-11-27 12:09:41: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-11-27 12:09:41: Exiting...
Zigbee2MQTT:error 2021-11-27 12:09:42: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
npm ERR! code 1
npm ERR! path /opt/zigbee2mqtt
npm ERR! command failed
npm ERR! command sh -c node index.js

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-11-27T11_09_42_365Z-debug.log
</pre>

First boot:
<pre>
$ DEBUG="zigbee-herdsman*" npm start

> zigbee2mqtt@1.22.0 start
> node index.js

Zigbee2MQTT:info  2021-11-27 12:21:52: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-11-27.12-21-47' filename: log.txt
[... 48 lines cut ...]
  zigbee-herdsman:adapter:zStack:startup:backup fetched adapter address manager table (capacity=0, used=0) +18s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":76} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,76,0,124] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":76} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,76,0,124] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":76} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,76,0,124] +6s
  zigbee-herdsman:adapter:zStack:startup:backup fetched adapter security manager table (capacity=0, used=0) +18s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":513} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,1,2,51] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":513} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,1,2,51] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":513} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,1,2,51] +6s
  zigbee-herdsman:adapter:zStack:startup:backup fetched adapter aps link key data table (capacity=0, used=0) +18s
  zigbee-herdsman:adapter:zStack:startup:backup fetched adapter tclk seed +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":273} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,17,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":273} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,17,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":273} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,17,1,32] +6s
  zigbee-herdsman:adapter:zStack:startup:backup fetched adapter tclk table (capacity=0, used=0) +18s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength 
</pre>

Koenkk, 2021-11-28:
Flashing the fw from https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_Home_1.2 should fix the problem.

damocles-dev, 2021-11-29:
Thanks Koen, I missed that update. Now it works.

closed 2021-11-29 by damocles-dev