Koenkk/zigbee2mqtt #9866: Groups ignore invert_cover flag on blinds

Sangoku, 2021-11-27:
## What happened
Added 4 blinds to a Zigbee Groups via the UI.
One of them hast the flag **invert_cover** on true.

Using the newly generated Blinds Group in HA to set the percentage of the cover causes the blinds to have an opposite direction. 

## What did you expect to happen
That the blinds  don't wink at me 😵‍💫  setting them to 80% causes one to go to 20 and the rest to 80%

## How to reproduce it (minimal and precise)
Add 2 https://www.zigbee2mqtt.io/devices/MS-108ZR.html#moes-ms-108zr tp ha. Set one of them to invert_cover = true and try to use them. 

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531 Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001CD4C277
Adapter firmware version: TI\x20CC2531\x20USB\x20CDC

Koenkk, 2021-11-28:
The way invert_cover works is by simply inverting the position send to the devices. This means you cannot use for only 1 device in a group.

Sangoku, 2021-11-29:
> The way invert_cover works is by simply inverting the position send to the devices. This means you cannot use for only 1 device in a group.

So this means i can assign the group the flag "invert_cover "? 
**That is what i was looking for**. 

Because now the group entity reports it is closed when the normal entities are open and vice versa. 
I have every blind separately with the flag "invert_cover" set. 
But when i use them with the group entity in HA they are reversed. 
Aka the group ignores the flag "invert_cover" which i set for every blind separately.

Koenkk, 2021-11-29:
> I have every blind separately with the flag "invert_cover" set.

Do all members have `invert_cover: true` or just one? From your OP I understood one:

> One of them hast the flag invert_cover on true.

Sangoku, 2021-11-29:
Didn't want to explain the whole proces. 
But here it goes.
 I have 6 blinds. If i didn't set the flag then alexa would close the blinds when i say to open them. And when i inverted the motors the Ha ui woulg go reverse.

closed 2021-11-29 by Sangoku

Sangoku, 2021-11-30:
Didn't want to explain the whole proces. 
But here it goes.
 I have 6 blinds. If i didn't set the flag then alexa would close the blinds when i say to open them. And when i inverted the motors the Ha ui would go reverse. One i was to lazy to open and kept it in reverse without the flag because there was no alexa in the room. Fast forward to me creating a zigbee group to turn all blinds and the one without the flag was the only one in reverse in group.  Now i fixe d the one and now all have the flag set. 
Buuut 
Ha sees them reversed. When i open the  group to 100 they go to 0 and i see it on other entities going to 0 while the group goes to 100.