Koenkk/zigbee2mqtt #9867: IKEA E2001/E2002 STYRBAR remote controller - incorrect long arrows press recognition

gasecki, 2021-11-27:
## What happened

IKEA E2001/E2002
Incorrect event recognition when long press left arrow / right arrow. Instead of expected event **arrow_left_hold** / **arrow_right_hold** Event **on** is recognized. 

## What did you expect to happen
Z2M should report arrow_left_hold, arrow_left_release, arrow_right_hold, arrow_right release

## How to reproduce it (minimal and precise)
Long press left / right arrow

## Debug info
Zigbee2MQTT version: 1.22.0-3
Adapter hardware: Conbee II
Adapter firmware version: unknown

MattWestb, 2021-11-28:
Thats is because the remote is sending one reset / sync sequence to the grope of light is controlling and setting all light in the same scene mode and on with 100% brightness.
Its one light controller and dont sending "click click hold" is sending light commands as it suppose doing as one **real** light controller device.

Read the manual for the device for knowing how to using it right.

If you like have it working in one different  way pleas make one new firmware for it so user can using it as one multi click devices in there system ala Aqara (that cant controlling lights).

gasecki, 2021-11-28:
Looks like one of us didn’t get the point right.

It is difficult to answer you as I find your comment not only irrelevant to the matter but also rude.

I checked zigbee2mqqt product page and based on that I know what to expect in my MQTT feed. What I get is not what I expect, but only for left/right buttons. Buttons brightness_up/down report three events as described.

No it is not light controller. You can turn on your fishtank air pump if you have such use case. You will not find it in manual. 

https://www.zigbee2mqtt.io/devices/E2001_E2002.html

—
Tomasz

MattWestb, 2021-11-28:
OK its up to you but i have only sniffing all IKEA controller over 100 hours so i knowing how they is working and shall being used (and not abused).

You can looking what "profile_id": 260 and "device_type": "0x0820" is in the Zigbee  specification. (https://zigbeealliance.org/wp-content/uploads/2019/11/docs-15-0014-05-0plo-Lighting-OccupancyDevice-Specification-V1.0.pdf)

And light controller is sending on/off commands and also possible binding so they can setting other device like outlets that's not real lights ala Zigbee standard.

If i may explaining my point with one example: You is baying one Nissan Micra and like using it for transporting fridges that is 2 meter high = no good match but expected its not working.

Im sorry i dont intend being rude but its still one light controller device.

gasecki, 2021-11-28:
This is zigbee2mqtt forum right? 

I have other Ikea remotes including 5 button one, other aqara remotes, other noname zigbee led controllers. 
All of them are exposing values and enumerate action values as described in Z2M relevant product page.
Z2M is not the use case as producers want. Producers want their devices to work with their apps and within their network.
Z2M is to collect whichever message whichever device sends with ZIGBEE protocol and convert it into MQTT message. Do I still need to explain that? 
Unfortunately this will not solve the issue I’m trying to but I hope it closes this off-topic. 
—
Tomasz

Rawq86, 2021-12-05:
The same problem, 
Incorrect action recognition when long press left arrow / right arrow. Instead of expected action arrow_left_hold / arrow_right_hold Event on is recognized.

Any ideas?

nodamnway, 2021-12-14:
If you will hold button long enough, `arrow_*_hold` and `arrow_*_release` will be reported after `on` event.

So technically they are reported, but not in the way we want.

rbalaev, 2021-12-23:
Same problem. with a long hold, the ON command is first sent, then the required arrow_*_hold.

Koenkk, 2021-12-23:
@kampari89  could you check this?

mbauwens, 2022-01-08:
Have there been any updates on this? I also have the same problem with `on` being sent prior to the correct arrow hold messages.

hschwane, 2022-01-16:
I have the same problem. It seems the remote itself is sending an on command before the arrow down command is send. So this is not easily solved on the mqtt side. 
@MattWestb you mentioned custom firmware to remove this and run the TYRBAR as 4 individual buttons. Do you have any idea on where to find that or anyone who would be able to help modifying the firmware?

namadori, 2022-01-31:
Same issue. I specifically got the Styrbar to use as two independent remotes, for the light and the rolling shutter of the room, but with this "on" command it will turn on the light every time i move the shutter. Really annoying.

I understand this is not Z2M fault, but at least it would be useful to explain it in the docs. I would do it, but I'm not a native english speaker, I believe someone else can do the job better.

umekzg, 2022-02-12:
Just bought one as the old 5 button remote went EOL.
So basically left/right hold is unusable for custom integration.

git-developer, 2022-03-15:
This issue should not be closed before the documentation is updated.

mbauwens, 2022-03-15:
I'm willing to update the [docs](https://www.zigbee2mqtt.io/devices/E2001_E2002.html) in a PR to reflect this issue but I'm not familiar with the documentation guidelines. I'm sufficiently fluent in English though 😄

Should this be mentioned as a "Known issue" under "Notes", for example, or should this be mentioned as a note under the "Exposes" - "Action (enum)" heading? @Koenkk ?

Nebula83, 2022-04-13:
Same issue here, and kind of a let down to be honest. Esthetically the new remote is a lot more pleasing, behaviorally: not so much. Since I directly bound the remote to the light, it isn't easily ignored as well. I restored to a tailor made double click system on the automation side, but it's ugly. Any updates in a custom firmware?

noobadmin, 2022-05-05:
Bumping as it's really annoying. It would be awesome if someone came up with a firmware patch. Also, @MattWestb congrats on being an expert on ikea switches / zigbee stuff 👍🏼 Hope, you're getting roasted at least as much as you did OP when asking for help.

antonio1475, 2022-05-29:
Damn, I just found this after running into the issue with my first Styrbar remote to replace an E1810 that I was using to control 2 lights (one with up/down, one with left/right).

Now _left/right hold_ causes the other one to turn on because the payload is the same (on) with _up_ than _left/right hold_
This doesn't happen with E1810 and it's a pity IKEA designed it like this :(

I'm trying to figure out a workaround since I don't have the light binded (controll all through automations and conditions for more advanced controls), but the payload is so simple there's nothing to work with...

namadori, 2022-05-29:
Yep, same issue here. I am still hoping that someone smarter than me will find a workaround.

penroseg, 2022-06-16:
Not a fix and maybe what people are already doing but as the [arrow_*_hold] _is_ actually sent after the On command, I have temporarily disabled the automation associated with the Press On (Brightness Up) button and this allows my Left Hold & Right Hold automations to work as expected.  It's not great but it gives me an extra working button (7 out of 8 potential)...  

The only thing is it's possibly more frustrating not having the On button working but as it's the only indented button on the switch I'm using it for a 'master' action to trigger a Default Area scene rather than associating to a device or devices.

FYI, I don't have my switch connected to a bulk/device and am solely using it for Zigbee2MQTT so YMMV.

antonio1475, 2022-06-17:
@penroseg I went beyond that, while still keeping the up button working (even though not ideal, but better than disabled):

1. Up button short click action triggers a Script.
2. The script waits for 1.5 seconds listening to see if arrow_right_hold or arrow_left_hold happens (that means the action_button_up_short was triggered by a left/right long press). 1,5 seconds works 98% of the time for me.
3. If 2 happens, then the script is stopped, allowing the left/right hold to happen without up_button happening too.
If 2 doesn't happen, it means button_up was triggered by actually pressing the up remote, so it does the expected action (light on in my case).

I hope I made sense.
It's not great because up_button has a delay of 1.5 seconds, but at least it's not disabled. To turn on the light quicker, I have Up Double Press. Up Long press also works (because I'm using brightness_move_onoff and not just brightness_move)

The automation section:
```
    action_button_up_short:
      - service: script.turn_on_living_room_e2002
        data: {}
      - wait_for_trigger:
          - platform: mqtt
            topic: zigbee2mqtt/Tradfri Styrbar E2002 Comedor/action
            payload: arrow_right_hold
          - platform: mqtt
            topic: zigbee2mqtt/Tradfri Styrbar E2002 Comedor/action
            payload: arrow_left_hold
        continue_on_timeout: false
      - service: script.turn_off
        data: {}
        target:
          entity_id: script.turn_on_living_room_e2002
```
_Using [EPMatt 2002 Blueprint](https://github.com/EPMatt/awesome-ha-blueprints/blob/main/blueprints/controllers/ikea_e2001_e2002/ikea_e2001_e2002.yaml)_

The script:
```
alias: Turn on living room - E2002
sequence:
  - delay:
      hours: 0
      minutes: 0
      seconds: 1
      milliseconds: 500
  - service: light.turn_on
    target:
      entity_id: light.comedor
   
mode: restart
```
**Recommend to use Restart mode in the script!**

Note just in case: my remote isn't paired to a light. I run services or automations on the different buttons.

Hope this helps. Questions are welcome if I wasn't clear 😅 

CC @namadori

penroseg, 2022-06-17:
> @penroseg I went beyond that, while still keeping the up button working (even though not ideal, but better than disabled):
> 
> 1. Up button short click action triggers a Script that will turn on the light in 1.5 seconds.
> 2. In parallel, the automation waits for 1.5 seconds to see if arrow_right_hold or arrow_left_hold happens (that means the action_button_up_short was triggered by a left/right long press). 1,5 seconds works 98% of the time for me.
> 3. If 2 happens, then the script is stopped, allowing the left/right hold to happen without up_button happening too.
>    If 2 doesn't happen, it means button_up was triggered by actually pressing the up remote, so it does the expected action (light on in my case).
> 
> I hope I made sense. It's not great because up_button has a delay of 1.5 seconds, but at least it's not disabled. To turn on the light quicker, I have Up Double Press. Up Long press also works (because I'm using brightness_move_onoff and not just brightness_move)
> 
> The automation section:
> 
> ```
>     action_button_up_short:
>       - service: script.turn_on_living_room_e2002
>         data: {}
>       - wait_for_trigger:
>           - platform: mqtt
>             topic: zigbee2mqtt/Tradfri Styrbar E2002 Comedor/action
>             payload: arrow_right_hold
>           - platform: mqtt
>             topic: zigbee2mqtt/Tradfri Styrbar E2002 Comedor/action
>             payload: arrow_left_hold
>         continue_on_timeout: false
>       - service: script.turn_off
>         data: {}
>         target:
>           entity_id: script.turn_on_living_room_e2002
> ```
> 
> _Using [EPMatt 2002 Blueprint](https://github.com/EPMatt/awesome-ha-blueprints/blob/main/blueprints/controllers/ikea_e2001_e2002/ikea_e2001_e2002.yaml)_
> 
> The script:
> 
> ```
> alias: Turn on living room - E2002
> sequence:
>   - delay:
>       hours: 0
>       minutes: 0
>       seconds: 1
>       milliseconds: 500
>   - service: light.turn_on
>     target:
>       entity_id: light.comedor
>    
> mode: restart
> ```
> 
> **Recommend to use Restart mode in the script!**
> 
> Note just in case: my remote isn't paired to a light. I run services or automations on the different buttons.
> 
> Hope this helps. Questions are welcome if I wasn't clear 😅
> 
> CC @namadori

That's brilliant...!! I haven't really got my head around Scripts yet but will try this over the weekend and sense another HA rabbit hole will follow....

antonio1475, 2022-06-17:
> That's brilliant...!! I haven't really got my head around Scripts yet but will try this over the weekend and sense another HA rabbit hole will follow....

How I look at it: Scripts are automations that are (only) run (or called) manually instead of a trigger (automation). Or the other way around: an Automation is a Script that runs when something happens (trigger), other than me running/calling it.
The body (actions) are the same, although Automations can have a bit more complexity running If-Then or Choose based on the trigger that started it.

If you want to stick automations: my script above can be also made as an automation, that is run by the first automation (Styrbar) - since automations can also be manually called to run. Just put any trigger (eg. an event that won't happen) and call the automation from the Styrbar one. Scripts allow you to have the same, without forcing you to add a trigger.

I was a bit repetitive but hope it's clear! I had a hard time understanding the difference between Automations and Scripts when I started! 😅 

Could it be all done in the sequence within the Styrbar short up button action, without an extra script/automation? Maybe, but I'm unable to see how. :)

closed 2022-07-27 by github-actions[bot]

JackTalisker, 2022-07-30:
I have the same issue. arrow_left_hold and arrow_right_hold actions fire the "on" payload.

JackTalisker, 2022-07-30:
Affected users please join issue #13335

Maarten112, 2022-08-04:
I discovered the same issue yesterday. There's always an ON action payload before an ' arrow_left_hold' action payload. 

I'm not super familiair with how all zigbee things works in detail, but it seems like this is not a zigbee2mqtt issue but the way how the styrbar remote sents out their signals?

antonio1475, 2022-08-04:
> I discovered the same issue yesterday. There's always an ON action payload before an ' arrow_left_hold' action payload.
> 
> I'm not super familiair with how all zigbee things works in detail, but it seems like this is not a zigbee2mqtt issue but the way how the styrbar remote sents out their signals?

Yes. The received payload is what it is. The Styrbar unfortunately works like that as far as I can tell.

I believe unless IKEA changes it (unlikely?) or a custom firmware is created (is it possible?) not much can be done by Z2M. But of course I'm hopeful someone can find a solution!

Maarten112, 2022-08-04:
> Yes. The received payload is what it is. The Styrbar unfortunately works like that as far as I can tell.
>
> I believe unless IKEA changes it (unlikely?) or a custom firmware is created (is it [...]

Already thought so. To bad, since now we need to always filter this on payload in all our logic (giving delay for a regular on command). 

I didn't open the styrbar yet, so I don't know what uC they are using nor if we can manually reflash it with custom code. 

Too bad that they implemented the arrow behaviour this way, since I really like the look of the new vesions.

MattWestb, 2022-08-04:
Its having one original Silabs MG21 Light module and can being flashed with one real J-Link probe (many cheep clones is not working) and you can flash coordinator firmware on the module (EZSP  6.10.3) that i have cooking and its working great.
https://github.com/MattWestb/EFR32-FW/tree/main/Markus_EZSP.

Silabs SDK you find on the hut-hub https://github.com/SiliconLabs/gecko_sdk.

[image]

(hocked up and dumped main flash  and user data with Silabs WSTK)

gasecki, 2022-09-15:
Hi Guys, 
I lost my hope this issue will be resolved and I stopped to follow this thread. But, it seems that arrows long press and release is reported properly now. Same for the other round IKEA remote. I don’t know from which Z2M version. Please check in your installations.

antonio1475, 2022-09-15:
> Hi Guys, I lost my hope this issue will be resolved and I stopped to follow this thread. But, it seems that arrows long press and release is reported properly now. Same for the other round IKEA remote. I don’t know from which Z2M version. Please check in your installations.

You're right! It's actually explained and solved in https://github.com/Koenkk/zigbee2mqtt/issues/13335 , pushed on Aug 5

Didn0t really understand the how, but it works perfectly (other than left/right arrows long press taking 1.5 seconds to kick in, but it is what it is).

Thank you for pointing it here, as it had gone unnoticed for me. I now simplified my blueprint and everything is perfect - so much better than the convoluted workaround I had! :+1: