Koenkk/zigbee2mqtt #9878: Zigbee Model TS110E  Zigbee Manufacturer _TZ3210_ngqk6jia

Anatol6868, 2021-11-28:
## Information about the device + link

Device model ID: TS110E
Manufacturer name: _TZ3210_ngqk6jia
Aliexpress link: https://es.aliexpress.com/item/4001279149071.html?spm=a2g0s.9042311.0.0.2ea863c00bivCw

## data/database.db entry of the device
"id":7,"type":"Router","ieeeAddr":"0x00","nwkAddr":52247,"manufId":4417,"manufName":"_TZ3210_ngqk6jia","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[5,4,6,8,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS110E","manufacturerName":"_TZ3210_ngqk6jia","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"moesStartUpOnOff":1}},"genLevelCtrl":{"attributes":{"61440":78,"64515":10,"64516":1000}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1674843135},"lastSeen":1638119810879}  

 !!!!  This model does not work like https://user-images.githubusercontent.com/90213726/143667984-2a3bf402-0c53-44c9-8361-4afe2be027ed.jpg

Configured the operation of this device. In the file .src/zcl/definition/cluster.ts you need to add a new command for the 0008 cluster:
```
.........
            moveToLevelTuya: {
                ID: 240,
                parameters: [
                    { name: 'level', type: dataType_1.default.uint16 },
                    { name: 'transtime', type: dataType_1.default.uint16 },
                ],
            },
...........
```
Works with external converter :
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const DataType = {
    uint8: 0x20,
    uint16: 0x21,
};

const fzLocal = {
    ts110e_brightness: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('61440')) {
 //               return {brightness: utils.mapNumberRange(msg.data['61440'], 10, 1000, 1, 100, 0)};
                return {brightness: msg.data['61440']};
            }
        },
    },
};

const fzLoca = {
    brightness_min: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('64515')) {
                return {brightness_min: msg.data['64515']};
            }
        },
    },
};

const fzLoc = {
    brightness_max: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('64516')) {
                return {brightness_max: msg.data['64516']};
            }
        },
    },
};

const fzled = {
     led_type: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const lookup1 = {0: 'led', 1: 'incandescent', 2: 'halogen'};
            if (msg.data.hasOwnProperty('64514')) {
                const property = utils.postfixWithEndpointName('led_type', msg, model);
                return {[property]: lookup1[msg.data['64514']]};
            }
        },
    },
};

const tzLocal = {
ts110e_brightness: {
   key: ['brightness'],
      convertSet: async (entity, key, value, meta) => {
               let level = value;
               const payload1 = {level, transtime: 0}
               await entity.command('genOnOff', 1, {}, utils.getOptions(meta.mapped, entity));
               await utils.sleep(1000)
               await entity.command('genLevelCtrl', 'moveToLevelTuya', payload1, utils.getOptions(meta.mapped, entity));

     },
     convertGet: async (entity, key, meta) => {
            await entity.read('genLevelCtrl', [61440]);
     },
   },
};

const tzLoca = {
ts110e_brightness_min: {
   key: ['brightness_min'],
      convertSet: async (entity, key, value, meta) => {
            let payload = {64515: {'value': value, 'type': DataType.uint16}};
            await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));

     },
     convertGet: async (entity, key, meta) => {
           await entity.read('genLevelCtrl', [64515]);
     },
   },
};

const tzLoc = {
ts110e_brightness_max: {
   key: ['brightness_max'],
      convertSet: async (entity, key, value, meta) => {
            let payload = {64516: {'value': value, 'type': DataType.uint16}};
            await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));

     },
     convertGet: async (entity, key, meta) => {
           await entity.read('genLevelCtrl', [64516]);
     },
   },
};

const tzLo = {
ts110e_led_type: {
   key: ['led_type'],
      convertSet: async (entity, key, value, meta) => {
           value = value.toLowerCase();
           const lookup1 = {'led': 0, 'incandescent': 1, 'halogen': 2};
           newValue = parseInt(lookup1[value]);
           payload = {'64514': {'value': newValue, 'type': DataType.uint8}};
           await entity.write('genLevelCtrl', payload, utils.getOptions(meta.mapped, entity));

     },
     convertGet: async (entity, key, meta) => {
           await entity.read('genLevelCtrl', [64514]);
     },
   },
};

const definition = {
    fingerprint: [{ modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia' }],
    model: 'TS110E_dimmer',
    vendor: 'TuYa',
    description: 'Smart dimmer module with neutral',
    fromZigbee: [fz.on_off, fzLocal.ts110e_brightness, fz.moes_power_on_behavior, fzLoca.brightness_min, fzLoc.brightness_max, fzled.led_type],
    toZigbee: [tz.on_off, tz.moes_power_on_behavior, tzLocal.ts110e_brightness,tzLoca.ts110e_brightness_min,tzLoc.ts110e_brightness_max, tzLo.ts110e_led_type],
    exposes: [exposes.numeric('brightness', ea.ALL).withValueMin(10).withValueMax(1000).withDescription('Brightness of this light'),
              e.switch(),
              exposes.numeric('brightness_min', ea.ALL),
              exposes.numeric('brightness_max', ea.ALL),
              exposes.enum('led_type', ea.ALL, ['led', 'incandescent', 'halogen']).withDescription('Controls the type led'),
              exposes.enum('power_on_behavior', ea.ALL, ['off', 'previous', 'on']).withDescription('Controls the behaviour when the device is powered on')],
        configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
   },
};

module.exports = definition;

```                            
https://disk.yandex.by/d/nE7OD2tIewCOvA
key 1f50ddb684c75fbe6258eb9f269398bf

Koenkk, 2021-11-28:
Could you make a pull request for this?

Anatol6868, 2021-11-28:
What is it like ? Or what?

Anatol6868, 2021-11-28:
My English is only through a translator. )

pavax, 2021-11-29:
Thx @Anatol6868 I tried the mentioned converter, and I was able to set/get my light-switch, set/get the min and max brightness and read the current brightness (set by my external push-button). To set the brightness, I guess I'd need to be able to patch my cluster.js file within the home-assistant zigbee2mqtt addon which I haven't tried yet.

rsettembrini, 2021-11-29:
Thx @Anatol6868 it is exactly what I am searching for at the moment.
@pavax How did you execute docker command? I have homeassistant installed as OS on a raspberry pi4, but if I will try to execute docker command it says command not found. Also homeassistant system files are not visible

adekloet, 2021-11-29:
thanks @Anatol6868 for the work you have put into this one, much appreciated. I wait for it to be committed to Z2MQTT edge pr production and test it out on the devices I have here.

pavax, 2021-11-29:
> how did you execute docker command?

@rsettembrini  I'm not using the ha ssh addon as this won't give you privileges to the host. I instead followed the following guide to access my home-assitant host: https://developers.home-assistant.io/docs/operating-system/debugging/#ssh-access-to-the-host. After that you should be able to list all your running containers: `docker ps`

pavax, 2021-11-29:
After applying the converter and cluster.js patch the following entities are registered in Home-Assistant:

[image]

...unfortunately, it doesn't expose a light entity having the features state and brightness like this [familiar dimmer module:](https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-L.html). Is that by design?

rsettembrini, 2021-11-29:
> > how did you execute docker command?
> 
> @rsettembrini I'm not using the ha ssh addon as this won't give you privileges to the host. I instead followed the following guide to access my home-assitant host: https://developers.home-assistant.io/docs/operating-system/debugging/#ssh-access-to-the-host. After that you should be able to list all your running containers: `docker ps` and follow my guide. But be aware that every time the z2m-addons restarts your changes to the cluster.js are gone.

thx @pavax I did as you suggested and I am able to see core files.

Anatol6868, 2021-11-29:
The max and min values ​​are relevant for the physical switch.
Single press is on / off.
Long press to increase the brightness to the set max value,
Next long press decrease to the set min value.
Deliberately did not use utils.mapNumberRange, as it becomes necessary to adjust
Min / max values. Or do it in%, 1000 - 100%, 10 - 1%. I don't use HA.
Also thanks to @danieledwardgeorgehitchcock.

Anatol6868, 2021-11-29:
[image]

pavax, 2021-11-29:
> Min / max values. Or do it in%, 1000 - 100%, 10 - 1%. I don't use HA.

It's exactly what my converter does. Except that it guarantees, that the brightness won't get below the brightness_min / max values. So, it behaves like the physical switch. The converter makes the device identically to [the exsting Lonsonho 1 gang smart dimmer](https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-L.html#light). You can still adjust the brightness and the light states my sending the proper mqtt messages as it combines the two in one topic.

Z2M:
[image]

MQTT:
[image]

adekloet, 2022-01-01:
Hi,

Was the converter added to the Z2MQTT supported devices? 

Thanks for all the work on this one.
Happy newyear

half-monkey, 2022-01-06:
Hi,

Thanks to everyone above who has contributed to this!
I have some of these dimmers, so have been tinkering...
@pavax the code you've listed mostly works for me (including Home Assistant integration) - thank you!
I added some logic to prevent out-of range dimming values being reported (e.g. when dimmer reports "0" when off:
```
      if (msg.data.hasOwnProperty("61440")) {
        let minBrightness = meta.state.hasOwnProperty("brightness_min")
          ? meta.state.brightness_min
          : 0;
        let maxBrightness = meta.state.hasOwnProperty("brightness_max")
          ? meta.state.brightness_max
          : 1000;
        let level = utils.mapNumberRange(
          msg.data["61440"],
          minBrightness,
          maxBrightness,
          0,
          255
        );
	if (level < 0) {
		level = 0;
	}
	if (level > 255) {
		level = 255;
	}
```

which stopped some of the negative values I was getting in the GUI under some circumstances. I do find that the level tends to jump slightly after being set, but I'm assuming that's due to rounding errors in the range translation, so not a problem as it's typically only by 1 or 2 increments.

I still have one issue though: When I try to refresh the dimming value, it always comes back as 0, but when I actually change it, it reports back correctly... Anybody else experiencing similar? The one difference I can see is that the two produce slightly different messages in the log:
_debug 2022-01-06 22:54:41: Received Zigbee message from 'Hallway_Light', type 'readResponse', cluster 'genLevelCtrl', data '{"61440":0}' from endpoint 1 with groupID 0_ <-- not working (dimmer was set to full brightness, so should report 1000 (or close to that))
_debug 2022-01-06 22:55:08: Received Zigbee message from 'Hallway_Light', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":997}' from endpoint 1 with groupID 0_ <-- working

I don't know enough to understand the difference between a "readResponse" and an "attributeReport" (I guess one is transmitted whenever the unit feels like it, but the other happens when we query it?), but guessing that's the problem. Is that something that could be fixed here, or a bug in the dimmer itself? If I generally avoid the value refresh button, it seems to work.

MrTheFirst, 2022-01-24:
I bought such a device (1 gang no natural)
Device model ID: TS110E
Manufacturer name: _TZ3210_weaqkhab (not _TZ3210_ngqk6jia)
Aliexpress link: https://es.aliexpress.com/item/4001279149071.html?spm=a2g0s.9042311.0.0.2ea863c00bivCw

zigbee2mqtt does not support it. I use the converter described above. The device works strangely, can't get attributes.

brightness

> Publish 'get' 'brightness' to 'Switch Dimmer' failed: 'Error: Read 0xa4c13814d311a0af/1 genLevelCtrl([61440], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

led_type get
> Publish 'get' 'led_type' to 'Switch Dimmer' failed: 'Error: Read 0xa4c13814d311a0af/1 genLevelCtrl([64514], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

led_type set
> Publish 'set' 'led_type' to 'Switch Dimmer' failed: 'Error: Write 0xa4c13814d311a0af/1 genLevelCtrl({"64514":{"value":0,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

power_on_behavior get

> Publish 'get' 'power_on_behavior' to 'Switch Dimmer' failed: 'Error: Read 0xa4c13814d311a0af/1 genOnOff(["moesStartUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

power_on_behavior set
> Publish 'set' 'power_on_behavior' to 'Switch Dimmer' failed: 'Error: Write 0xa4c13814d311a0af/1 genOnOff({"moesStartUpOnOff":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

how can i fix this?

Imaginous, 2022-02-08:
I also have received this device, it's not completely clear to me what the current status is.

Will it be included in the next version of Zigbee2mqtt?

MHerus, 2022-02-13:
I'm very disappointed..

[url=https://postimg.cc/MvKYkHJm][img]https://i.postimg.cc/MvKYkHJm/Zrzut-ekranu-2022-02-13-123338.png[/img][/url]

RealZack, 2022-02-24:
Hey,

I'm using the converter from @pavax (thank you for that) and it's working fine.

Could it be integrated natively ?

DesertF0x, 2022-03-16:
With the code @pavax posted I have the problem that the device reports 0 brightness when reading the brightness even though it is not zero:
setting brightness
```shell
debug 2022-03-16 10:12:26: Received MQTT message on 'zigbee2mqtt/LichtEGFlur/set' with data '{"brightness":119}'
debug 2022-03-16 10:12:26: Publishing 'set' 'brightness' to 'LichtEGFlur'
debug 2022-03-16 10:12:26: TS110E: TZ-Brightness level changed to: 533
debug 2022-03-16 10:12:26: Received Zigbee message from 'LichtEGFlur', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-03-16 10:12:26: MQTT publish: topic 'zigbee2mqtt/LichtEGFlur', payload '{"brightness":-153,"brightness_max":800,"brightness_min":300,"led_type":"led","linkquality":72,"power_on_behavior":"previous","state":"ON"}'
debug 2022-03-16 10:12:26: Received Zigbee message from 'LichtEGFlur', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":533}' from endpoint 1 with groupID 0
debug 2022-03-16 10:12:26: TS110E: FZ-Brightness level changed to: 119
info  2022-03-16 10:12:26: MQTT publish: topic 'zigbee2mqtt/LichtEGFlur', payload '{"brightness":119,"brightness_max":800,"brightness_min":300,"led_type":"led","linkquality":72,"power_on_behavior":"previous","state":"ON"}'
debug 2022-03-16 10:12:27: Received Zigbee message from 'LichtEGFlur', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":529}' from endpoint 1 with groupID 0
debug 2022-03-16 10:12:27: TS110E: FZ-Brightness level changed to: 117
info  2022-03-16 10:12:27: MQTT publish: topic 'zigbee2mqtt/LichtEGFlur', payload '{"brightness":117,"brightness_max":800,"brightness_min":300,"led_type":"led","linkquality":72,"power_on_behavior":"previous","state":"ON"}'
debug 2022-03-16 10:12:32: Received Zigbee message from 'LichtEGFlur', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2022-03-16 10:12:32: Received Zigbee message from 'LichtKuecheninsel', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":49,"imageType":2031,"manufacturerCode":4107}' from endpoint 1 with groupID 0
```
reading brightness
```shell
debug 2022-03-16 10:14:16: Received MQTT message on 'zigbee2mqtt/LichtEGFlur/get' with data '{"brightness":""}'
debug 2022-03-16 10:14:16: Publishing get 'get' 'brightness' to 'LichtEGFlur'
debug 2022-03-16 10:14:16: Received Zigbee message from 'LichtEGFlur', type 'readResponse', cluster 'genLevelCtrl', data '{"61440":0}' from endpoint 1 with groupID 0
debug 2022-03-16 10:14:16: TS110E: FZ-Brightness level changed to: -153
info  2022-03-16 10:14:16: MQTT publish: topic 'zigbee2mqtt/LichtEGFlur', payload '{"brightness":-153,"brightness_max":800,"brightness_min":300,"led_type":"led","linkquality":65,"power_on_behavior":"previous","state":"ON"}'
```

SkyByte, 2022-03-24:
Be aware that there are several Manufacturer names for the same model. I bought a device with following data. Unfortunately it is unsupported at all, you even can not switch on and off.

Device model ID: TS110E
Manufacturer name: _TZ3210_**weaqkhab**
Aliexpress link: https://es.aliexpress.com/item/1005002821229566.html?gatewayAdapt=glo2esp&spm=a2g0o.order_list.0.0.21ef194d0Z9tRJ

The connectors are not "L-L1-L1-com-S1-S1", but "L-N-L1-L1-com-S", but it is Without Neutral. You have to connect a bridge between N and L1.

So if this device will be adapted, please include this variant, too. Thanks.

DesertF0x, 2022-03-24:
Have you tried it with the config posted in this issue? Setup like [here](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_1-pairing-the-device-with-zigbee2mqtt).

SkyByte, 2022-04-03:
Today i tried the code posted by @pavax, i only changed the manufacturerName to "_TZ3210_weaqkhab".
The code is working as described above, also with the issue returning zero brightness when reading the brightness even though it is not zero. But this would be okay for me, because i only want be able to set the brightness level, not to read it.
Nevertheless I would appreciate the fully functional integration of this device.

DesertF0x, 2022-04-06:
Reading the devices is most likely not supported. This seems to be usual for Tuya devices.

AcidSleeper, 2022-04-10:
Hey guys.
I have tried to do this but I cant understand where to find and edit "cluster.ts"

Im running Home Assistant and Z2M as an addon in HA.
I have created a file "ts110e.js" and placed this in the same folder as configuration.yaml (zigbee2mqtt configfile) and pasted in the code above. 
Also in Zigbee2mqtt´s settings I have external_converters: - ts110e.js

But where is cluster.ts?

DesertF0x, 2022-04-21:
As far as  I remember the cluster.ts modification is already in and you do not need to do it anymore.

AcidSleeper, 2022-04-21:
> As far as I remember the cluster.ts modification is already in and you do not need to do it anymore.

Sorry to say it dont work.

external converters settings: should this be in z2m settings or in HA settings for z2m?

RealZack, 2022-04-21:
you only have to create TS110E.js and put it into /config/zigbee2mqtt

then inside your configuration.yml still in /config/zigbee2mqtt you have to add : "external_converters": [TS110E.js]

AcidSleeper, 2022-04-24:
> you only have to create TS110E.js and put it into /config/zigbee2mqtt
> 
> then inside your configuration.yml still in /config/zigbee2mqtt you have to add : "external_converters": [TS110E.js]

Did that but still says "Unsupported" and cant handle on/off or dim.

closed 2022-06-01 by github-actions[bot]