Koenkk/zigbee2mqtt #9883: Aqara switch 2 gang QBKG24LM cannot be used to pair other devices

iqbalaydrus, 2021-11-29:
## What happened
I tried to join aqara pir and door sensors using the permit join via specific router on the frontend and it didn't work. Tried with a few of those. But pairing worked with aqara 3 gang switches (QBKG26LM).

## What did you expect to happen
I expect pairing should work with any router devices.

## How to reproduce it (minimal and precise)
Join new device via QBKG24LM switch.

## Debug info
Zigbee2MQTT version: 1.22.0 docker image
Adapter hardware: zzh
Adapter firmware version: 20210708

insipiens, 2021-11-29:
Surely an issue for the manufacturer? Z2m doesn’t handle pairing.

iqbalaydrus, 2021-11-29:
I'm not sure. Are you saying that all QBKG24LM switches might not support pairing through it? Can we verify that somehow?

closed 2022-01-06 by github-actions[bot]