Koenkk/zigbee2mqtt #9887: Sensor Aqara temperature WSDCGQ11LM frozen after some hours

dattardi1, 2021-11-29:
After some hours temperature sensor WSDCGQ11LM stop to updated value.

I need to repair the sensor to back at normal fucntionality, but however after some hours it will froze again

## Debug info
Zigbee2MQTT version:  1.22.0-3
Adapter hardware: Conbee II
Adapter firmware version: deCONZ_ConBeeII_0x26720700.bin.GCF

Tonio16, 2021-11-30:
Hello

May be linked also to this issue:
https://github.com/Koenkk/zigbee2mqtt/issues/9752

Personnaly, old devices are recognized but lost after a few hours. New one stay as unrecognized. 

Antoine

Flurkmark, 2021-12-12:
Have you had it working before?
I have had 6 of them working perfectly for over a year. Now when the batteries die and I replace, they will only connect to the coordinator. Before they connected to a router. All routers are IKEA bulbs/panels. 

One of them also showed your behavior. After a remove, restart of z2m, re-join it works, but only to coordinator. I am uncertain if this is a IKEA firmware problem (have updated during the year) or something with zigbee2mqtt.

closed 2022-01-19 by github-actions[bot]

MrMiracles, 2022-12-18:
Exactly the same problem, battery died. Installed a new one.
Did not report anything at first, so I tried resetting and rejoining the sensor. It worked for a couple of hours now it's stuck again.

Looking at the deconz software it is indeed only connecting to the coordinator, it has a good connection but it is somehow not updating. Even if I press the button on the sensor (that should send an update) it won't update the values.

Did you or someone had any luck resolving this?