Koenkk/zigbee2mqtt #9890: TuYa TS0601_thermostat_1 apparently does not have modes defined

losip, 2021-11-29:
## What happened
Thermostat based on this radiator valve has all modes displayed, even inappropriate ones such as 'cool'

## What did you expect to happen
Need to see just Off, Heat and Auto like its brother TuYa TS0601_thermostat. Modes definiton missing?

## How to reproduce it (minimal and precise)
Create thermostat and see six modes displayed

## Debug info
Zigbee2MQTT version: Zigbee2mqtt Edge
Adapter hardware: CC26X2R1
Adapter firmware version:
[image]

vladi1234, 2021-11-29:
Look here:

https://github.com/Koenkk/zigbee-herdsman-converters/issues/2937#issuecomment-917696264

vladi1234, 2021-11-29:
I think that doesn't leave any point if it works.
You can program with extra code so that HA is correct as the default, but that makes problems with other home automation programs. It doesn't make TVR as flexible as it should be.

losip, 2021-11-29:
Thanks for the advice

closed 2021-11-29 by losip