Koenkk/zigbee2mqtt #9896: IKEA E1812 (Shortcut button) and IEKA E1743 (On/Off button) can not pair

atmel14, 2021-11-29:
Hello.
I have bought a IKEA E1812 (Shortcut button) and IEKA E1743 (On/Off button).
I have use CC2531 as coordinator and pairing work fine. Button works aslo fine. No Problems.

Now I want to pair the buttons to my friend network, but they use "zzh Multiprotocol RF Stick (CC2652R1) as coordinator.
Here I could not pair the device.
Interview was started but after some time the device left the network. by the next try I got an error. It is reproducible!
Please see log:
`Zigbee2MQTT:info  2021-11-29 18:19:31: Device '0x0c4314fffeab04fe' joined
Zigbee2MQTT:info  2021-11-29 18:19:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffeab04fe","ieee_address":"0x0c4314fffeab04fe"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-11-29 18:19:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0c4314fffeab04fe"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-11-29 18:19:31: Starting interview of '0x0c4314fffeab04fe'
Zigbee2MQTT:info  2021-11-29 18:19:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffeab04fe","ieee_address":"0x0c4314fffeab04fe","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-11-29 18:19:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0c4314fffeab04fe"},"type":"pairing"}'
Zigbee2MQTT:info  2021-11-29 18:19:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffeab04fe","ieee_address":"0x0c4314fffeab04fe"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-11-29 18:19:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0c4314fffeab04fe"},"type":"device_announced"}'
Zigbee2MQTT:warn  2021-11-29 18:19:41: Device '0x0c4314fffeab04fe' left the network
Zigbee2MQTT:info  2021-11-29 18:19:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffeab04fe","ieee_address":"0x0c4314fffeab04fe"},"type":"device_leave"}'
Zigbee2MQTT:info  2021-11-29 18:19:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x0c4314fffeab04fe"},"type":"device_removed"}'
Zigbee2MQTT:info  2021-11-29 18:20:42: Device '0x0c4314fffeab04fe' joined
Zigbee2MQTT:info  2021-11-29 18:20:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffeab04fe","ieee_address":"0x0c4314fffeab04fe"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-11-29 18:20:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0c4314fffeab04fe"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-11-29 18:20:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffeab04fe","ieee_address":"0x0c4314fffeab04fe"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-11-29 18:20:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0c4314fffeab04fe"},"type":"device_announced"}'
Zigbee2MQTT:warn  2021-11-29 18:20:53: Device '0x0c4314fffeab04fe' left the network
Zigbee2MQTT:info  2021-11-29 18:20:53: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffeab04fe","ieee_address":"0x0c4314fffeab04fe"},"type":"device_leave"}'
Zigbee2MQTT:info  2021-11-29 18:20:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x0c4314fffeab04fe"},"type":"device_removed"}'

C:\Apps\zigbee2mqtt-1.22.0\lib\zigbee.ts:147
        const name = data.device.name;
                                 ^
TypeError: Cannot read property 'name' of undefined
    at Zigbee.logDeviceInterview (C:\Apps\zigbee2mqtt-1.22.0\lib\zigbee.ts:147:34)
    at Controller.<anonymous> (C:\Apps\zigbee2mqtt-1.22.0\lib\zigbee.ts:84:18)
    at Controller.emit (events.js:315:20)
    at Controller.onDeviceJoined (C:\Apps\zigbee2mqtt-1.22.0\node_modules\zigbee-herdsman\src\controller\controller.ts:563:20)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.22.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.22.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.`

Have you seen it before? Could you help me!
Thanks in advance.

atmel14, 2021-11-29:
Hello.
What I have seen it that this error occure only when the device is very close to the coordinator (<10cm).
When the distannce is bigger (like 1m) pairing is succesfully.
It cloud be relate to this #8670

misternerd, 2021-12-14:
I can confirm both the error as well as the weird solution. Thanks @atmel14 for the 1m hint, that really helped me.
My errors on Linux using slae.sh's coordinator are pretty similar. I get a few join/announce/leave cycles (left out for brevity) before the error occurs.
```
Zigbee2MQTT:info  2021-12-14 21:46:50: Device 'IKEA Tradfri OnOff 01' joined                                                                                   
Zigbee2MQTT:info  2021-12-14 21:46:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"IKEA Tradfri OnOff 01","ieee_address":"0x680ae2fffe0c8a16"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-12-14 21:46:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"IKEA Tradfri OnOff 01","ieee_address":"0x680ae2fffe0c8a16"},"type":"device_announce"}'
Zigbee2MQTT:warn  2021-12-14 21:47:01: Device 'IKEA Tradfri OnOff 01' left the network                                                                         
Zigbee2MQTT:info  2021-12-14 21:47:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"IKEA Tradfri OnOff 01","ieee_address":"0x680ae2fffe0c8a16"},"type":"device_leave"}'
TypeError: Cannot read property 'name' of undefined                                                                                                            
    at Zigbee.logDeviceInterview (/app/lib/zigbee.ts:144:34)                                                                                                   
    at Controller.<anonymous> (/app/lib/zigbee.ts:84:18)                                                                                                       
    at Controller.emit (events.js:400:28)                                                                                                                      
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:531:22)
```
Also, I'd like to point out that it crashes the whole app. Please let me know if I can provide additional info to help fix the issue.

atmel14, 2022-01-03:
Could not be repreoduce with Release Version 1.22.2. Tested on CC2531 and CC2552R.
Therefor item is closed.

closed 2022-01-03 by atmel14