Koenkk/zigbee2mqtt #9898: battery value not correct

timoline, 2021-11-29:
https://www.zigbee2mqtt.io/devices/ICZB-RM11S.html#icasa-iczb-rm11s

just added a few new batteries in this device but always shows 50%

If I put the battery in another devoce it shows 100%

sjorge, 2021-11-29:
I’m guessing it might need the flag to tell it not to device the value, or if it’s using voltage it might je tagged wrong https://github.com/Koenkk/zigbee-herdsman-converters/blame/master/README.md#L32

Koenkk, 2021-11-30:
Should be fixed now, please confirm.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

timoline, 2021-11-30:
just pulled the dev branch and unfortunatly still 50%

Koenkk, 2021-11-30:
Make sure you are on #c90171d2cc3022f88987cf1e9be978bd7fa7ecd8 and wait until the next battery report comes in (could take a day)

timoline, 2021-11-30:
ok good point, I'll let you know

timoline, 2021-12-01:
looking good now, thx

[image]

closed 2021-12-02 by Koenkk