Koenkk/zigbee2mqtt #9899: Stability of the Network - have to restart every few days

Jens-Wymeersch, 2021-11-29:
## What happened
Basically I've restarted Z2M on Saturday. Made sure the whole network is covered LQI. And today Tuesday, I need to restart my network again as I'm unable to control my devices.

## What did you expect to happen
Once setup, I wouldn't have to reset it every few days, weeks.

## How to reproduce it (minimal and precise)

I get the following error message as an example.

`Publish 'set' 'state' to ' Lamp Lower Stairs' failed: 'Error: Command 0x0017880106a86e31/11 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'`

## Debug info
`Zigbee2MQTT:error 2021-11-29 23:16:29: Publish 'set' 'state' to 'Lamp upper stairs' failed: 'Error: Command 0x0017880106a86e16/11 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
Zigbee2MQTT:info  2021-11-29 23:16:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Lamp upper stairs' failed: 'Error: Command 0x0017880106a86e16/11 genOnOff.on({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"Lamp upper stairs"},"type":"zigbee_publish_error"}'
2021-11-29T21:16:35.742Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":18222,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":83,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,103,0,5,5,8,5,11,5]}}
2021-11-29T21:16:35.743Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,46,71,1,1,4,11,83,0,30,9,16,103,0,5,5,8,5,11,5,96]
2021-11-29T21:16:35.747Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x0017880106a86e31:49895/11 (0,0,859)
2021-11-29T21:16:35.748Z zigbee-herdsman:controller:endpoint Command 0x0017880106a86e31/11 genOnOff.on({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)`

Zigbee2MQTT version: 1.22.0 commit: unknown
Adapter hardware: CC2652R (ZZH)
Adapter firmware version:
Coordinator type zStack3x0
Coordinator revision 20210708
Frontend version 0.6.30

Jens-Wymeersch, 2021-11-29:
Please also note that if I look to my Mosquitto Broker, it doesn't make any connection.

thiscantbeserious, 2021-11-30:
Try to throw out the Zigbee Lamps and see if it makes a difference. 

They act as a router, but whenever I added those I would have nothing but troubles with my whole network freezing up (with them naturally running super-hot ...). 

If that solves your issue then I'd throw in a set of dedicated routers and try again. 

Best would be 2652RB based like Slae.sh stick for now since the 2652P based ones with amplifiers currently dont seem to work properly as a router (pairs but never gets a client connected, so maybe they're unsupported or bugged ... which I didn't have time to look into further - but maybe that's just an issue on my side).

If that still doesn't help put them in a seperate Zigbee network, switch them over to Wifi/Bluetooth or use regular lamps with a Smart Zigbee-(Dimmer)-Switch instead.

Another possibility could be that you're using a RaspberryPi and the power that the new Sticks require are too much for it to handle properly. Use an active USB-Hub (had to do that too) to solve that part, but be carefull about it being protected from running the Power back into the Host port, unless you want to damage your RaspberryPi - there's a lot of shit Hubs in the market for that saving a few cents in return).

Jens-Wymeersch, 2021-11-30:
I have 150 zigbee devices in my house. At least half of them are routers, ranging for lights, switches, curtain covers, power plugs, etc... 
I'm running Z2M on an HA instance on RPI 3 linked to UPS. Purpose is Z2M, Room Assistant and NUT. Nothing else. 

For lamps, I have a few cheaper lamps but most are HUE or IKEA. 
@thiscantbeserious you really think this is the problem that nothing works any more and it doesn't even make a connection to MQTT ? A reboot of the Z2M app solves it again till the next time.

Koenkk, 2021-11-30:
Please try the fw from https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0 , this has an increased memory heap which could prevent crashes with larger networks.

lux73, 2021-12-01:
try to remove all IKEA Lamps - my Brother has 1 (ONE) IKEA Lamp integrated which has crashed the whole Zigbee Network from timt to time

now he has removed this Lamp (replaced with HUE) and his System is much more stable - he could not imagine that this single Lamp would cause so much trouble

me personally has only HUE Lamps & Dresden Electronic LED/RGB Controllers - not a single glitch

fegyosz, 2021-12-14:
I can confirm similar issues caused by Ikea plugs and bulbs, routers. I expected them to make my mesh stronger but in reality they weakened it. Just removed all yesterday. I hope they just need more delicate handling and these issues will be resolved.
I could see them failing causing issues for end devices, unplug replug resolved the issue, so it seems the device itself is failing.
This appears to be the same issue: https://github.com/Koenkk/zigbee2mqtt/issues/10074

awalsum, 2021-12-19:
I'm getting exactly the same error as @Jens-Wymeersch but with a Tuya TS0121_plug. I have a number of these plugs in the network, but it's just this one plug that seems to be causing trouble. I'm going to disable it and see if that resolves my problems. I'm having to restart my network multiple times a week. Some days even twice a day, which is very annoying. A smart home is supposed to make life easier, but it's not when you have to play fulltime administrator to keep things running.

awalsum, 2022-01-13:
The plot thickens. As I was constantly having trouble with just one of the smart plugs, I decided to remove it from the network and paired a brand new plug, a Tuya 011s.

The strange thing is, now this new plug, which is powering the same device (a servomotor on a 3-way valve) from the same 230V outlet, is causing the exact same problem. When the problems started occurring, it wasn't the last device to be added (so not evidently an overload issue) and now with the new plug, it also isn't the last one to be added as I have added two others since this one and they don't give me any problems. The error I'm getting is:

2022-01-13 10:05:41: Publish 'set' 'state' to 'Omschakelen boiler' failed: 'Error: Command 0x540f57fffe684463/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

Jens-Wymeersch, 2022-01-17:
@Koenkk 
- I've moved to the new firmware as you recommended 
- I've removed all non-HUE bulbs from my network
Still, the problem remains and i'll need to restart Z2M. Any other ideas ?

`2022-01-17T16:40:05.167Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x60a423fffe652cc8:1164/1 (0,0,576)
2022-01-17T16:40:05.168Z zigbee-herdsman:controller:endpoint Read 0x60a423fffe652cc8/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
2022-01-17T16:40:11.169Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":39886,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":163,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,242,0,5,5,8,5,11,5]}}
2022-01-17T16:40:11.170Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,206,155,1,1,4,11,163,0,30,9,16,242,0,5,5,8,5,11,5,57]
2022-01-17T16:40:11.173Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x60a423fffe621007:63162/1 (0,0,575)
2022-01-17T16:40:11.174Z zigbee-herdsman:controller:endpoint Read 0x60a423fffe621007/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)`

Koenkk, 2022-01-17:
@Jens-Wymeersch given that you are running on a RPI 3, do you have any other USB devices connected to it?

Jens-Wymeersch, 2022-01-17:
@Koenkk  As this is a critical component of my network, I've connected the RPI to a UPS which is linked to the RPI (NUT). Would a migration to an RPI 4 solve the problem ? 
If so, how would a migration look like ? Do I need to start from scratch or I can copy and past my existing configs ?

Koenkk, 2022-01-17:
but do you have eg an ssd connected via usb?

Jens-Wymeersch, 2022-01-17:
@Koenkk No, I don't have an SSD connected via USB.  
I have a SD card on the RPI on which I'm running HA for the sole purpose of running Z2M (and room assistant).

Koenkk, 2022-01-17:
@Jens-Wymeersch is anything logged to syslog (`cat /var/log/syslog`) at the time z2m crashes (first time you see a `SRSP - AF - dataRequest after 6000ms`)?

Jens-Wymeersch, 2022-01-17:
@Koenkk I'm using the HA version of the product. How would I be able to get to it. By the way, i've already restarted the service.

Koenkk, 2022-01-18:
But is anything logged to `cat /var/log/syslog` at the moment z2m starts to fail?

Jens-Wymeersch, 2022-01-18:
@Koenkk sorry for my ignorance, but how would I be able to find this in Home Assistant ?

Koenkk, 2022-01-18:
@Jens-Wymeersch login to your system via ssh and execute `cat /var/log/syslog`

Jens-Wymeersch, 2022-01-18:
@Koenkk I went to the shell of Home Assistant and executed the command without success. 
[image]

The only directory I find log data is related to the event /config/zigbee2MQTT/log/2022-01-17.18-49-40/log.txt
I don't know what to do more

Jens-Wymeersch, 2022-01-20:
@Koenkk it looks like you can't get from the HA addon (check with the community). Z2M addon went down again today.

Jens-Wymeersch, 2022-01-20:
@Koenkk would a move to RPI4 help my problem in your opinion ?

Koenkk, 2022-01-20:
@Jens-Wymeersch I don't think that will help. Also; are you using a good power supply for your Pi? And is the adapter connected through a usb extension cable?

Jens-Wymeersch, 2022-02-12:
@Koenkk Thank you for the hints. I've changed the power supply for the Pi and till it's been running smoothly.

c-soft, 2022-02-15:
I have very similar issue: every couple of days it stops working - no device is working. I need to restart addon to get the network back again. 

Error in the logs:

```
Zigbee2MQTT:error 2022-02-14 15:00:06: Publish 'set' 'state' to 'hol_wtycz_kamera' failed: 'Error: Command 0xa4c1382c7822f11a/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
```
My setup:
* Raspberry PI 4
* Good quality, powered USB hub (7 port)
* Sonoff 3.0 USB Dongle on ~1m USB cable from the USB hub (just to be sure, I had the same error before when being close Raspi)

Any other hints? Can the issue be related to particular device types in the network?

closed 2022-03-26 by github-actions[bot]