Koenkk/zigbee2mqtt #9900: GL-SD-002 Gledopto Zigbee Switch module

BeardedTinker, 2021-11-29:
## Information about the device + link
Zigbee 3.0 Switch module with ability to be controlled by wall switch - No Neutral wire required
https://www.gledopto.com/h-col-391.html

## data/database.db entry of the device
..
```
{"id":55,"type":"EndDevice","ieeeAddr":"0xcc86ecfffebb3ae1","nwkAddr":45872,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-SD-002","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-SD-002","manufacturerName":"GLEDOPTO","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20210830","swBuildId":"10276801"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00214f1d31","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20210830","swBuildId":"10276801","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1638218029320}
```

Created following custom converter that works:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['GL-SD-002'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'GL-SD-002', // Vendor model number, look on the device for a model number
    vendor: 'GLEDOPTO', // Vendor of the device (only used for documentation and startup logging)
    description: 'GELDOPTO Zigbee 3.0 Smart Home Switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```
Device can now be controlled (ON/OFF), but looks that there is problem with reporting state - maybe the same thing as with other device from same manufacturer (GL-W-001Z requires reading on/off state) - but not sure how to do it.

pedrolamas, 2021-12-03:
@BeardedTinker you should be able to replace these lines:

```javascript
fromZigbee: [fz.on_off],
toZigbee: [tz.on_off],
exposes: [e.switch()],
```

with just this:

```javascript
extend: extend.switch(),
```

As for the reporting, your code looks correct to me, so I assume that this device does suffer from the same issue as Gledopto GL-W-001Z.

I assume you've checked that the z2m logs in debug mode don't show anything when changing the on/off state directly on the physical button?

BeardedTinker, 2021-12-03:
That's right - nothing else was present in the logs. Was thinking on implementing same thing from GL-W-001Z, but had no idea on how to add this to custom converter...

closed 2022-01-10 by github-actions[bot]