Koenkk/zigbee2mqtt #9906: IKEA E1812 (Shortcut button) after firmware update to 2.3.080 does not send actions

atmel14, 2021-11-30:
Hello.

I had to IKEA E1812 (Shortcut button) with firmeware 2.3.015.
Both works fine and the action was receviecd from both devices.

Now I have updated one of them to firmeware 2.3.080.
The device joined sucesfully, but no action is receving by pushing the button.

Joining the device again show an error:
`Zigbee2MQTT:warn  2021-11-30 13:17:02: Device '0x04cd15fffe2d3b70' left the network
Zigbee2MQTT:info  2021-11-30 13:17:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe2d3b70","ieee_address":"0x04cd15fffe2d3b70"},"type":"device_leave"}'
Zigbee2MQTT:info  2021-11-30 13:17:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x04cd15fffe2d3b70"},"type":"device_removed"}'
Zigbee2MQTT:info  2021-11-30 13:17:08: Device '0x04cd15fffe2d3b70' joined
Zigbee2MQTT:info  2021-11-30 13:17:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe2d3b70","ieee_address":"0x04cd15fffe2d3b70"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-11-30 13:17:08: Configuring '0x04cd15fffe2d3b70'
Zigbee2MQTT:info  2021-11-30 13:17:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cd15fffe2d3b70"},"type":"device_connected"}'
Zigbee2MQTT:debug 2021-11-30 13:17:10: Device '0x04cd15fffe2d3b70' announced itself
Zigbee2MQTT:info  2021-11-30 13:17:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe2d3b70","ieee_address":"0x04cd15fffe2d3b70"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-11-30 13:17:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cd15fffe2d3b70"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-11-30 13:17:12: Retrieving state of '0x04cd15fffe2d3b70' after reconnect
Zigbee2MQTT:error 2021-11-30 13:17:18: Failed to configure '0x04cd15fffe2d3b70', attempt 1 (Error: Bind 0x04cd15fffe2d3b70/1 genPowerCfg from '901' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (C:\Apps\zigbee2mqtt-1.22.0\node_modules\zigbee-herdsman\src\utils\waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
Zigbee2MQTT:info  2021-11-30 13:17:20: Configuring '0x04cd15fffe2d3b70'
Zigbee2MQTT:debug 2021-11-30 13:17:28: Received Zigbee message from '0x04cd15fffe2d3b70', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":60}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-11-30 13:17:28: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe2d3b70', payload '{"battery":60,"elapsed":425136,"linkquality":123}'
Zigbee2MQTT:info  2021-11-30 13:17:28: Successfully configured '0x04cd15fffe2d3b70'`

atmel14, 2021-11-30:
The problem has already a workaround which is already mentioned in this issue some comments above.
#9439 (comment)

closed 2021-11-30 by atmel14

kornelremete, 2021-12-15:
same problem for me