Koenkk/zigbee2mqtt #9907: Can't disable "new availability feature"

LeoCal, 2021-11-30:
## What happened
On the latest version of zigbee2mqtt (1.22.0-8-gdf0543d4), it seems that the "new availability feature" is enabled by default (without any explicit config for that). As I've read this might be the possible cause of the issues I'm seeing, I've tried to disable it by adding "availability: false" in the configuration.yaml. I've tried both to put it directly under the "permit_join:" config and as a sub-item of the "advanced:". Despite this, I always see the following warning when starting zigbee2mqtt:

`Using experimental new availability feature`

## What did you expect to happen
I expected the new availability feature to be disabled.

## How to reproduce it (minimal and precise)
Simply start zigbee2mqtt and inspect the boot logs.

## Debug info
Zigbee2MQTT version: 1.22.0-8-gdf0543d4
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20201127

LeoCal, 2021-11-30:
mmm, actually, I've tried to enable it via a "availability: true" and I see that some more "availability" in the logs.

I was wondering, is it possible that the warning "Using experimental new availability feature" is printed to the logs regardless of whether the feature is actually enabled or not?

amccre, 2021-12-02:
Seeing the same thing:

Configuration...

frontend:
  port: 8099
experimental: {}
availability: false

Log...

Zigbee2MQTT:warn  2021-12-02 11:41:52: Using experimental new availability feature

Koenkk, 2021-12-03:
should be fixed in z2m 1.22.1

Krystm, 2021-12-14:
Doesn't seem to be as my config is marked false as well. 
[21:59:03] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.22.1 start
> node index.js
Zigbee2MQTT:warn  2021-12-13 21:59:06: Using experimental new availability feature

Koenkk, 2021-12-14:
Whoops, was indeed not fixed, it is now. Note that it is only cosmetically since the availability feature will be disabled when it is not in your config.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-21 by github-actions[bot]