Koenkk/zigbee2mqtt #9918: Firmware data not updated after both re-join / remove/join

Bascht74, 2021-11-30:
## What happened
I updated a hue dimmer with my hue bridge. For that I joined the device to the hue bridge and updated the firmware. 
After that, I re-joined the device in Z2M, but the firmware shown is still the old one. 
After that I updated two dimmers that I never joined within Z2M and after that I joined both in Z2M. Here, the firmware is shown right. 
[image]

After that I deleted/removed the one dimmer with the wrong firmware in Z2M and joined again. Still, the wrong firmware is shown.

[image]

Somehow it seems to be cached... !?!?
I checked with hue bridge and joined the device there again. In the hue bridge the firmware is shown OK. 

I have another hue device that was updated within Z2M that still shows the old firmware after OTA. 
I joined the device to the hue bridge and the new firmware is shown.

Same with a IKEA switch. Here it shows one device where no update is available (was already updated)  and another which has still the old firmware:
[image]

## What did you expect to happen
Check the firmware during the join process and update the firmware data within Z2M
Any idea?

## How to reproduce it (minimal and precise)
Update a firmware outside Z2M and delete and rejoin the device. The old firmware is shown.

## Debug info
Zigbee2MQTT version: 1.22.0-dev commit: 82f5ddb
Adapter hardware: ConBee2/RaspBee2
Adapter firmware version: 0x26720700
Frontend version: 0.6.42

Bascht74, 2021-12-05:
Is somebody else observing this problem?

Bascht74, 2021-12-05:
Maybe it is the same problem as in :
https://github.com/Koenkk/zigbee2mqtt/issues/9541

pswid, 2021-12-06:
Yes. I have the same problem. I have Hue Being LTC001. It has been succesfully upgraded to "20210331 1.88.1" (I can confirm it by linking temporarly to HUE bridge), but it's still displayed as "20160907 1.16.1_r19181" in frontend.

I'm using:
Zigbee2MQTT version: 1.22.1 commit: c2b5229f
Coordinator type: zStack30x
Coordinator revision: 20201010
Frontend version: 0.6.46

ryan-sondgeroth, 2021-12-09:
I have the same issue as well on a number of of my Hue bulbs.    If you read the swBuildId attribute value from the genBasic cluster for your Hue bulb using the Dev Console, it will report the correct firmware value.    The issue seems to stem from the fact that the firmware info is listed in two different places in the /opt/zigbee2mqtt/data/database.db file for each bulb, and when the firmware data is updated, it only gets updated in the first of those two locations.  The UI apparently uses the second of these two locations for populating the value sin the UI.   

I stopped Zigbee2mqtt, and edited the lines with the incorrect firmware manually, and that resulted in the correct firmware being displayed in the UI when I restarted it.   

While it doesn't really fix the larger issue behind this phenomenon, it may help point someone much smarter than me in the correct direction to to do so.

Bascht74, 2021-12-10:
@Koenkk @nurikk I usually don't address the maintainers directly in github, but this seems to be the best idea for solving this as the solution can be on both sides here...

Can you both maybe take a quick look at this issue. I think the infos from @ryan-sondgeroth are pointing in the right direction. 

I don't know what the two fields for firmware information in the database.db are for but either they both need to be updated by z2m or they have a different use (I don't have a clue) and the zm2 frontend is probably using the wrong entry for showing the firmware version of a device.
 
A fix would probably help a lot of people that get update buttons for only some entries - even they got the same firmware version displayed... 

Thx and have a great weekend!
Sebastian

Koenkk, 2021-12-10:
Probably fixed after https://github.com/Koenkk/zigbee2mqtt/issues/10108 is solved, current workaround is to remove device, stpo z2m, start z2m and pair device.

Bascht74, 2021-12-12:
Hi @Koenkk 
I tried a new edge version and still the old firmware (2.44.0) is shown after delete / join...
But I get an configure error in the log...

```
Zigbee2MQTT Version
1.22.1-dev commit: 28cb8b0
Coordinator Typ
ConBee2/RaspBee2
Coordinator Version
0x26720700
Frontend Version
0.6.60
```

```
Info Zigbee: allowing new devices to join.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"xgb9c-9"}'
Info Removing device '0x0017880109a62dee' (block: false, force: false)
Warning Device '0x0017880109a62dee' left the network
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880109a62dee","ieee_address":"0x0017880109a62dee"},"type":"device_leave"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x0017880109a62dee"},"type":"device_removed"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0017880109a62dee/battery/config', payload 'null'
Info MQTT publish: topic 'homeassistant/sensor/0x0017880109a62dee/action/config', payload 'null'
Info MQTT publish: topic 'homeassistant/sensor/0x0017880109a62dee/linkquality/config', payload 'null'
Info MQTT publish: topic 'homeassistant/sensor/0x0017880109a62dee/update_state/config', payload 'null'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0017880109a62dee/update_available/config', payload 'null'
Info MQTT publish: topic 'zigbee2mqtt/0x0017880109a62dee', payload ''
Info Successfully removed device '0x0017880109a62dee' (block: false, force: false)
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":false,"id":"0x0017880109a62dee"},"status":"ok","transaction":"xgb9c-10"}'
Info Device '0x0017880109a62dee' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880109a62dee","ieee_address":"0x0017880109a62dee"},"type":"device_joined"}'
Info Configuring '0x0017880109a62dee'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0017880109a62dee"},"type":"device_connected"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0017880109a62dee/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x0017880109a62dee/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x0017880109a62dee"],"manufacturer":"Philips","model":"Hue dimmer switch (929002398602)","name":"0x0017880109a62dee","sw_version":"2.44.0_hBB3C188"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x0017880109a62dee","name":"0x0017880109a62dee battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x0017880109a62dee","unique_id":"0x0017880109a62dee_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0017880109a62dee/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x0017880109a62dee/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x0017880109a62dee"],"manufacturer":"Philips","model":"Hue dimmer switch (929002398602)","name":"0x0017880109a62dee","sw_version":"2.44.0_hBB3C188"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x0017880109a62dee","name":"0x0017880109a62dee action","state_topic":"zigbee2mqtt/0x0017880109a62dee","unique_id":"0x0017880109a62dee_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0017880109a62dee/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x0017880109a62dee/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x0017880109a62dee"],"manufacturer":"Philips","model":"Hue dimmer switch (929002398602)","name":"0x0017880109a62dee","sw_version":"2.44.0_hBB3C188"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x0017880109a62dee","name":"0x0017880109a62dee linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x0017880109a62dee","unique_id":"0x0017880109a62dee_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x0017880109a62dee/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x0017880109a62dee/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x0017880109a62dee"],"manufacturer":"Philips","model":"Hue dimmer switch (929002398602)","name":"0x0017880109a62dee","sw_version":"2.44.0_hBB3C188"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x0017880109a62dee","name":"0x0017880109a62dee update state","state_topic":"zigbee2mqtt/0x0017880109a62dee","unique_id":"0x0017880109a62dee_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x0017880109a62dee/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x0017880109a62dee/availability"}],"availability_mode":"all","device":{ [... cut ...]
Error Failed to configure '0x0017880109a62dee', attempt 3 (Error: Bind 0x0017880109a62dee/1 genOnOff from '0x00212effff0657d6/1' failed (waiting for response TIMEOUT) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:753:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:517:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/philips.js:1763:13) at Configure.configure (/app/lib/extension/configure.ts:115:13))
```

Bascht74, 2021-12-12:
After force delete, restart Z2M, and rejoin, everything is OK...

[image]

I have some more devices that needs that fix. 
So https://github.com/Koenkk/zigbee2mqtt/issues/10108 didn't solve it (or a least there is something still to be fixed because of that error...

groenmarsmannetje, 2021-12-14:
> So #10108 didn't solve it (or a least there is something still to be fixed because of that error...

#10108 is still open and waiting for full fix and release.

Bascht74, 2021-12-14:
@groenmarsmannetje Thx for that information. It was my understanding to give it a try at this point as some thinks were already fixed. So let's wait.

Bascht74, 2022-01-08:
@groenmarsmannetje So now it should be fixed, right? Pull request is merged and new version is out...
So can I test it again? (have to wait for a new firmware release of a device. Maybe other people can confirm earlier)

groenmarsmannetje, 2022-01-08:
Yes it should be fixed now.

closed 2022-02-15 by github-actions[bot]