Koenkk/zigbee2mqtt #9919: Heiman HS1SA recognized as HS3SA-E

Aurel-14, 2021-11-30:
### Discussed in https://github.com/Koenkk/zigbee2mqtt/discussions/9408

<div type='discussions-op-text'>

<sup>Originally posted by **Aurel-14** October 30, 2021</sup>
Hello !

I have a CC2531 dongle with the firmware CC2531_DEFAULT_20201127.zip
I have the dev branch of zigbee2mqtt (just updated today).
I have jeedom on a RPi 4.
I bouth HEIMAN SMOKE SENSOR HS1SA-E Zigbee 3.0 version 
I have this model : https://www.zigbee2mqtt.io/devices/HS1SA.html
But it is reconized as this model : https://www.zigbee2mqtt.io/devices/HS3SA.html
Consequenses are that exposes working are "Battery" and "link quality". Smoke and Battery are not remonted (noted N/A in zigbee2mqtt control panel)

I can post log if needed but i don't know the good command to see then.

Sorry for my bad english :)
Thanks</div>

I sorry but i don't have answer for my issue and do not find solution today.

Olliek95, 2021-12-05:
Hi. I seem to have the exact same issue. I ordered (and got) three HS1SA-E (Zigbee 3.0) devices but they seem to be recognized as type HS3SA. This unfortunately results in the smoke detection not working (only battery and link quality are exposed). I have the same CC2531 dongle as the OP and I think I have a similar firmware as well. Is there a way to fix this?

Koenkk, 2021-12-06:
@Olliek95 
- what is logged in the debug log when triggering smoke on the HS1SA-E
- can you provide the data/databse.db entry of this device?

Olliek95, 2021-12-06:
Hi Koen,

It seems like I was a little too fast to judge that smoke wasn't detected properly.
When smoke is detected, I actually see it updated properly in Home Assistant (I assumed that because it wasn't shown as "expoded" that it wasn't working, but now it also shows the status there).
For completeness, this is the debug logging of zigbee2mqtt when pressing the test button on the device:

`Debug Received Zigbee message from '0xec1bbdfffea38947', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffea38947', payload '{"ac_status":false,"battery":100,"battery_low":false,"enrolled":false,"linkquality":23,"restore_reports":true,"smoke":false,"supervision_reports":false,"tamper":false,"trouble":false}'`

The device type however remains the incorrect HS3SA instead of the HS1SA.

Here is the entry from database.db:

`{"id":29,"type":"EndDevice","ieeeAddr":"0xec1bbdfffea38947","nwkAddr":63803,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"SmokeSensor-EF-3.0","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,1282,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SmokeSensor-EF-3.0","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":2,"appVersion":17,"stackVersion":2,"hwVersion":16,"dateCode":"2019.8.29"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0019430a40","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0019430a40","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":17,"stackVersion":2,"hwVersion":16,"dateCode":"2019.8.29","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1638818908813}`

Koenkk, 2021-12-06:
Fixed.

Changes will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-06 by Koenkk

Aurel-14, 2021-12-11:
Hy i confirm that now my smoke detector is recognized as HS1SA with last dev branch. Thanks !

Olliek95, 2021-12-11:
> Hy i confirm that now my smoke detector is recognized as HS1SA with last dev branch. Thanks !

Jep, same here :). Thanks @Koenkk .

GoNzCiD, 2022-01-27:
I'm in a similar issue between a xiaomi relay recognized as a xiaomi motion detector, using an old CC2531 with the routing firmware (not the latest one)
I have upgraded the docker container to the latest version, and I'll try to flash the CC2531, but for now the device still looks like a motion sensor. 

It's necessary to re-pair the device to be recognized correctly, or exists any other method to 'reload' the configuration.
Since the relay is hidden I'd prefer to modify something if it is possible, I don't know I can modify the `database.db` file and copy the JSON config from other relay preserving the `id` and `ieeeAddr` attributes.

Apart from the identification, the two relays works correctly.

Thanks in advance