Koenkk/zigbee2mqtt #9920: Philips hue bulbs needs to be paired again after power loss

ahocquet, 2021-12-01:
## What happened

When I physically switch off and on any of my hue lights - a bulb, a pot gui10 pot light - then Z2M can't connect anymore to the light and I need to pair it again. I don't have this problem with my Ikea bulbs.

Here's the error log I get : 
```
 Publish 'set' 'state' to 'Bureau Anthony - Hue Bulb' failed: 'Error: Command 0x00178801093fd3fb/11 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 58453 - 11 - 27 - 8 - 11 after 10000ms)'
 ```

## What did you expect to happen
It should not need to pair again.

## How to reproduce it (minimal and precise)
Physically power off the hue light.
Power on.
Light can't be controlled anymore

## Debug info
Zigbee2MQTT version:     1.22.0-dev commit: 0f33150 
Adapter hardware:   zzh (CC2652R Stick)
Adapter firmware version: CC2652R_coordinator_20211114

danieledwardgeorgehitchcock, 2021-12-01:
This may be a Zigbee network issue rather than a Z2M issue.

It takes a while for a Zigbee network to self-heal. If the device is in a low-signal area then it will be off-network until a nearby device detects it and brings it back in

ahocquet, 2021-12-01:
The thing is that it never happens with my Ikea bulbs but every single times with my Hue bulbs. And also, even if I wait a day or two, it doesn't get back to the network, I do have to pair it again.

Tonio16, 2021-12-03:
Hello

I would try to forced erase them  next time, reboot and to pair them again.

Antoine

ahocquet, 2021-12-09:
Problem seems to be solved. I had a different zigbee network with deconz running along zigbee2mqtt. After removing that network, the problem doesn't popup anymore. 

I close the issue.

closed 2021-12-09 by ahocquet