Koenkk/zigbee2mqtt #9939: MQTT template error in HA for Xiaomi ZNCLDJ11LM when homeassistant_legacy_entity_attributes is disabled

andreypolyak, 2021-12-01:
## What happened
I disabled homeassistant_legacy_entity_attributes in configuration.yaml. After that, I started to get following errors in my Home Assistant logs:
```
2021-12-02 00:02:21 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"last_seen":1638392541117,"linkquality":69,"position":0,"running":false,"update_available":null}, template: {{ value_json['update']['state'] == "available" }})
```

I only get this error for Xiaomi ZNCLDJ11LM device.

This is how MQTT topic for this device looks like (notice, there is no "update" attribute):
```json
{"last_seen":1638392870095,"linkquality":99,"position":0,"running":false,"update_available":null}
```

Discovery topic despite this has mentions "update" attribute:
```json
{"availability":[{"topic":"zigbee2mqtt_living_room/bridge/state"},{"topic":"zigbee2mqtt_living_room/Living Room Cover/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d00040c6454"],"manufacturer":"Xiaomi","model":"Aqara curtain motor (ZNCLDJ11LM)","name":"Living Room Cover"},"device_class":"update","enabled_by_default":true,"entity_category":"diagnostic","name":"Living Room Cover update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt_living_room/Living Room Cover","unique_id":"0x00158d00040c6454_update_available_zigbee2mqtt_living_room","value_template":"{{ value_json['update']['state'] == \"available\" }}"}
```

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: CC2562
Adapter firmware version:

Koenkk, 2021-12-03:
Should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

michelotten, 2021-12-13:
I had the same issue with the same device (ZNCLDJ11LM). I can confirm switching to dev branch, fixes the issue.

IgorSimic, 2021-12-13:
I can confirm that switching to dev branch resolve this issue, but in dev branch my thermostat, Moes BHT series Thermostat (BHT-002-GCLZB) not working and also, zigbee2mqtt own configuration web page not working.
I "combined" master branch with dist and lib directories from dev branch and then everithing works as expected.

jaminbenuk, 2021-12-29:
Sorry I'm a bit of a newb to this but I still see the following warning in Home Assistant... the error has now been resolved.

2021-12-29 13:44:25 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received
2021-12-29 13:44:25 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value received

Are these something to worry about?

Running the Dev branch for both HA and Z2M as of 15 mins ago.

closed 2022-02-05 by github-actions[bot]