Koenkk/zigbee2mqtt #9943: Negative temperature not correct below -20°C

fariouche, 2021-12-01:
## What happened
I have 2 new CASA.AI thermometers, that I've put in my fridge.
I've paired them fine (cc26X2R1 zzh) and the room temperature is reported fine (compared to my 2 other thermometers in the room)
One is put in the refrigerator, around 6°C, and temperature is fine.
The other is in the freezer, where the temperature is expected to be around -18°C.. (should be because I suspect my freezer to be way below, around -26 but not sure if the is my old school thermometer that is not working)
This thermometer should be able to handle -30°C, and I've seen the temperature drop to -18, then -19.5 and after that the temperature is displayed as -128.42 then -129.62 as if -108 is added for temperature below -20)
Not sure if this is the device itself, as I don't have a CASA.AI gateway to compare :( but they say they support a range -30 to 100°C)

## What did you expect to happen
Temperature below -20 should be correctly displayed

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: latest from github directly, and 1.22
Adapter hardware: CC26X2R1 zzh from electrolama
Adapter firmware version: flashed with latest available firmware from july 2021

Thank you! Love this project :)

fariouche, 2022-01-01:
Hello,

So it seems that this is a bug in the firmware from casaia. No answer from them... I then propose a workaround for this thermometer. Values below -20 are invalid, most likely the thermometer does not handle such low values despite their claim!
Adding an offset of 108 for values below -128 seems to fix the problem and I can at last monitor my freezer temperature!
There is still a small glitch around -20, the this offset make the temperature curve continuous.

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const utils = require('zigbee-herdsman-converters/lib/utils');

const fzLocal = {
    casaia_temperature: {
    cluster: 'msTemperatureMeasurement',
        type: ['attributeReport', 'readResponse'],
        options: [exposes.options.precision('temperature'), exposes.options.calibration('temperature')],
        convert: (model, msg, publish, options, meta) => {
            temperature = parseFloat(msg.data['measuredValue']) / 100.0;
            const property = utils.postfixWithEndpointName('temperature', msg, model);
            if(temperature <= -128.0)
		temperature += 108.0;
            return {[property]: utils.calibrateAndPrecisionRoundOptions(temperature, options, 'temperature')};
        },
    }
}

module.exports = [
    {
        zigbeeModel: ['CTHS317ET'],
        model: 'CTHS-317-ET',
        vendor: 'CASAIA',
        description: 'Remote temperature probe on cable',
        fromZigbee: [fzLocal.casaia_temperature, fz.battery],
        toZigbee: [],
        meta: {battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(3);
            await reporting.bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement']);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.temperature(endpoint);
            await reporting.batteryVoltage(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.temperature(), e.battery_low(), e.battery()],
    }
];

fariouche, 2022-02-02:
just to keep the issue open

closed 2022-02-10 by github-actions[bot]

pubFbx, 2024-01-22:
Hello,
It's not working for me. I have the message :

debug 2024-01-22 18:42:46: Received Zigbee message from 'CyTeCongel', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":-2000}' from endpoint 3 with groupID 0
error 2024-01-22 18:42:46: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'logger')}
debug 2024-01-22 18:42:46: TypeError: Cannot read properties of undefined (reading 'logger')

Is there any detailed documentation to understand how to program this kind of thing.  Apart from "Support new devices" and these few examples, I haven't found anything really educational.

fariouche, 2024-01-23:
I have no idea about this error 'logger'. In my case it is still working with version 1.33.1.
I never had the time to open a PR so that @Koenkk can merge this fix, but your problem is unrelated and I suggest you open a new ticket

pubFbx, 2024-01-27:
thank you for your reply.  It's too complicated for me to make a PR.  I'll try to find it myself and if I can't, I'll create a fix at Openhab level (even if it's less clean)