Koenkk/zigbee2mqtt #9948: Error: Error Resource temporarily unavailable Cannot lock port - after reboot

KevinHubert-Dev, 2021-12-02:
## What happened
After downloading the newest version of zigbee2mqtt as a separated folder to my existing instance of zigbee2mqtt I've stopped the previous zigbee2mqtt instance and rebooted my raspberry pi 3B. After the system has started again I've tried to get the default environment of zigbee2mqtt (without any special configuration) running and then migrate my devices to the newest version using the Zigbee2mqtt-Guide. For my bad the zigbee2mqtt instance does not start and throw the following error:
`2021-12-02 08:22:16: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port`
Log: https://pastebin.com/e3uRAJnm

When I now try to run my previous instance, the same error occures. The zigbee dongle is a CC2531 and has ran before without problems even after reboots.

FAQ and Install guide has been both viewed. Problem still exists. Following steps have been tried to solve the problem using other reported issues in this repo:

From: https://github.com/Koenkk/zigbee2mqtt/issues/4323
sudo ls -l /proc/[0-9]/fd/ |grep /dev/ttyACM0
Does not return anything

From: https://github.com/Koenkk/zigbee2mqtt/issues/208
ls -lat /dev/ttyACM0
> crw-rw---- 1 root dialout 166, 0 Dec  2 08:35 /dev/ttyACM0

From: https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#cc2531-coordinators-only-press-the-reset-button-on-the-device
Reset button of CC2531 has been pressed 
Green led has not turned of but keeped on the whole time

From FAQ:
Power source is stable, I'm running the system as root, dongle has worked before.

## What did you expect to happen
Even if the newest version of zigbee2mqtt may have problems with the dongle I would at least assume that the previous and unchanged version of zigbee2mqtt does work with the CC2531 given.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: Master branch - 1.22.1 https://github.com/Koenkk/zigbee2mqtt/commit/c2b5229f762358055bce4760c33212b89a04e6d7
Adapter hardware: CC2531
[Logs](https://pastebin.com/e3uRAJnm) (same as above)

Thank you in advance

KevinHubert-Dev, 2021-12-03:
Update from myself: Problem seems to be on my raspberry pi. I've setup a different raspberry with the same CC2531 and this works without problem. Even if checks for USB availability work flawless on both systems.

Any future help welcomed anyway :)

closed 2022-01-10 by github-actions[bot]

snack04, 2022-08-24:
Have you been able to find a solution?

I am faced with the same issue I believe. 

Thank you for the detailed post and commands - I do not have a huge amount of experience and it has been very thorough and easy to follow. I am however still stuck with this issue.

I am in the process of setting up a Homebridge on Raspberry Pi 4 and am not able to get past this step to run npm start.

snack04, 2022-08-24:
I have managed to find a solution for my situation, if anyone else runs into this issue with limited experience. As I have seen in other posts, the port is being used by something else... it turned out out zigbee2mqtt was already running, therefore the npm start command produced the error highlighted above. 

I used the command below to stop it manually
`sudo systemctl stop zigbee2mqtt`

then using `npm start` I got it to work.

Note: This was during initial setup, and not for every restart.

gnolnos, 2023-01-26:
> I have managed to find a solution for my situation, if anyone else runs into this issue with limited experience. As I have seen in other posts, the port is being used by something else... it turned [...]

This is the solution!

turnipsforme, 2023-02-08:
THIS WAS IT OH MY GOD I'VE BEEN AT THIS FOR HOURS THANK YOU

grandamemt, 2024-03-23:
A little late, but since I'm having the same issue I didn't want to start a new thread.   I am unable to run "sudo systemctl stop zigbee2mqtt" or "npm start".  
I get: "bash: systemctl: command not found" and "bash: npm: command not found".
Thanks in advance for any suggestions!

Grogger, 2024-05-11:
Also a little late, but I solved this by stopping the "Silicon Labs Multiprotocol" add-on, which was for the multi-pan firmware on my coordinator. I just moved off of the multi-pan firmware to plain-old ezsp and the add-on was trying to connect, apparently locking up the coordinator.

poida97, 2024-06-09:
> he "Silicon Labs Multiprotocol" add-on, which was for the multi-pan firmware on my coordinator. I just moved off of the multi-pan firmware to plain

OMG I have lost count of how many random settings and config I have changed while this was the answer all along. Thank you!

poida97, 2024-06-09:
> I have managed to find a solution for my situation, if anyone else runs into this issue with limited experience. As I have seen in other posts, the port is being used by something else... it turned [...]

OMG I have lost count of how many random settings and config I have changed while this was the answer all along. Thank you!

poycorp, 2024-11-10:
2 hours of research to finally land here and find my solution ! Thank you !

iamlunaburger, 2025-11-21:
What it ended up being for me was old firmware for my Sonoff dongle.  I flashed the latest version and everything worked after that.