Koenkk/zigbee2mqtt #9952: Battery information of Xiaomi WSDCGQ12LM not loading.

Murianwind, 2021-12-02:
## What happened
Temperature, humidity and barometric pressure from the Xiaomi WSDCGQ12LM are loaded correctly, but not just the battery.

## What did you expect to happen
I want the battery information to be printed.

## How to reproduce it (minimal and precise)
Register Xiaomi WSDCGQ12LM to Z2M. Over time, temperature and humidity data will come up. However, only Null values are raised for battery information.

## Debug info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: CC2652P SMARTLIGHT SLZB-02
Adapter firmware version: CC1352P2_CC2652P_other_coordinator_20210708

lux73, 2021-12-02:
sometimes Battery Value need 12-24 hrs to publish

Murianwind, 2021-12-02:
I have been using this sensor for a month already

FelipeDamasio, 2021-12-02:
Same here with Aqara Opple WXCJKG13LM. Even after repair the device, the batery status become 0%.

salopette, 2021-12-02:
> Same here with Aqara Opple WXCJKG13LM. Even after repair the device, the batery status become 0%.

I now have temperature adjustment on the WXCJKG13LM? What for? Battery is also 0% at all switches

[image]
[image]

1igeza, 2021-12-03:
Also for me Opple switch show 0%, when I saw it first I replaced battery, after replacing it was still showing 0%.  It helps pushing "C" button but not for long. 
`debug 2021-12-03 09:55:23: Received Zigbee message from 'Opple', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,76,12,3,40,25,4,33,168,19,5,33,53,0,6,36,1,0,0,0,0,8,33,17,1,10,33,0,0,12,32,1,100,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 4 with value 5032
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 5 with value 53
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown vtype=36, pos=16
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 6 with value null
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown vtype=1, pos=17
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 36 with value null
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown vtype=0, pos=18
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown vtype=0, pos=19
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 0 with value null
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown vtype=0, pos=20
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 0 with value null
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown vtype=0, pos=21
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 0 with value null
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown vtype=8, pos=22
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 0 with value null
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 8 with value 273
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 10 with value 0
debug 2021-12-03 09:55:23: lumi.remote.b686opcn01: unknown index 12 with value 1
info  2021-12-03 09:55:24: MQTT publish: topic 'zigbee2mqtt/Opple', payload '{"action":null,"battery":0,"last_seen":"2021-12-03T09:55:23+01:00","linkquality":66,"operation_mode":"event","state":"OFF","temperature":25,"voltage":null}'
`

FelipeDamasio, 2021-12-03:
Even after 24, rebot, repair... the battery for Opples switch is 0%.

[image]

sv87411, 2021-12-05:
> > Same here with Aqara Opple WXCJKG13LM. Even after repair the device, the batery status become 0%.
> 
> I now have temperature adjustment on the WXCJKG13LM? What for? Battery is also 0% at all switches
> 

I'm not great at reading/understanding the Zigbee herdsman code, but when I last checked the Opple sections of code for the WXCJKG13LM and the messages that come from the switch, they appear to output temperature values, but from my WXCJKG13LM devices' it's always 25C. These are the periodic messages I get from my switches through Z2M:

`payload '{"battery":0,"linkquality":42,"operation_mode":null,"state":"OFF","temperature":25,"voltage":null}`

I am pretty sure these switches don't support temperature information and the herdsman code is incorrectly interpreting the device messages and assuming they do.

I also only ever get battery info after pressing the 'C' button once on the back of the switch. It soon reverts back to null as you can see above.

I guess these switches have buggy firmware that reports battery state if you press the 'C' button, but doesn't report battery info in the periodic status messages. Unless you are using the Xiaomi hub you probably can't update the firmware.

I record all my sensor battery states to an influxDB and I've given up with the battery info for the WXCJKG13LM switches and just filter those devices out of my queries.

They are great reasonably priced switches (I have 5), but I guess along with that comes poor firmware.

FelipeDamasio, 2021-12-05:
Yes I agree. My device has the same behavior.
The problem is that the new version of the Z2M is using the 0% battery info (it only sends the battery info a few times a day to save battery life), whereas the previous versions only considered non-zero battery values. I believe it's something simple to solve, and I hope they fix it as soon as possible.
[image]

1igeza, 2021-12-05:
> I guess these switches have buggy firmware that reports battery state if you press the 'C' button, but doesn't report battery info in the periodic status messages. Unless you are using the Xiaomi [...]

I checked my database and for me it's started 24 November, now I will check commits made in this time. 
[image]

Koenkk, 2021-12-06:
@1igeza is any voltage reported in your case? If not, could you provide me a debug log of 24 hours of z2m operation.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

1igeza, 2021-12-06:
> @1igeza is any voltage reported in your case? If not, could you provide me a debug log of 24 hours of z2m operation.
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

If the battery is shown correctly then I also have voltage. If battery is "0" then voltage is "null"
`2021-12-03 09:55:24: MQTT publish: topic 'zigbee2mqtt/Opple', payload '{"action":null,"battery":0,"last_seen":"2021-12-03T09:55:23+01:00","linkquality":66,"operation_mode":"event","state":"OFF","temperature":25,"voltage":null}' `
In HA I don't have option to enable voltage sensor in opple device (diagnostic section can only enable battery, last seen and link quality) (I can enable voltage sensor in other zigbee devices). 
I update z2m to newest dev and started debugging.

Koenkk, 2021-12-06:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Murianwind, 2021-12-06:
I posted a problem about the WXCJKG12LM, but they are all talking about the
WXCJKG13LM.

Is WXCJKG12LM also modified?
-----
Murian Song

SW Testing Evangelist & Consultant / TOC Evangelist & Consultant

Blog: http://murianwind.blogspot.com

Twitter: http://twitter.com/murianwind

Facebook: http://www.facebook.com/murianwind

odelma, 2021-12-06:
Thanks for referring to my issue (https://github.com/Koenkk/zigbee2mqtt/issues/10033) @salopette. While I read the couple of first messages of this issue before opening mine, I did not realize it might be about the same thing (as device in question was of different model and headline was only about battery). But indeed it very likely is about the same thing or at least very much related. 

@Koenkk - should the patch above also fix the issue I reported in #10033?

1igeza, 2021-12-07:
Since yesterday it's showing battery correctly. So I think it is fixed for me.

leroyloren, 2021-12-07:
Thank you, it works. It wouldn't be possible to edit "details" like

1) The temperature is permanently 25 in the messages, but it is not displayed anywhere.
`MQTT publish: topic 'zigbee2mqtt/6 tlacitko', payload '{"action":"","battery":100,"last_seen":"2021-12-07T14:25:48+01:00","linkquality":236,"operation_mode":null,"state":"OFF","temperature":25,"voltage":3261}'`

[image]
[image]

2) Display the strange firmware version "2019www."
[image]

sv87411, 2021-12-09:
I can confirm after updating to latest dev build a day ago this is fixed for me too. All of my WXCJKG13LM devices are reporting battery correctly and it's not reverting to 'null' and 0.

👍️

Koenkk, 2021-12-14:
@arteck it is fixed in the latest zigbee-herdsman-converters.

closed 2022-01-21 by github-actions[bot]