Koenkk/zigbee2mqtt #9954: SONOFF SNZB-02 stoppen functioning with ZZH

pznamenskii, 2021-12-02:
## What happened

about 3 months ago I bound 2 SONOFF SNZB-02 sensors into my HA setup with ZZH stick successfully.
A week ago I bought a 3rd sensor and failed to bind it whatever I tried. At this moment I realized my previous 2 sensor have not been reporting any data for a long while. I tried removing (forced) one of them and was not able to join it back.
Logs do not show a single thing even on debug.

## What did you expect to happen

sensors join Zigbee network successfully

## How to reproduce it (minimal and precise)

Try joining SONOFF SNZB-02

## Debug info
Zigbee2MQTT version: 1.22.1-1
Adapter hardware: ZZH! CC2652R  
Adapter firmware version: 20210708

jahu, 2021-12-07:
I confirm the issue - I've just realized that my snzb-02 (all of them) stoped working after update to latest version of Z2m (from 1.18.x) :/ is there any chance of fixing this or revert of Z2m is only option?

afics, 2022-01-09:
Not stale, still happening to me.

sanyafifa, 2022-01-09:
the problem is urgent

[image]

sebastianovide, 2022-01-09:
same here. I've thought it was broken and I've bought a new one... they are seen but don't report any useful data

Koenkk, 2022-01-09:
This device can be tricky to pair, there are some tips here: https://github.com/Koenkk/zigbee2mqtt/issues/8971#issuecomment-933473038

Radlagerkok, 2022-01-11:
the problem appeared after upgrading from 1.18.  Before that, more than a year worked without problems on 2531. The new sonoff 2652 did not solve the problem

simie7, 2022-01-11:
I have the same problem:

- Used successfully [this](https://sonoff.tech/product/smart-home-security/snzb-02/) device on older version (not sure about version, installed and paired in the begining of December 2021)
- Did upgrade to 1.22.2 version.
- Re-paired device.
- Now getting the same errors like others.

Current version:
[image]

Using [this](https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/) as coordinator.

Tried different combinations of re-pairing, with battery removal, usb stick removal, reboot - nothing helps.

simie7, 2022-01-11:
This sequence worked for me:
- `systemctl stop zigbee2mqtt`
- make backup of `data` folder
- `git checkout tags/1.22.1`
- `systemctl start zigbee2mqtt`
- re-pair `SNZB-02` device
- `systemctl stop zigbee2mqtt`
- restore from backup of `data` folder (in my case wasn't needed)
- `git checkout master`
- `systemctl start zigbee2mqtt`

I guess something messed between `1.22.1` and `1.22.2` versions.

sanyafifa, 2022-01-11:
> This sequence worked for me:

Hi!
Does SNZB-02 connect directly to the coordinator?
I have no pairing after 3 flashes

simie7, 2022-01-11:
In my case - yes, directly.

simie7, 2022-01-11:
Interesting fact: today got another one sensor, and successfully paired with the first try, without version downgrade.

sanyafifa, 2022-01-12:
found a similar problem :)
The author writes what helped:
https://notenoughtech.com/home-automation/is-sonoff-zigbee-dongle-plus-finally-ready-for-launch/

`Zigbee2MQTT:error 2022-01-08 19:11:30: Configure failed: Error: Bind 0x0012xxxxxxx/1 msTemperatureMeasurement from '0x00124xxxxxxxx1/1' failed (AREQ - ZDO - bindRsp after 10000ms)`
>No matter what I’ve done, I could not receive any feedback from the sensor even though ZigBee2MQTT was reporting successful pairing. What had fixed it for me, was wiping all of my devices and adding this sensor first. It’s a weird issue, and I may bother Koen about it.

evilworm, 2022-01-26:
Having the same problem with SONOFF SNZB-02. After upgrading to 1.22.2 the device fails to bind.

```
info Device '0x00124b001c4b2951' joined
info MQTT publish: topic 'raspberrypi/bridge/event', payload '{"data":{"friendly_name":"0x00124b001c4b2951","ieee_address":"0x00124b001c4b2951"},"type":"device_joined"}'
info Configuring '0x00124b001c4b2951'
info MQTT publish: topic 'raspberrypi/bridge/log', payload '{"message":{"friendly_name":"0x00124b001c4b2951"},"type":"device_connected"}'
info Starting interview of '0x00124b001c4b2951'
info MQTT publish: topic 'raspberrypi/bridge/event', payload '{"data":{"friendly_name":"0x00124b001c4b2951","ieee_address":"0x00124b001c4b2951","status":"started"},"type":"device_interview"}'
info MQTT publish: topic 'raspberrypi/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b001c4b2951"},"type":"pairing"}'
error Configure failed: Error: Bind 0x00124b001c4b2951/1 msTemperatureMeasurement from '0x00124b0021600162/1' failed (AREQ - ZDO - bindRsp after 10000ms)
info Successfully configured '0x00124b001c4b2951'
```

I tracked down it to devices/sonoff.js:15 (zigbee-herdsman-converters package) line:
`await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);`

this line throws an error which is caught a few lines later and loggs an error

if it helps, I dumped parameters from the "bind" method:

endpoint=`{"ID":1,"profileID":260,"deviceID":770,"inputClusters":[0,3,1026,1029,1],"outputClusters":[3],"deviceNetworkAddress":57608,"deviceIeeeAddress":"0x00124b001c4b2951","clusters":{},"_binds":[],"_configuredReportings":[],"meta":{},"pendingRequests":[]}`

coordinatorEndpoint=`{"ID":1,"profileID":260,"deviceID":5,"inputClusters":[],"outputClusters":[],"deviceNetworkAddress":0,"deviceIeeeAddress":"0x00124b0021600162","clusters":{},"_binds":[],"_configuredReportings":[],"meta":{},"pendingRequests":[]}`

bindClusters=`["msTemperatureMeasurement","msRelativeHumidity","genPowerCfg"]`

evilworm, 2022-01-27:
after playing around with the sensor and trying to bind it literally about 100 times I figured that if you rapidly remove/insert battery it will eventually register correctly ... also true if you use the "reconfigure" and play with the battery.
I suspect that the 10 sec timeout for the bind is too small for this particular sensor

sanyafifa, 2022-02-05:
> I figured that if you rapidly remove/insert battery it will eventually register correctly

Hey
is it possible to give a short algorithm how did you get the conjugation?

closed 2022-03-15 by github-actions[bot]