Koenkk/zigbee2mqtt #9958: Failed to setup reporting for ZNLDP12LM

iStitch07, 2021-12-02:
## What happened
Every time when I restart z2m I have error message in log:
```
Failed to setup reporting for '0x00158d00054ac101' - Error: Read 0x00158d00054ac101/1 lightingColorCtrl(["colorCapabilities"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
```

0x00158d00054ac101 in my case is ZNLDP12LM

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: CC26X2R1
Adapter firmware version: 20210708

iStitch07, 2021-12-03:
```
Zigbee2MQTT:error 2021-12-02 15:14:41: Failed to setup reporting for '0x00158d00054ac069' - Error: Read 0x00158d00054ac069/1 lightingColorCtrl(["colorCapabilities"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
,    at start (/app/index.js:97:5)
,    at Controller.start (/app/lib/controller.ts:152:9)
,    at Controller.callExtensions (/app/lib/controller.ts:310:17)
,    at Report.start (/app/lib/extension/legacy/report.ts:179:17)
,    at Report.setupReporting (/app/lib/extension/legacy/report.ts:95:72)
,    at Object.condition (/app/lib/extension/legacy/report.ts:47:63)
,    at getColorCapabilities (/app/lib/extension/legacy/report.ts:24:9)
,    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:430:22)
,    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:310:23)

```

Koenkk, 2021-12-04:
The problem is caused by the report feature which is deprecated, see second point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.17.1 how to migrate.

closed 2021-12-04 by Koenkk