Koenkk/zigbee2mqtt #9962: returning log message

strandbein, 2021-12-02:
Now that all groups and scenes are working fine for me - thanks for your great work - i have this repeating message. I am on AH addon and Conbee II. What might be the problem?

(node:361) UnhandledPromiseRejectionWarning: TypeError: Cannot read property '0' of undefined
    at publish (/app/node_modules/mqtt-packet/writeToStream.js:352:44)
    at Object.generate [as writeToStream] (/app/node_modules/mqtt-packet/writeToStream.js:34:14)
    at sendPacket (/app/node_modules/mqtt/lib/client.js:96:27)
    at MqttClient._sendPacket (/app/node_modules/mqtt/lib/client.js:1069:7)
    at MqttClient.publish (/app/node_modules/mqtt/lib/client.js:542:14)
    at /app/lib/mqtt.ts:153:25
    at new Promise (<anonymous>)
    at MQTT.publish (/app/lib/mqtt.ts:152:16)
    at Controller.publishEntityState (/app/lib/controller.ts:269:33)
    at publish (/app/lib/extension/receive.ts:150:22)
(node:361) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 128)

Koenkk, 2021-12-03:
Could you provide the debug log of what happens before this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

strandbein, 2021-12-03:
hopefully this is the right one
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7652084/log.txt)

Koenkk, 2021-12-04:
I don't see the error in this log (would expect to find `(node:361) UnhandledPromiseRejectionWarning: TypeError: Cannot read property '0' of undefined`)

strandbein, 2021-12-04:
may be  i am too stupid to generate a proper log. tried it with docker logs xxx > log.txt 2>&1 but the log file is empty

Koenkk, 2021-12-04:
Could you provide what is logged just above `(node:361) UnhandledPromiseRejectionWarning: TypeError: Cannot read property '0' of undefined`? That may already be enough

strandbein, 2021-12-04:
ok, I will do tonight

strandbein, 2021-12-06:
zigbee2mqtt

The message only appears in the web browser log, here at 19:14:40. Here the log from the browser tab, attached the log.txt.

advanced:
  log_level: info
zigbee_herdsman_debug: true

Browser log:
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/Thermometer Herrenzimmer', payload '{"battery":100,"humidity":40.7,"temperature":20.3}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/Ute Motion Sensor', payload '{"battery":100,"battery_low":false,"occupancy":false,"voltage":2100}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/Herrenzimmer Orbi Plug', payload '{"state":"ON"}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/Wohnungstür Kontakt', payload '{"battery":100,"contact":true}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/Wohnzimmer Balkontür', payload '{"battery":91,"contact":true}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/Herrenzimmer Balkontür', payload '{"battery":100,"contact":true}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/WZ Schrank 2', payload '{"brightness":1,"color_temp":370,"state":"ON"}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/WZ Schrank', payload '{"brightness":1,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_temp":370,"state":"ON"}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/WZ Schrank 3', payload '{"brightness":1,"color_temp":370,"state":"ON"}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/Küche HomePod Plug', payload '{"state":"OFF"}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/Wohnzimmer Motion Sensor', payload '{"battery":71,"battery_low":false,"occupancy":true}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/HZ mit Globus', payload '{"brightness":1,"color_temp":370,"state":"OFF"}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/Herrenzimmer Globus', payload '{"brightness":1,"color_temp":370,"state":"OFF"}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/WZ Schrank 1', payload '{"brightness":1,"color_temp":370,"state":"ON"}'
Zigbee2MQTT:info  2021-12-06 19:14:40: MQTT publish: topic 'z2m/WZ Schrank 4', payload '{"brightness":1,"color_temp":370,"state":"ON"}'
(node:363) UnhandledPromiseRejectionWarning: TypeError: Cannot read property '0' of undefined
    at publish (/app/node_modules/mqtt-packet/writeToStream.js:352:44)
    at Object.generate [as writeToStream] (/app/node_modules/mqtt-packet/writeToStream.js:34:14)
    at sendPacket (/app/node_modules/mqtt/lib/client.js:96:27)
    at MqttClient._sendPacket (/app/node_modules/mqtt/lib/client.js:1069:7)
    at MqttClient.publish (/app/node_modules/mqtt/lib/client.js:542:14)
    at /app/lib/mqtt.ts:153:25
    at new Promise (<anonymous>)
    at MQTT.publish (/app/lib/mqtt.ts:152:16)
    at Controller.publishEntityState (/app/lib/controller.ts:269:33)
    at Controller.start (/app/lib/controller.ts:158:26)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:363) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:363) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
2021-12-06T18:14:40.257Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x1bbf EP:1 SeqNr. 20 request id: 5
2021-12-06T18:14:40.259Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-12-06T18:14:40.259Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-12-06T18:14:40.290Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xd733 EP:1 SeqNr. 21 request id: 6
2021-12-06T18:14:40.290Z zigbee-herdsman:deconz:driver query aps data indication
2021-12-06T18:14:40.297Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xa7df EP:1 SeqNr. 22 request id: 7
2021-12-06T18:14:40.299Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 25
2021-12-06T18:14:40.301Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 5 status: 0
2021-12-06T18:14:40.301Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-12-06T18:14:40.302Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110
2021-12-06T18:14:40.302Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-12-06T18:14:40.304Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x3394 EP:1 SeqNr. 23 request id: 8
2021-12-06T18:14:40.307Z zigbee-herdsman:deconz:driver query aps data confirm
2021-12-06T18:14:40.309Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xc779 EP:1 SeqNr. 24 request id: 9
2021-12-06T18:14:40.313Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 26
2021-12-06T18:14:40.317Z zigbee-herdsman:deconz:driver query aps data indication
[... cut ...]
2021-12-06T18:14:40.375Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xc779 request id: 9 confirm status: 0
2021-12-06T18:14:40.375Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-12-06T18:14:40.376Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 6
2021-12-06T18:14:40.376Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.:

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7662803/log.txt)

Koenkk, 2021-12-06:
Can you proivde me your `configuration.yaml`?

strandbein, 2021-12-06:
I usually use qos: 0. Not a good idea?

{
  "external_converters": [],
  "devices": [
    "devices.yaml"
  ],
  "groups": [
    "groups.yaml"
  ],
  "homeassistant": false,
  "permit_join": false,
  "mqtt": {
    "base_topic": "z2m",
    "server": "mqtt://core-mosquitto:1883",
    "user": "xxx",
    "password": "xxx"
  },
  "serial": {
    "port": "/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2407034-if00",
    "adapter": "deconz"
  },
  "advanced": {
    "log_level": "error",
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blocklist": [],
    "availability_passlist": [],
    "ikea_ota_use_test_url": false
  },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "frontend": {
    "port": 8099
  },
  "experimental": {},
  "availability": true,
  "zigbee_herdsman_debug": false
}

Koenkk, 2021-12-06:
what is in `devices.yaml` and `groups.yaml`?

strandbein, 2021-12-06:
You are absolutely right: one device had an invalid qos value. Deleted and the error is gone. Great work, thank you very much!

Koenkk, 2021-12-07:
Good, added a check for this, z2m will refuse to start now when an invalid QOS value is used.

closed 2021-12-07 by Koenkk