Koenkk/zigbee2mqtt #9965: Aqara MCCGQ14LM not sending state

NdR91, 2021-12-02:
## What happened

After the pairing, two of my brand new MCCGQ14LM are not sending any state except the LQI.
I tried to repai both of them, but without any result.

[image]

## What did you expect to happen

Receiving the states of Contact, Battery and Voltage

## How to reproduce it (minimal and precise)

Pair a MCCGQ14LM device

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2652
Adapter firmware version: Latest Z-Stack 3.x.0

NdR91, 2021-12-02:
Sorry, my bad.

I updated to the latest FW (didn't see the develop branch): Z-Stack_3.x.0 20211114

I ha dto re-pair again the device, now it's working perfectly

closed 2021-12-02 by NdR91

orbsmiv, 2021-12-04:
@Koenkk Is this 3.x.0 firmware on the develop branch good to use or should I wait on a master branch release?

NdR91, 2021-12-04:
> @Koenkk Is this 3.x.0 firmware on the develop branch good to use or should I wait on a master branch release?

I'm using it since 2 days and no issue found.

My ZigBee network counts abt 40 devices of different brands.

Koenkk, 2021-12-04:
> @Koenkk Is this 3.x.0 firmware on the develop branch good to use or should I wait on a master branch release?

Should be fine, running it for 2 weeks already and no problems, will probably made official with the next z2m release.