Koenkk/zigbee2mqtt #9969: can't set valve position for EUROTRONIC SPIRIT SPZB0001

ngrataloup492, 2021-12-02:
## What happened
can't set valve position for EUROTRONIC SPIRIT SPZB0001, Have an error INVALID VALUE when I want to change the value with the slider. 
I have a conbee 2 and i'm in the last version of zigbee2mqtt (1.22.1)

I've 4 valves with the same pb

Equipemnt in zigbee2mqtt database 
```
{"id":19,"type":"EndDevice","ieeeAddr":"0x00158d00053d36a8","nwkAddr":45993,"manufId":4151,"manufName":"Eurotronic","powerSource":"Battery","modelId":"SPZB0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,513,25,10],"outClusterList":[0,1,3,4,513,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"SPZB0001","manufacturerName":"Eurotronic","powerSource":3,"zclVersion":2,"appVersion":22,"stackVersion":5,"hwVersion":35,"dateCode":"20191014","swBuildId":"22190930"}},"hvacThermostat":{"attributes":{"16384":1,"16385":74,"16386":0,"16387":2100,"16392":1,"unoccupiedHeatingSetpoint":1600,"occupiedHeatingSetpoint":2100,"localTemp":2050,"pIHeatingDemand":37,"localTemperatureCalibration":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":80,"batteryAlarmState":0}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff060dcc","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00212effff060dcc","endpointID":1}],"configuredReportings":[{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":600,"repChange":25},{"cluster":513,"attrId":8,"minRepIntval":0,"maxRepIntval":600,"repChange":1},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":600,"repChange":25},{"cluster":513,"attrId":20,"minRepIntval":0,"maxRepIntval":600,"repChange":25},{"cluster":513,"attrId":16387,"minRepIntval":0,"maxRepIntval":600,"repChange":25},{"cluster":513,"attrId":16392,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}}},"appVersion":22,"stackVersion":5,"hwVersion":35,"dateCode":"20191014","swBuildId":"22190930","zclVersion":2,"interviewCompleted":true,"meta":{"configured":3},"lastSeen":1638082822389}
```

Debug log :
```
Publish 'set' 'valve_position' to 'Salon_Vanne1' failed: 'Error: Write 0x00158d00053d36a8/1 hvacThermostat({"16385":{"value":50,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'
Debug Error: Write 0x00158d00053d36a8/1 hvacThermostat({"16385":{"value":50,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_VALUE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:310:23) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:392:22) at Object.convertSet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:3453:13) at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:247:36)
```

[image]
I already check that the parameters are good
[image]
[image]

[image]

## What did you expect to happen
I expect the value on the valve will modifiy when I'm in TRV_mode 1 and move the slider of the valve_position

## How to reproduce it (minimal and precise)
Set the TRV_Mode to 1
and just try to move the slider valve_position

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: Conbee II
Adapter firmware version: 0x26720700

ngrataloup492, 2022-01-02:
issue still open, did anyone have this pb ?

sygys, 2022-01-12:
I have no problem with this whatsoever. TRV_mode needs to be 1 and after that you can set the valve position just fine.

Keep in mind (just figured this out today) that you have to post the valve position every so much time to keep the valve in TRV_mode 1. i guess its a safety measurement. If the system goes down and the valve doesnt receive a position for some time it falls back to automatic mode. I am testing it now with posting the valve_position every 5 minutes. and so far it stays in trv 1

DapDallas, 2022-01-14:
> issue still open, did anyone have this pb ?

i have the same problem .. i bought the device specifically because of the feature. If you have an fix, please post it here.

sygys, 2022-01-14:
post the valve position every 5 minutes to the valve and it will stay in the valve mode. there is one problem last night the temperature dropped below 15 degrees here and for some reason the valve did go back to automatic temperature based mode when it dropped under the 15 degrees. Im still testing what could have caused this. The heating mode was on auto so it could be that this was the problem. i now put heating mode to off, wet trv_mode to 1 and now sending the valve position to the device every 5 minutes. it needs to run a few days before i can confirm or deny this works. But i have good hopes it will

closed 2022-02-22 by github-actions[bot]