Koenkk/zigbee2mqtt #9971: Moes BRT-100-TRV Errors after update to 1.22

dedors, 2021-12-03:
After the update to 1.22.1 from 1.21 I have the following 2 errors in my HA log:

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:566
First occurred: 05:11:44 (6 occurrences)
Last logged: 05:12:04

Error parsing value: access to attribute 'update' of 'dict' object is unsafe. (value: {"battery":69,"boost_heating":"OFF","boost_heating_countdown":0,"boost_heating_countdown_time_set":301,"child_lock":"UNLOCK","current_heating_setpoint":20,"eco_mode":"OFF","eco_temperature":20,"last_seen":"2021-12-03T04:11:44.215Z","linkquality":39,"local_temperature":22.5,"local_temperature_calibration":-2,"max_temperature":45,"min_temperature":5,"position":0,"preset":"programming","program_saturday":{"saturday":"6h:0m 24°C, 12h:0m 23°C, 14h:30m 22°C, 17h:30m 21°C "},"program_sunday":{"sunday":" 6h:0m 19°C, 12h:30m 20°C, 14h:30m 21°C, 18h:30m 20°C "},"program_weekday":{"weekday":" 5h:0m 20°C, 11h:30m 21°C, 13h:30m 22°C, 17h:30m 23°C "},"programming_mode":"05:00/20°C 11:30/21°C 13:30/22°C 17:30/23°C 06:00/24°C 12:00/23°C 14:30/22°C 17:30/21°C 06:00/19°C 12:30/20°C 14:30/21°C 18:30/20°C","update_available":null,"window":"CLOSED","window_detection":"OFF"}, template: {{ value_json['update']['state'] == "available" }})
```

```
Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:165
Integration: MQTT (documentation, issues)
First occurred: 05:11:44 (6 occurrences)
Last logged: 05:12:04

Invalid value for number.thermostat_moes_brt_100_black_boost_heating_countdown_time_set: 300 (range 0.0 - 100.0)
```
the z2m UI shows a range from 100-900:
```
boost_heating_countdown_time_set
Boost Time Setting 100 sec - 900 sec, (default = 300 sec)
```

Device: Moes BRT-100-TRV / TS0601 / _TZE200_b6wax7g0

FIY: With my other device Moes HY368 / TS0601 / _TZE200_ckud7u2l no such error occures.

I did a firmware update just to be sure. 

## Debug info
Zigbee2MQTT version: 1.22.1
Adapter hardware: CC1352P-2
Adapter firmware version: 20211114

robertalexa, 2021-12-03:
Similar issue described here:
https://github.com/Koenkk/zigbee2mqtt/issues/9972

Might be worth keeping an eye there

robertalexa, 2021-12-04:
First error might already be fixed on dev. Have a look here 
https://github.com/Koenkk/zigbee2mqtt/issues/9939

closed 2021-12-05 by Koenkk

dedors, 2021-12-13:
Should I open up a new one or still continue here?

After I switched to dev, all "range problems" were gone but a new one came:
` [homeassistant.components.mqtt.number] Invalid value for number.thermostat_moes_brt_100_black_max_temperature: 45 (range -10.0 - 35.0)`
I was able to change that value to a acceptable one (it only proposed values up to 35 there too) on the device itself, and I was able to change it's value with HA (confirmed valuechange on the device itself), still the same error continued to show up.

So I removed the device from z2m and rejoined, since then almost all values are NULL (I let it run for over 24h):
`MQTT publish: topic 'zigbee2mqtt/Thermostat Moes BRT-100 Black', payload '{"battery":null,"boost_heating":null,"boost_heating_countdown":null,"boost_heating_countdown_time_set":null,"child_lock":null,"eco_mode":null,"eco_temperature":null,"last_seen":"2021-12-13T03:37:16.910Z","linkquality":72,"local_temperature_calibration":null,"max_temperature":null,"min_temperature":null,"position":null,"programming_mode":null,"update":{"state":null},"update_available":null,"window":null,"window_detection":null}'`