Koenkk/zigbee2mqtt #9973: Ability to expose switch as light from inside WebUI

thinkl33t, 2021-12-03:
## Is your feature request related to a problem? Please describe
I currently use zigbee2mqtt almost exclusively through the web interface, which is amazing!  However, for the light switches in my house, i use Xiaomi QBKG04LM light switches, which by default show up to homeassistant (and hence google assistant) as switches.

## Describe the solution you'd like
I'd like a setting added to the device in the webui which allows automatically sets up the [correct config](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#exposing-switch-as-a-light) for homeassistant autodiscovery, without having to log into the docker host, stop z2m, edit the config file manually for 20+ devices, and restart it.  A simple checkbox or dropdown would be perfect.

This would also allow for keeping the autodiscovery code updated and working on new homeassistant releases, to avoid [issues](https://github.com/Koenkk/zigbee2mqtt/issues/8033) like we saw in July.

## Describe alternatives you've considered
Current alternative is editing the config file manually, which can be a bit error prone.

## Additional context
Add any other context or screenshots about the feature request here.

adizag, 2022-10-11:
@Koenkk  It is marked as closed, but there is still no GUI option for exposing as light. Couldn't find it in the guide as well.
That option is very necessary, could be great  to have it please.

cwildfoerster, 2023-02-13:
ping @Koenkk

closed 2023-03-25 by github-actions[bot]

thargy, 2023-03-30:
Yeah, this has me stumped, can never find the place in the docs with the example of how to configure in YAML and couldn't get it working.  Would love a simpler GUI way of doing this.

thargy, 2023-07-05:
For anyone else, you may wish to look at the Home Assistant Helper - "Change device type of a switch", which you can create to change a switch to a light.

flackovic, 2024-01-14:
> For anyone else, you may wish to look at the Home Assistant Helper - "Change device type of a switch", which you can create to change a switch to a light.

After following [steps their documentation](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#exposing-switch-as-a-light) three times, unsucessfully, this worked with 2 clicks. Thanks!