Koenkk/zigbee2mqtt #9974: Eurotronics Spirit SPZB0001 does not expose "occupied_heating_setpoint" to HA

breti169, 2021-12-03:
Hi All,

I have Home Assistant OS + Zigbee2Mqtt + CC2531. I try to use Eurotronics Spirit SPZB0001 thermostate - but it seems that its integration is not really usable.

Can you please have a look, why the target temerature "**occupied_heating_setpoint**" is not exposed to HA as entity? 

There are a bunch of more or less useful properties exposed - f.i "update available" or "local temperature calibration". For my point of view, the most important property of a thermostate is "occupied_heating_setpoint". But exactly this is not available as HA entity. Is there a special reason for this?

The wanted behavior would be:

- expose occupied_heating_setpoint as sensor, to read out its current value by HA automation
- expose occupied_heating_setpoint  as actor, to set its value by HA automation 

As I have seen - integration of Eurotronics Spirit seems to have a long history. Good to know, that a lot of people are working hard to make this ingredible smarthome Platform possible. Thank you for this and for your great work!

Viele Grüße von Stefan

BTW: I have bougt 8 identical Eurotronics Spirit Devices. 6 of them joins without problems - 2 can't be connected. The working devices all have the same Firmwareversion (22190930) and Firmwaredate (20191014). Very strange - but its not important - it only a hint for others they have the same experience.

Koenkk, 2021-12-03:
Should already be fixed in z2m 1.22.1

breti169, 2021-12-03:
Thanks for responding. But unfortenately it is'nt fixed. To be sure to have the lastest z2m version, I have installed z2m-edge. Also I have deleted and resetted and joined one of my SPZB0001 completely new.

Here you see my added SPZB0001 device in z2m exposes-tab
[image]

And here you see it in ha
[image]

There is no occupied_heating_setpoint entity in HA.

What is wrong?

Koenkk, 2021-12-04:
Ah I was wrong, it is indeed not exposed as a separate entity. This value should be available in the `climate` entry exposed by this device, the occupied heating setpoint is set as the [temperature state topic](https://www.home-assistant.io/integrations/climate.mqtt/#temperature_state_topic)

Koenkk, 2021-12-06:
I don't understand what you expect to be fixed, the `occupied_heating_setpoint` is already exposed to the `climate` entry as I explained in https://github.com/Koenkk/zigbee2mqtt/issues/9974#issuecomment-986033660, all thermostats are integrated in this way.

breti169, 2021-12-09:
Oh - sorry - I dit'nt understud the implemented climate logic correctly. For automation purposes, I have access the target temperature by Service "climate.set_temperature" and select wanted device(s). My first try was to automate the "occupied_heating_setpoint " entity  directly. This was wrong. Thanks for your support. I close this Issue.

closed 2021-12-09 by breti169